sdeigm/uni-meter #4: not an issue, but would be very helpful for many users - docker compose instructions

martin-st-81, 2024-12-20:
Is it possible to provide instructions for docker compose?

sdeigm, 2024-12-23:
Sure, I will try to provide a docker image together with some instructions ... just give me some time :-)

closed 2025-01-31 by sdeigm

sdeigm, 2025-01-31:
So, i created a Dockerimage. It will be pushed as

sdeigm/uni-meter:latest

to dockerhub.

I have set up a separate uni-meter-docker github project which contains a docker-compose file.

The project has no readme yet. The docker-compose file will map 3 ports to the container 80/TCP, 1010/UDP and 5353/UDP

The first port if for the Shelly's WebServer, the second one for RCP over UDP and the last one for the avahi mDNS daemon.

If you don't need one of the features, you can just disable the port mapping.

For the avahi daemon you have to decide if you want to use the one from the real host or the one from the docker container. If you want to use the one from the docker container you have to stop the one from the host (if installed at all):