sdeigm/uni-meter #5: SHRDZM Smartmeter Kundenschnittstellen Modul mittels UDP einbinden möglich ?

martin-st-81, 2025-01-03:
Mit meinen Kenntnissen konnte ich dieses tolle Script leider nicht erweitern. 
Aufgrund baulicher Gegebenheiten kann ich für meinen Speicher Hoymiles MS-A2 keinen Shelly Pro3EM verbauen. 

Zumindest in Österreich werden die SmartMeter oft direkt mit einem Device von SHRDZM ausgelesen.
https://cms.shrdzm.com/produkt/smartmeter-modul/

Dieses bietet die Möglichkeit mittels "UDP Send" die Daten weiterzuleiten 

hier wäre 192.168.25.54 die IP des Ziel Devices -  zB das UniMeter - und der Port zB 9522 frei wählbar
[image]

Die Daten werden dann folgendermaßen via UDP übertragen, wobei "AC0BFBDA717B" die ID vom SHRDZM ist

{"id":"AC0BFBDA717B","data":{
"timestamp":"2025-01-03T20:38:45",
"1.8.0":"12570070",
"2.8.0":"1860924",
"1.7.0":"403",
"2.7.0":"0",
"32.7.0":"229.60",
"52.7.0":"230.20",
"72.7.0":"232.10",
"31.7.0":"0.68",
"51.7.0":"1.81",
"71.7.0":"0.58",
"13.7.0":"57.00",
"16.7.0":"403",
"uptime":"0000:00:01:21",
"UTC":"2025-01-03T19:38:52"
}}

Wäre es möglich das Device SHRDZM in das Script UniMeter einzubauen ? Oder muss man hier nach einer anderen Möglichkeit suchen.

sdeigm, 2025-01-04:
Hallo Martin, klar versuche ich das einzubauen. Du musst mich dann nur beim Live-Test unterstützen.

martin-st-81, 2025-01-04:
Danke - natürlich stehe ich für Tests und Feedback zur Verfügung. 
Die Grundinstallation vom UniMeter laut Anleitung ist bereits durchgeführt.

sdeigm, 2025-01-04:
Ich sitze hier gerade schon an der Implementierung. Ich habe mir gerade mal deine Beispieldaten von oben angesehen. Da kommen bei mir gleich ein paar Fragen auf. Prinzipiell macht das SHRDZM Device ja genau das selbe was bei mir der VzLogger über den Infrarot-Lesekopf macht. 

Der Wert "16.7.0" ist bei SML die aktuelle Leistung. Bei dir steht dort der Wert 403 was bei HTTP ja der "Forbidden" Statuscode ist. Diesen Wert muss man in Deutschland explizit im Stromzähler freigeben in dem man eine PIN eingibt. Kann es sein, das dein Stromzähler dafür im Augenblick auch noch gesperrt ist? Oder ist das wirklich Zufall und die Leistung hat zum dem Zeitpunkt gerade 403 Watt betragen? Den PIN um den Kanal 16.7.0 im meinem Zähler freizuschalten habe ich hier in Norddeutschland von meinem Netzbetreiber bekommen.

Die Werte 32.7.0, 52.7.0 und 72.7.0 scheinen ja die Spannung auf den 3 Phasen zu sein, und 31.7.0, 51.7.0 und 71.7.0 der momentane Strom auf den 3 Phasen. Ggf. kann ich also die Leistung auf den 3 Phasen auch über diese Werte ausrechnen. Ich hoffe nur das ist dann auch wirklich die Wirkleisung und nicht Scheinleistung. 

Ich baue das jedenfalls erst mal so ein.

martin-st-81, 2025-01-04:
Das war nur Zufall, anbei die derzeitigen Werte
und ja den PIN musste ich vom Netzbetreiber erfragen und eintragen, der SHDRZM überträgt die Werte aufgrund des PINs jetzt aber unverschlüsselt via UDP

[image]

Das ist der Stromzähler welcher in NÖ und bei mir verwendet wird, darin wird ebenfalls auch auf die Werte eingegangen.
https://www.netz-noe.at/Download-(1)/Smart-Meter/218_3_SmartMeter_Drehstromzahler.aspx

Kennzahl 1.7.0: aktueller Stromverbrauch, Screenshot 15W
Kennzahl 2.7.0: aktuelle Einspeisung, Screenshot 0W 
Kennzahl 1.8.0: Dieser Wert zeigt den Strombezug. 
Kennzahl 2.8.0: Dieser Wert zeigt die Menge an Strom, welche durch Ihre Erzeugungsanlage (z.B. Photovoltaikanlage) in das Stromnetz des Netzbetreibers eingespeist wurde

martin-st-81, 2025-01-04:
bezgl. dem Wert 16.7.0, das ist der tatsächliche Bezug/Einspeisung mit + oder -
- ->Einspeisung
+ -> Bezug 

Kennzahl 1.7.0 & Kennzahl 2.7.0 sind immer positiv

zb: Bezug 1200 W
Kennzahl 1.7.0: 1200
Kennzahl 2.7.0: 0
Kennzahl 16.7.0: 1200

Einspeisung 700W
Kennzahl 1.7.0: 0
Kennzahl 2.7.0: 700
Kennzahl 16.7.0: -700

sdeigm, 2025-01-04:
Ok, sehr gut. Das ist alles analog zum VzLogger.

Ich habe jetzt eine erste Version fertig die du testen könntest. Kannst du dir das selber mit maven bauen oder soll ich auf github ein Package veröffentlichen?

martin-st-81, 2025-01-04:
bitte ein Package auf github veröffentlichen, mit maven habe ich noch keine Erfahrungen.
Wenn die Stromwerte im Log auftauchen kann ich diese dann heute auch noch testen.

sdeigm, 2025-01-04:
Es gibt jetzt ein 0.9.4-SNAPSHOT Package. Das könntest du bitte einmal testen. Die uni-meter.conf Datei müsste folgendermassen aussehen:

uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.shrdzm"

  input-devices {
	shrdzm {
	  port = 9522
	}
  } 
}

Den ganzen Teil unter input-devices musst du nicht unbedingt angeben. Das ist der Default. Nur wenn du einen anderen Port verwenden willst bitte hier ändern.

martin-st-81, 2025-01-04:
Derzeit habe ich keinen Zugang zum Hoymiles Speicher. Sind die Messwerte ersichtlich wenn ich das LOG auf Debug stelle ?
Der Shelly wird jedenfalls in der Shelly App am Handy gefunden

[image]

"starting receive loop for notifications from the SMA energy meter" ist das optisch oder ein Fehler ?

meine uni-meter.conf lautet:

uni-meter {
output = "uni-meter.output-devices.shelly-pro3em"

input = "uni-meter.input-devices.shrdzm"

input-devices {
shrdzm {
port = 9522
}
}
}

sdeigm, 2025-01-04:
Das SMA in der Ausgabe ist ein Copy & Paste Fehler von mir.

Wenn du den Loglevel auf Debug stellst solltest du Ausgabe wie folgt sehen:

`25-01-04 17:40:35.332 INFO  uni-meter                - ###########################################################################
25-01-04 17:40:35.332 INFO  uni-meter                - # Universal electric meter converter 0.9.4-SNAPSHOT (2025-01-04 15:55:14) #
25-01-04 17:40:35.332 INFO  uni-meter                - ###########################################################################
25-01-04 17:40:35.332 INFO  uni-meter                - initializing actor system
25-01-04 17:40:35.820 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-01-04 17:40:35.924 INFO  uni-meter.controller     - creating Shelly3EM output device
25-01-04 17:40:35.930 INFO  uni-meter.controller     - creating ShrDzm input device
25-01-04 17:40:36.129 INFO  uni-meter.input          - starting receive loop for notifications from the SMA energy meter
25-01-04 17:40:36.755 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-01-04 17:43:35.092 DEBUG uni-meter.input          - received ShrDzm packet: ShrDzmPacket[id=AC0BFBDA7XYZ, data={timestamp=2025-01-03T20:38:45, 1.8.0=12570070, 2.8.0=1860924, 1.7.0=403, 2.7.0=0, 32.7.0=229.60, 52.7.0=230.20, 72.7.0=232.10, 31.7.0=0.68, 51.7.0=1.81, 71.7.0=0.58, 13.7.0=57.00, 16.7.0=403, uptime=0000:00:01:21, UTC=2025-01-03T19:38:52}]
25-01-04 17:43:35.108 DEBUG uni-meter.input          - power data found for channel 16.7.0: 403
25-01-04 17:43:35.108 DEBUG uni-meter.input          - consumption data found for channel 1.8.0: 12570070
25-01-04 17:43:35.108 DEBUG uni-meter.input          - production data found for channel 2.8.0: 1860924
25-01-04 17:43:35.108 DEBUG uni-meter.output         - power phase 0: PowerData[power=134.33333333333334, apparentPower=134.33333333333334, powerFactor=1.0, current=0.5840579710144927, voltage=230.0, frequency=50.0]
25-01-04 17:43:35.124 DEBUG uni-meter.output         - power phase 1: PowerData[power=134.33333333333334, apparentPower=134.33333333333334, powerFactor=1.0, current=0.5840579710144927, voltage=230.0, frequency=50.0]
25-01-04 17:43:35.124 DEBUG uni-meter.output         - power phase 2: PowerData[power=134.33333333333334, apparentPower=134.33333333333334, powerFactor=1.0, current=0.5840579710144927, voltage=230.0, frequency=50.0]
25-01-04 17:43:35.124 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=4190.0233333333335, totalProduction=620.308]
25-01-04 17:43:35.124 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=4190.0233333333335, totalProduction=620.308]
25-01-04 17:43:35.124 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=4190.0233333333335, totalProduction=620.308]`

sdeigm, 2025-01-04:
Die aktuellen Daten des virtuellen Shelly kannst du auch per Browser abfragen:

http://<shelly-ip>/rpc/EM.GetStatus?id=0
http://<shelly-ip>/rpc/EMData.GetStatus?id=0

martin-st-81, 2025-01-04:
Für mein Verständnis klappt es wie gewünscht, danke für die rasche Hilfe und Umsetzung.
Sobald ich es mit dem Hoymiles testen kann, gebe ich bescheid 

25-01-04 20:05:57.519 DEBUG uni-meter.input          - received ShrDzm packet: ShrDzmPacket[id=AC0BFBDA7XYZ, data={timestamp=2025-01-04T20:05:50, 1.8.0=12576089, 2.8.0=1862878, 1.7.0=526, 2.7.0=0, 32.7.0=228.30, 5          2.7.0=229.40, 72.7.0=229.20, 31.7.0=0.66, 51.7.0=1.42, 71.7.0=1.37, 13.7.0=66.60, 16.7.0=526, uptime=0000:03:52:07, UTC=2025-01-04T19:05:56}]
25-01-04 20:05:57.521 DEBUG uni-meter.input          - power data found for channel 16.7.0: 526
25-01-04 20:05:57.522 DEBUG uni-meter.output         - power phase 0: PowerData[power=175.33333333333334, apparentPower=175.33333333333334, powerFactor=1.0, current=0.7623188405797102, voltage=230.0, frequency=50.          0]
25-01-04 20:05:57.522 DEBUG uni-meter.output         - power phase 1: PowerData[power=175.33333333333334, apparentPower=175.33333333333334, powerFactor=1.0, current=0.7623188405797102, voltage=230.0, frequency=50.          0]
25-01-04 20:05:57.523 DEBUG uni-meter.output         - power phase 2: PowerData[power=175.33333333333334, apparentPower=175.33333333333334, powerFactor=1.0, current=0.7623188405797102, voltage=230.0, frequency=50.          0]
25-01-04 20:05:57.524 DEBUG uni-meter.input          - consumption data found for channel 1.8.0: 12576089
25-01-04 20:05:57.525 DEBUG uni-meter.input          - production data found for channel 2.8.0: 1862878
25-01-04 20:05:57.525 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=4192.029666666666, totalProduction=620.9593333333333]
25-01-04 20:05:57.525 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=4192.029666666666, totalProduction=620.9593333333333]
25-01-04 20:05:57.525 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=4192.029666666666, totalProduction=620.9593333333333]

[image]
[image]
[image]

martin-st-81, 2025-01-05:
Bezgl. deiner Frage "Ich hoffe nur das ist dann auch wirklich die Wirkleistung und nicht Scheinleistung."
13.7.0 ist der Leistungsfaktor, der das Verhältnis von Wirk zu Scheinleistung ist.

Ich sehe, dass hier derzeit die power phase 0/1/2 das Drittel von 16.7.0 - 526 - sind

power phase 0 -> 32.7.0 x 31.7.0 x (13.7.0 / 100) = 228,30 x 0,66 x 0,666 = 100,351548
power phase 1 -> 52.7.0 x 51.7.0 x (13.7.0 / 100) = 229,40 x 1,42 x 0,666 = 216,948168
power phase 2 -> 72.7.0 x 71.7.0 x (13.7.0 / 100) = 229,20 x 1,37 x 0,666 = 209,126664

macht in Summe 526,42638 ~ 526W

[image]

Wäre super wenn man dies noch implementieren könnte. Seitens Hoymiles ist es ja angeblich angedacht, dass man mehrere Hoymiles MS-A2 zukünftig an jeweils eine Phase hängen kann. 

1.8.0 – Wirkleistung A+
2.8.0 – Wirkleistung A-
1.7.0 – Momentanleistung P+
2.7.0 – Momentanleistung P-
32.7.0 – Spannung L1
52.7.0 – Spannung L2
72.7.0 – Spannung L3
31.7.0 – Strom L1
51.7.0 – Strom L2
71.7.0 – Strom L3
13.7.0 - Leistungsfaktor
16.7.0 - Momentane Wirkleistung kombiniert in + & -

closed 2025-01-05 by sdeigm

plustig73, 2025-01-05:
> Die aktuellen Daten des virtuellen Shelly kannst du auch per Browser abfragen:
> 
> `http://<shelly-ip>/rpc/EM.GetStatus?id=0` `http://<shelly-ip>/rpc/EMData.GetStatus?id=0 `

Hallo, sorry wenn ich mich hier einfach einklinke.

Beim Aufrufen von

http://<shelly-ip>/rpc/EMData.GetStatus?id=0

kommen bei mir nur Nullwerte zurück. Normal oder sollte da bei mir auch was kommen? 

`{
    "id": 0,
    "a_total_act_energy": 0,
    "a_total_act_ret_energy": 0,
    "b_total_act_energy": 0,
    "b_total_act_ret_energy": 0,
    "c_total_act_energy": 0,
    "c_total_act_ret_energy": 0,
    "total_act": 0,
    "total_act_ret": 0
}`

Frage die Daten über den SMA SunnyHomeManager2 ab.

FYI

http://<shelly-ip>/rpc/EM.GetStatus?id=0 sieht so aus:

`{
    "id": 0,
    "a_current": 1.765,
    "a_voltage": 232.32,
    "a_act_power": 234,
    "a_aprt_power": 308.6,
    "a_pf": 0.7582631237848347,
    "a_freq": 50,
    "b_current": 3.61,
    "b_voltage": 232.365,
    "b_act_power": 782.1,
    "b_aprt_power": 825.8,
    "b_pf": 0.98,
    "b_freq": 50,
    "c_current": 0.953,
    "c_voltage": 232.437,
    "c_act_power": 68.7,
    "c_aprt_power": 204.3,
    "c_pf": 0.98,
    "c_freq": 50,
    "total_current": 6.328,
    "total_act_power": 1084.8,
    "total_aprt_power": 0
}`

Grüße aus dem PV-Forum

sdeigm, 2025-01-05:
@plustig73: Ja da sollte etwas kommen. Ich schaue mir das mal an, ob da beim SunnyHomeManager etwas falsch ist. Ggf. könntest du mir eine Debug-Log zukommen lassen, da ich keinen SunnyHomeManager zum Testen habe.

sdeigm, 2025-01-05:
@martin-st-81: Die getrennte Ausweisung der 3 Phasen kann ich noch einbauen. Kannst du dafür bitte wenn die Sonne scheint mal testen, ob ein Strom negativ wird, wenn du einspeist? Sonnst kann ich so die Gesamtleistung nicht berechnen. Das ist aber nur Kosmetik, dass du dann in der Übersicht für alle 3 Phasen unterschiedliche Werte angzeigt bekommst. Im Augenblick drittele ich nur die Gesamtleistung und verteile sie gleichmäßig auf alle 3 Phasen.

Bzgl. des Hoymiles und Phasen weiß ich nicht ganz genau, was da implementiert werden soll. Ein Hoymiles Speicher hängt über die Schuko-Steckdose immer nur an einer Phase.

sdeigm, 2025-01-05:
@plustig73: in meinem Test-Setup werden da auch für SMA Daten angezeigt. Verwendest du schon die Version 0.9.3? Ansonsten bitte einen neuen Issue erzeugen und ein DEBUG bzw. sogar TRACE Log anfügen.

plustig73, 2025-01-05:
> @plustig73: Ja da sollte etwas kommen. Ich schaue mir das mal an, ob da beim SunnyHomeManager etwas falsch ist. Ggf. könntest du mir eine Debug-Log zukommen lassen, da ich keinen SunnyHomeManager zum Testen habe.

Hallo,

ja, Version 0.9.3.

[uni-meter.log](https://github.com/user-attachments/files/18310095/uni-meter.log)

Im Anhang das log-file... 

Mach dir keinen Stress wegen mir, und falls du jemanden zum SMA-SHM2 testen brauchst sag Bescheid.

Grüße

plustig73, 2025-01-05:
> Hallo,
>
> ja, Version 0.9.3.
>
> [uni-meter.log](https://github.com/user-attachments/files/18310095/uni-meter.log)
>
> Im Anhang das log-file...
>
> Mach dir keinen Stress wegen mir, und falls du jemanden zum [...]

So, falscher Alarm, nachdem ich den Dienst fürs logfile erstellen neu gestartet habe kommen bei Aufrufen Werte zurück!

Sorry dafür!

martin-st-81, 2025-01-06:
Anbei ein Bsp für die Einspeisung. Strom bleibt positiv beim Einspeisen.
Wäre super wenn du die getrennte Ausweisung der 3 Phasen bitte noch einbauen könntest.

denke so sollte es passen
power phase 0 -> 32.7.0 x 31.7.0 x (13.7.0 / 100)
power phase 1 -> 52.7.0 x 51.7.0 x (13.7.0 / 100)
power phase 2 -> 72.7.0 x 71.7.0 x (13.7.0 / 100)

Ev. gleich auch den optischen copy/paste Fehler  "starting receive loop for notifications from the SMA energy meter" mitmachen, denke dann könnt man das Thema SHRDZM als perfekt eingebunden ablegen, DANKE für deine tolle Arbeit. 

Bezug: 853W
[image]
Einspeisung: 103W
[image]

sdeigm, 2025-01-06:
Die fehlerhafte SMA Ausgabe ist schon entfernt. Es gibt eine richtige Version 0.9.4 (kein Snapshot) in der diesbezüglich alles korrekt sein sollte.

Das getrennte Ausweisen der 3 Phase ist aber problematisch. Es gilt nämlich die Bedingung, dass die Summe der 3 Phasen der Gesamtbezug/die Gesamteinspeisung ist. 

In deinem ersten Beispiel mit 853 Watt Bezug ist das alles eindeutig. Beim Beispiel mit den 103 Watt Einspeisung ergibt sich aber nach deiner Formel von oben erstmal ohne Beachtung von Vorzeichen:

Phase 1: 90,764 Watt
Phase 2: 75,643 Watt
Phase 3: 117,855 Watt

Da du insgesamt 103 Watt einspeist nehme ich aus den Zahlen an, das du auf den Phasen 2+3 einspeist, zusammen 193,498 Watt
und auf der Phase 1 gleichzeitig 90,764 Watt beziehst.

Rechnerisch ergibt sich damit ein ungefähre Gesamteinspeisung von ungefähr 103 Watt.

Irgendwoher brauche ich noch die Info welche Phase gerade einspeist und welche bezieht.

martin-st-81, 2025-01-06:
"Irgendwoher brauche ich noch die Info welche Phase gerade einspeist und welche bezieht." 
Dies gibt das SHRDZM Modul bzw. das saldierende SmartMeter leider nicht her.
Aus dem Grund kann wohl keine allgemeine Aussage getroffen werden. Ich weiß an welchen Phasen ich einspeise, dass muss allerdings nicht pauschal für andere zustimmen.
Belassen wir es so, in der derzeitigen Funktionsweise des Hoymiles MS-A2 wird es egal sein.
Danke für deinen tollen Support und Leistung bei diesem Script.

martind40, 2025-02-22:
Hab nun auch UDP am SHRDZM aktiviert und die Standardkonfiguration aus https://github.com/sdeigm/uni-meter] funktioniert umgehend.

https://github.com/sdeigm/uni-meter/issues/56#issuecomment-2676114409 kann geschlossen werden, danke für die rasche Hilfe.