sdeigm/uni-meter #8: Feature Request: generic MQTT source as Input?

krikk, 2025-01-13:
it would be nice to have a configurable MQTT source as input, this way your "emulated" shelly pro 3em could get data from way more source (for me from my old shelly 3EM for example) 

great work so long!!

krikk, 2025-01-13:
no, there is no "standard" how different energy meters provide der readings, so best would be the following configurable properties:
MQTT-Server: IP or Hostname
MQTT-Port: Listening Port (usually 1883)
MQTT-User
MQTT-Password (for password protected MQTT Brokers

the most simple mqtt-topics would be like that, where the topic directly provides the needed data:

MQTT Topic Power 1: solar/powermeter/power1
MQTT Topic Power 2: solar/powermeter/power3
MQTT Topic Power 3: solar/powermeter/power3

[image]

the more advanced setup would need an additional JSON-Path Propertyper power topic if the MQTT-Topic gives a JSON-Object like this:

`{"Time":"2025-01-13T15:03:41","ENERGY":{"TotalStartTime":"2021-02-15T23:59:31","Total":3.324,"Yesterday":0.005,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}`

StPinkert, 2025-01-18:
Same here, would absolutely appreciate MQTT as input device. I have an Hichi-Sensor (Tasmota with SML) on my electricity meter that pushes the values as JSON object to a mqtt-server. Here is an example of my readings:

tele/smlreader/STATE {"Time":"2025-01-18T01:57:30","Uptime":"54T04:34:34","UptimeSec":4682074,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"OFF","Wifi":{"AP":1,"SSId":"********","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:30","Main":{"power":466,"counter_pos":1306.661,"counter_neg":148.611}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:31","Main":{"power":470}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:31","Main":{"power":467}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:33","Main":{"power":470}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:34","Main":{"power":468}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:35","Main":{"power":465}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:36","Main":{"power":472}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:37","Main":{"power":469}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:38","Main":{"power":461}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:39","Main":{"power":464}}
tele/smlreader/SENSOR {"Time":"2025-01-18T01:57:40","Main":{"power":460}}

"tele/smlreader/SENSOR is the topic, the message a in JSON format. But be prepared other sensors may send a number instead of a JSON (as @krikk mentioned)

In my case the sensor pushes the power value immediately (one per second) and all other values every 10 seconds. But that is a custom setting in my Tasmota script. By default there is an update interval where all values are send together (10 seconds in my case - default is 60 seconds if i remember right)

As @krikk said, there is no standard. My previous electricity meter had much more data (power,current,voltage,phase angle for all three phases and frequency). My current one tells only the (combined) power via IR-Sensor, as you can see in the example above. (counter_pos and counter_neg are the readings of OBIS 1.8.0 and 2.8.0)

I'm a programmer by myself, but have absolutely no clue of Java. For my work i use mainly python (and sometimes php, if it's web based stuff). I could do testing and giving feedback, if it helps.

sdeigm, 2025-01-18:
@StPinkert thanks for providing that detailed input. The case that the same topic somtimes contains a value and sometimes not (like the energy readings in your setup) was not yet covered by my design. I will improve that.

sdeigm, 2025-01-18:
Hi @krikk and @StPinkert I have just created a first 0.9.7-SNAPSHOT test version. You can either download it from the packages or build it yourself using the branch of this feature request.

Attached you will find two uni-meter.conf files which should be already set up for your examples. You just have to rename them and adjust the IP addresses to your needs.

Happy testing and have a nice weekend!

[uni-meter - krikk.conf.txt](https://github.com/user-attachments/files/18463467/uni-meter.-.krikk.conf.txt)
[uni-meter - StPinkert.conf.txt](https://github.com/user-attachments/files/18463466/uni-meter.-.StPinkert.conf.txt)

StPinkert, 2025-01-18:
Wow, that was a fast response. Didn't expected that.

First test worked nearly like a charm. Had to change the ip and add username and password to the configuration file. In the file mqtt.java i saw, that it was already prepared for mqtt with authentication.

 Here's the log output (debug level):
`
25-01-18 13:00:38.820 INFO  uni-meter                - ###########################################################################
25-01-18 13:00:38.823 INFO  uni-meter                - # Universal electric meter converter 0.9.7-SNAPSHOT (2025-01-18 12:32:49) #
25-01-18 13:00:38.823 INFO  uni-meter                - ###########################################################################
25-01-18 13:00:38.823 INFO  uni-meter                - initializing actor system
25-01-18 13:00:39.060 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-01-18 13:00:39.174 INFO  uni-meter.controller     - creating Shelly3EM output device
25-01-18 13:00:39.180 INFO  uni-meter.controller     - creating MQTT input device
25-01-18 13:00:39.329 INFO  uni-meter.input          - subscribing to topic: tele/smlreader/SENSOR
25-01-18 13:00:39.590 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-01-18 13:00:39.729 INFO  uni-meter.input          - MQTT stream connected
25-01-18 13:00:39.947 DEBUG uni-meter.output         - power phase 0: PowerData[power=-84.66666666666667, apparentPower=-84.66666666666667, powerFactor=1.0, current=-0.3681159420289855, voltage=230.0, frequency=50.0]
25-01-18 13:00:39.958 DEBUG uni-meter.output         - power phase 1: PowerData[power=-84.66666666666667, apparentPower=-84.66666666666667, powerFactor=1.0, current=-0.3681159420289855, voltage=230.0, frequency=50.0]
25-01-18 13:00:39.959 DEBUG uni-meter.output         - power phase 2: PowerData[power=-84.66666666666667, apparentPower=-84.66666666666667, powerFactor=1.0, current=-0.3681159420289855, voltage=230.0, frequency=50.0]
25-01-18 13:00:39.959 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=0.0, totalProduction=0.0]
25-01-18 13:00:39.961 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=0.0, totalProduction=0.0]
25-01-18 13:00:39.961 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=0.0, totalProduction=0.0]
[...]
25-01-18 13:00:42.875 DEBUG uni-meter.output         - power phase 0: PowerData[power=-85.33333333333333, apparentPower=-85.33333333333333, powerFactor=1.0, current=-0.3710144927536232, voltage=230.0, frequency=50.0]
25-01-18 13:00:42.875 DEBUG uni-meter.output         - power phase 1: PowerData[power=-85.33333333333333, apparentPower=-85.33333333333333, powerFactor=1.0, current=-0.3710144927536232, voltage=230.0, frequency=50.0]
25-01-18 13:00:42.875 DEBUG uni-meter.output         - power phase 2: PowerData[power=-85.33333333333333, apparentPower=-85.33333333333333, powerFactor=1.0, current=-0.3710144927536232, voltage=230.0, frequency=50.0]
25-01-18 13:00:42.875 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=0.0, totalProduction=0.0]
25-01-18 13:00:42.876 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=0.0, totalProduction=0.0]
25-01-18 13:00:42.876 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=0.0, totalProduction=0.0]
25-01-18 13:00:42.876 DEBUG uni-meter.output         - power phase 0: PowerData[power=-85.33333333333333, apparentPower=-85.33333333333333, powerFactor=1.0, current=-0.3710144927536232, voltage=230.0, frequency=50.0]
25-01-18 13:00:42.876 DEBUG uni-meter.output         - power phase 1: PowerData[power=-85.33333333333333, apparentPower=-85.33333333333333, powerFactor=1.0, current=-0.3710144927536232, voltage=230.0, frequency=50.0]
25-01-18 13:00:42.876 DEBUG uni-meter.output         - power phase 2: PowerData[power=-85.33333333333333, apparentPower=-85.33333333333333, powerFactor=1.0, current=-0.3710144927536232, voltage=230.0, frequency=50.0]
25-01-18 13:00:42.876 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=436.231, totalProduction=0.0]
25-01-18 13:00:42.876 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=436.231, totalProduction=0.0]
25-01-18 13:00:42.877 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=436.231, totalProduction=0.0]
25-01-18 13:00:42.877 DEBUG uni-meter.output         - power phase 0: PowerData[power=-85.33333333333333, apparentPower=-85.33333333333333, powerFactor=1.0, current=-0.3710144927536232, voltage=230.0, frequency=50.0]
25-01-18 13:00:42.877 DEBUG uni-meter.output         - power phase 1: PowerData[power=-85.33333333333333, apparentPower=-85.33333333333333, powerFactor=1.0, current=-0.3710144927536232, voltage=230.0, frequency=50.0]
25-01-18 13:00:42.877 DEBUG uni-meter.output         - power phase 2: PowerData[power=-85.33333333333333, apparentPower=-85.33333333333333, powerFactor=1.0, current=-0.3710144927536232, voltage=230.0, frequency=50.0]
25-01-18 13:00:42.877 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=436.231, totalProduction=49.684666666666665]
25-01-18 13:00:42.878 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=436.231, totalProduction=49.684666666666665]
25-01-18 13:00:42.878 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=436.231, totalProduction=49.684666666666665]
[... cut ...]
25-01-18 13:00:42.889 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=436.231, totalProduction=49.684666666666665]
totalProduction=49.684666666666665]
`

So from my point of view it look's good so far. Unfortunately i haven't the time for further testing and setting up my MS-A2 (i've got just a few days ago) and connecting the uni-meter to it. I'll do it tomorrow and keep you updated.

Nevertheless, that project here is perfect timing. :-) Otherwise i had to drill a hole through the house wall to connect the hoymiles directly to the MS-A2.

StPinkert, 2025-01-19:
Ich wechsel mal ins deutsch, das geht schneller und einfacher ;-)
Ich hab mein MS-A2 angeschlossen, eingerichtet und Firmwareupdate durchgeführt. Lief alles problemlos. Der Shelly 3EMPro/uni-meter ließ sich ebenfalls problemlos hinzufügen. Soweit kein Problem. 
Danach fing es an wild zu werden. Die gemessene Leistung mittels Hichi-Sensor am Stromzähler schwankte innerhalb von 16-20 Sekunden von 450W zu -360W und wieder zurück. 
Wenn ich mir das Log anschaue, scheint der uni-meter genau das getan zu haben, was er soll. Werte vom MQTT-Topic lesen und auf Anfrage an den MS-A2 liefern. Die Frage ist jetzt, woran könnte es liegen, dass sich die Regelung nicht stabilisiert und ständig zw. 0W und 800W Einspeiseleistung (den Wert hab ich als Max beim MS-A2 festgelegt) oszilliert?
Der Sensor am Stromzähler liefert die Werte mit 1Hz (eine Aktualisierung pro Sekunde). Der MS-A2 fragt aber ca. mit 2Hz ab. Wenn das oszillieren damit zusammenhängt, wird's schwierig. Die Stromzähler liefern meines Wissens nach alle nur 1x pro Sekunde die Werte.
Logdatei hänge ich mal mit an zur Analyse.

[uni-meter.log](https://github.com/user-attachments/files/18466709/uni-meter.log)

sdeigm, 2025-01-19:
Hm, das ist seltsam. Das hört sich so an, als wenn die Leistungsdaten zu träge geliefert werden und der Hoymiles dann übersteuert. Ich habe ein ähnliches Setup (IR-Lesekopf am VzLogger) und bei mir steuert es perfekt. Der VzLogger liefert auch nur pro Sekunde einen neuen Wert. Ein paar Sachen sollten wir mal überprüfen:

- Bei einem Bekannten, bei dem den Hoymiles in Betrieb genommen hatte, war die Leistung auf dem Stromzähler noch nicht per PIN freigeschaltet. Ich hatte da eine Logik in mein VzLogger-InputDevice eingebaut, die versucht, dann die Leistung aus den sich ändernden Zählerständen zu ermitteln. Das hat ein ähnliches Verhalten wie bei dir ergeben. Deshalb meine Frage: der 16.7.0 Leistungskanal ist an deinem Zähler freigeschaltet? Nicht dass der Hitichi-Sensor diese Daten auch nicht hat und dann versucht die Leistung aus den Zählerständen zu berechnen?

- Aus irgendeinem Grund hattest du den Zähler ja schon. Ist da irgend ein anderes Gerät das sich einschaltet, sobald eine Einspeisung stattfindet? Nicht das sich dass überlagert. Denn wenn der Hoymiles richtig steuert schwankt die Leistung die ganze Zeit um +-0.

- Es kann natürlich sein, dass über die Strecke Hitchi -> MQTT -> Uni-Meter -> Hoymiles zu viel Verzögerung ins Spiel kommt und die Regelung deshalb übersteuert. Das würde mich aber eigentlich wundern.

Wenn keiner der ersten beiden Punkte zutrifft und es wirklich das Problem ist, dass der Hoymiles mit 2Hz abfragt und der Zähler nur mit 1Hz liefert kann ich da noch mal eine Verzögerung einbauen, so dass der Hoymiles die Werte langsamer bekommt. Das lässt sich über die Websocket Verbindung drosseln. 

Heute Abend wenn mein Hoymiles wieder einspeist kann ich auch noch mal ein Log wie von dir oben ziehen und dann können wir vergleichen. Im Augenblick lacht aber die Sonne und der Speicher wird mit 1KW geladen.

StPinkert, 2025-01-19:
Zu deinen Fragen: 
- Der Sensor am Stromzähler liest nur aus und leitet die Werte an den MQTT-Broker weiter. Der berechnet selbst keine Werte. PIN ist freigeschaltet. 
- Den Zähler hatte ich, um meine Verbrauchsdaten/Einspeisung mitzuverfolgen und zu berechnen, wie hoch meine Eigennutzung ist bzw. wie viel Strom ich dem Netzbetreiber schenke. Dafür läuft bei mir NodeRed, der die Daten von der Balkonsolar und dem Stromzähler empfängt und mir grafisch aufbereitet. (Seit einiger Zeit läuft auch parallel HomeAssistant) Ja, aktuell habe ich noch eine Steckdosenleiste, an der ein paar USB-Ladegeräte hängen. Die schaltet sich allerdings erst ein, wenn ich mind. eine Minute lang konstant >=50W Überschuss habe (gesteuert über NodeRed). Das kann's also nicht sein.

Ich werde auch gleich noch mal schauen, wie er sich jetzt verhält, wenn Überschuss da ist.

StPinkert, 2025-01-19:
Ich habe mal ein paar stümperhafte Änderungen vorgenommen. Herzstück ist in der `ShellyPro3EM.java` in der Methode `createRpcResult` die folgende Zeile:
`case "EM.GetStatus" ->  request.id()%5==0 ? rpcEmGetStatus() : rpcEmGetEmptyStatus();`

Die sorgt dafür, dass nur noch jede 5. Anfrage vom MS-A2 eine Antwort mit den aktuellen Werten bekommt. Dazwischen schicke ich jetzt leere Responses. Hier ein Auszug vom Log:

```
25-01-19 14:02:53.446 DEBUG uni-meter.websocket.a76f868c-5413-42b4-b966-4cff53ce3003 - send => {"id":1656,"src":"shellypro3em-e45f01a038ab","dst":"280024270341","result":{"id":0}}
25-01-19 14:02:53.732 DEBUG uni-meter.websocket.a76f868c-5413-42b4-b966-4cff53ce3003 - send => {"id":1657,"src":"shellypro3em-e45f01a038ab","dst":"280024270341","result":{"id":0}}
25-01-19 14:02:54.225 DEBUG uni-meter.websocket.a76f868c-5413-42b4-b966-4cff53ce3003 - send => {"id":1658,"src":"shellypro3em-e45f01a038ab","dst":"280024270341","result":{"id":0}}
25-01-19 14:02:54.756 DEBUG uni-meter.websocket.a76f868c-5413-42b4-b966-4cff53ce3003 - send => {"id":1659,"src":"shellypro3em-e45f01a038ab","dst":"280024270341","result":{"id":0}}
25-01-19 14:02:55.261 DEBUG uni-meter.websocket.a76f868c-5413-42b4-b966-4cff53ce3003 - send => {"id":1660,"src":"shellypro3em-e45f01a038ab","dst":"280024270341","result":{"id":0,"a_current":-0.03,"a_voltage":230,"a_act_power":-6,"a_aprt_power":-6,"a_pf":1,"a_freq":50,"b_current":-0.03,"b_voltage":230,"b_act_power":-6,"b_aprt_power":-6,"b_pf":0.98,"b_freq":50,"c_current":-0.03,"c_voltage":230,"c_act_power":-6,"c_aprt_power":-6,"c_pf":0.98,"c_freq":50,"total_current":-0.08,"total_act_power":-18,"total_aprt_power":-18}}
```

Damit funktioniert es jetzt deutlich besser. Oszilliert zwar immer noch ein wenig, aber ich vemute, dass das eher durch die schwankende Aufnahme der laufenden Geräte (insbesondere PC) verursacht wird da er dadurch ja jetzt deutlich langsamer regelt.

Also meine Idee wäre immer noch Semaphore, aber etwas veränderte Logik:
- Semaphore wird beim erstellen direkt gesperrt
- Input Device gibt Semaphore frei, sobald neue Werte empfangen werden
- In der Methode createRpcResult wird versucht die Semaphore zu setzen. Bei Erfolg werden die aktuellen Werte übermittelt, bei Fehlschlag eine leere Response.
-
Wie erwähnt, bin kein Java-Programmierer, daher übersteigt das meine mickrigen Java-Kenntnisse.

P.S. Mein erster Ansatz mit `Thread.sleep(1000)` schlug gnadenlos fehlt, da der MS-A2 auch in der Zwischenzeit permanent neue Anforderungen gesendet hat.
P.P.S. Bei der Lösung wäre noch zu prüfen, was passiert, wenn längere Zeit keine Werte vom MQTT-Broker kommen, da der MS-A2 ja ständig Antworten bekommt....nur halt keine verwertbaren. Wird der uni-meter komplett gestoppt, merkt der MS-A2 das nach einer Weile und geht quasi in Ruhemodus (keine Einspeisung, kein laden).

sdeigm, 2025-01-19:
So, ich habe nun mal einen Debug-Output bei mir auf dem System erzeugt, wo der Hoymiles den Netzbezug schön auf 0 hält. Das Logfile anbei.

Das mit dem Drosseln der Abfragegeschwindigkeit des Hoymiles werde ich mir mal ansehen. Ich hätte eigentlich erwartet, dass der Hoymiles in gewissen Grenzen noch keine neue Anfrage sendet, bevor die alte beantwortet ist. Bleibt sie natürlich zu lange aus, dann schlägt mit Sicherheit ein Retry zu. Das werde ich mal debuggen.

Da das ganze aber nichts mehr mit MQTT zu tun hat werde ich dieses Ticket dann schließen wenn ich die Doku aktualisiert habe. Alles weitere sollten wir dann in dem neuen Feature Ticket diskutieren

[uni-meter.log](https://github.com/user-attachments/files/18469405/uni-meter.log)

sdeigm, 2025-01-19:
Ich nochmal. Ich habe mir mal die Eingabedaten angesehen, die per MQTT geliefert werden. Die ändern sich manchmal mehrere Sekunden lang nicht (grep auf "power phase 0") (Wert die mehrere Sekunden anstehen z.B 11.33, 198, -90):

`25-01-19 00:11:54.638 DEBUG uni-meter.output         - power phase 0: PowerData[power=-33.0, apparentPower=-33.0, powerFactor=1.0, current=-0.14347826086956522, voltage=230.0, frequency=50.0]
25-01-19 00:11:54.873 DEBUG uni-meter.output         - power phase 0: PowerData[power=100.33333333333333, apparentPower=100.33333333333333, powerFactor=1.0, current=0.43623188405797103, voltage=230.0, frequency=50.0]
25-01-19 00:11:55.853 DEBUG uni-meter.output         - power phase 0: PowerData[power=100.33333333333333, apparentPower=100.33333333333333, powerFactor=1.0, current=0.43623188405797103, voltage=230.0, frequency=50.0]
25-01-19 00:11:56.884 DEBUG uni-meter.output         - power phase 0: PowerData[power=135.66666666666666, apparentPower=135.66666666666666, powerFactor=1.0, current=0.5898550724637681, voltage=230.0, frequency=50.0]
25-01-19 00:11:57.856 DEBUG uni-meter.output         - power phase 0: PowerData[power=35.0, apparentPower=35.0, powerFactor=1.0, current=0.15217391304347824, voltage=230.0, frequency=50.0]
25-01-19 00:11:58.834 DEBUG uni-meter.output         - power phase 0: PowerData[power=38.0, apparentPower=38.0, powerFactor=1.0, current=0.16521739130434784, voltage=230.0, frequency=50.0]
25-01-19 00:11:59.863 DEBUG uni-meter.output         - power phase 0: PowerData[power=-98.0, apparentPower=-98.0, powerFactor=1.0, current=-0.4260869565217391, voltage=230.0, frequency=50.0]
25-01-19 00:12:00.875 DEBUG uni-meter.output         - power phase 0: PowerData[power=-97.33333333333333, apparentPower=-97.33333333333333, powerFactor=1.0, current=-0.42318840579710143, voltage=230.0, frequency=50.0]
25-01-19 00:12:01.848 DEBUG uni-meter.output         - power phase 0: PowerData[power=-117.0, apparentPower=-117.0, powerFactor=1.0, current=-0.5086956521739131, voltage=230.0, frequency=50.0]
25-01-19 00:12:02.883 DEBUG uni-meter.output         - power phase 0: PowerData[power=-0.6666666666666666, apparentPower=-0.6666666666666666, powerFactor=1.0, current=-0.002898550724637681, voltage=230.0, frequency=50.0]
25-01-19 00:12:03.856 DEBUG uni-meter.output         - power phase 0: PowerData[power=11.333333333333334, apparentPower=11.333333333333334, powerFactor=1.0, current=0.04927536231884058, voltage=230.0, frequency=50.0]
25-01-19 00:12:04.639 DEBUG uni-meter.output         - power phase 0: PowerData[power=11.333333333333334, apparentPower=11.333333333333334, powerFactor=1.0, current=0.04927536231884058, voltage=230.0, frequency=50.0]
25-01-19 00:12:04.641 DEBUG uni-meter.output         - power phase 0: PowerData[power=11.333333333333334, apparentPower=11.333333333333334, powerFactor=1.0, current=0.04927536231884058, voltage=230.0, frequency=50.0]
25-01-19 00:12:04.643 DEBUG uni-meter.output         - power phase 0: PowerData[power=11.333333333333334, apparentPower=11.333333333333334, powerFactor=1.0, current=0.04927536231884058, voltage=230.0, frequency=50.0]
25-01-19 00:12:04.874 DEBUG uni-meter.output         - power phase 0: PowerData[power=87.0, apparentPower=87.0, powerFactor=1.0, current=0.37826086956521743, voltage=230.0, frequency=50.0]
25-01-19 00:12:05.856 DEBUG uni-meter.output         - power phase 0: PowerData[power=113.0, apparentPower=113.0, powerFactor=1.0, current=0.49130434782608695, voltage=230.0, frequency=50.0]
25-01-19 00:12:06.862 DEBUG uni-meter.output         - power phase 0: PowerData[power=105.66666666666667, apparentPower=105.66666666666667, powerFactor=1.0, current=0.4594202898550725, voltage=230.0, frequency=50.0]
25-01-19 00:12:07.878 DEBUG uni-meter.output         - power phase 0: PowerData[power=16.333333333333332, apparentPower=16.333333333333332, powerFactor=1.0, current=0.07101449275362319, voltage=230.0, frequency=50.0]
25-01-19 00:12:08.858 DEBUG uni-meter.output         - power phase 0: PowerData[power=-97.66666666666667, apparentPower=-97.66666666666667, powerFactor=1.0, current=-0.4246376811594203, voltage=230.0, frequency=50.0]
25-01-19 00:12:09.835 DEBUG uni-meter.output         - power phase 0: PowerData[power=-91.33333333333333, apparentPower=-91.33333333333333, powerFactor=1.0, current=-0.39710144927536234, voltage=230.0, frequency=50.0]
25-01-19 00:12:10.854 DEBUG uni-meter.output         - power phase 0: PowerData[power=-108.0, apparentPower=-108.0, powerFactor=1.0, current=-0.46956521739130436, voltage=230.0, frequency=50.0]
25-01-19 00:12:11.869 DEBUG uni-meter.output         - power phase 0: PowerData[power=-10.666666666666666, apparentPower=-10.666666666666666, powerFactor=1.0, current=-0.0463768115942029, voltage=230.0, frequency=50.0]
25-01-19 00:12:12.847 DEBUG uni-meter.output         - power phase 0: PowerData[power=85.33333333333333, apparentPower=85.33333333333333, powerFactor=1.0, current=0.3710144927536232, voltage=230.0, frequency=50.0]
25-01-19 00:12:13.862 DEBUG uni-meter.output         - power phase 0: PowerData[power=198.0, apparentPower=198.0, powerFactor=1.0, current=0.8608695652173913, voltage=230.0, frequency=50.0]
25-01-19 00:12:14.644 DEBUG uni-meter.output         - power phase 0: PowerData[power=198.0, apparentPower=198.0, powerFactor=1.0, current=0.8608695652173913, voltage=230.0, frequency=50.0]
[... cut ...]
25-01-19 00:12:42.844 DEBUG uni-meter.output         - power phase 0: PowerData[power=-80.33333333333333, apparentPower=-80.33333333333333, powerFactor=1.0, current=-0.34927536231884054, voltage=230.0, frequency=50.0]
`

Kann es sein, dass dein Lesekopf zwar jede Sekunde in das MQTT Topic schreibt aber die Werte nicht jede Sekunde aktualisiert? Habe auf die schnelle mal im Internet gesucht und z.B. folgendes gefunden 

https://community.simon42.com/t/stromzaehler-mit-smr-ir-lesekopf-jede-sekunde-auslesen/36232

Kann das nur eine Einstellung auf deinem Lesekopf sein die falsch ist?

StPinkert, 2025-01-19:
Feature Request schließen:  Stimmt. MQTT läuft, von daher Feature Request erfüllt.

Einstellungen am Lesekopf: Was hier im Log auftaucht kann etwas verwirrend sein. Aber bei genauen hinschauen sind es nie mehr als 2 Sekunden mit dem gleichen Wert. Tasmota (das auf dem Hichi-Zähler läuft) hat eine kleine Eigenart, wodurch der gleiche Wert 2x gesendet wird. Zum einen ist es bei mir so eingestellt, dass der Power-Wert sofort gesendet wird, wie im verlinkten Thread vom User "Nein_danke" erwähnt. Als zweites gibt es die "teleperiod". In dem Intervall werden gesammelt alle verfügbaren Werte gesendet. Die steht bei mir auf dem minimal möglichen Wert (10 Sekunden). D.h. Power wird sofort gesendet, wenn er vom Zähler per IR übertragen wird(1x pro Sekunde) und unabhängig davon werden alle Werte alle 10 Sek. übertragen. Bei den angegebenen Beispielen war immer eine Nachricht der "teleperiod" mit dabei - erkennt man daran, dass die nahezu zeitgleich oder nur ein paar Zehntel Sekunden nach einem vorherigen Wert kommt. Da in der Zwischenzeit aber kein neuer Wert vorliegt, entspricht der immer dem zuletzt gesendeten.
Dieses Verhalten hat mich einige Nerven in NodeRed gekostet, bis ich kapiert habe, dass pro Minute nicht 60 sondern 66 Werte übertragen werden, wovon ich 6 verwerfen kann, weil sie redundant sind. 

P.S. Mit der von mir vorgenommenen Änderungen lief die Regelung heute recht gut. Sowohl beim Laden, als auch beim Entladen, als die Solarenergie nicht mehr ausgereicht hat. Mittlerweile ist der Akku wier leer. (Hab nur 2 Panele mit 600W Wechselrichter, da bleibt für den Akku im Winter nicht viel übrig). Hoymiles ist derzeit in der Entwicklung der Firmware ja noch sehr aktiv. Immer wieder neue Features. Ich vermute, dass auch an bereits vorhandenen Features noch fleissig hinter den Kulissen gewerkelt wird. Vielleicht liegt das unterschiedliche Verhalten bei der Regelung an verschiedenen Firmware-Versionen?

Bei mir ist folgende Fimware installiert: Kommunikationsmodul: V01.03.06, Wechselrichter: V01.03.04, BMS: V01.01.13.00

closed 2025-01-20 by sdeigm