sdeigm/uni-meter #12: Throttling of input device polling frequency

sdeigm, 2025-01-19:


StPinkert, 2025-01-19:
Maybe i can try to get my quick&dirty solution nice&clean. 
Would you like to introduce a new configuration setting to enable or disable this feature and/or control the behaviour?

sdeigm, 2025-01-20:
I have now introduced a throttling for the EM.GetStatus request received via websocket. You can enforce a minium sampling period by adding the following section to your uni-meter.conf file. Adjust the min-sample-period to your needs:

```
uni-meter {
  #...
  output-devices {
    shelly-pro3em {
      min-sample-period = 5000ms
    }
  }
  #...
}
```

The Hoymiles storage does not backpressure, when it hasn't received an answer for a pending RPC request. Instead the storage fires a new request every 500ms. Only the last request withing the minimum sampling period will be answered by the uni-meter. All other requests will be basically dropped.

So tracing the above sampling period of 5 seconds leads to the following behavior:

`sudo /opt/uni-meter/bin/uni-meter.sh | grep -E "recv|total_aprt_power"`

```
25-01-20 08:34:11.280 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48099,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:11.888 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48100,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:12.472 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48101,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:12.719 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48102,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:13.129 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48103,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:13.618 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48104,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:14.198 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48105,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:14.739 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48106,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:15.363 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48107,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:15.401 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48108,"src":"280024310133","method":"EMData.GetStatus","params":{"id":0}}
25-01-20 08:34:15.856 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48109,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:16.087 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - send => {"id":48109,"src":"shellypro3em-b827eb364242","dst":"280024310133","result":{"id":0,"a_current":0.4,"a_voltage":230,"a_act_power":93,"a_aprt_power":93,"a_pf":1,"a_freq":50,"b_current":0.4,"b_voltage":230,"b_act_power":93,"b_aprt_power":93,"b_pf":0.98,"b_freq":50,"c_current":0.4,"c_voltage":230,"c_act_power":93,"c_aprt_power":93,"c_pf":0.98,"c_freq":50,"total_current":1.2,"total_act_power":279,"total_aprt_power":279}}
25-01-20 08:34:16.112 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48110,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:16.749 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48111,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:17.429 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48112,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:17.983 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48113,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:18.538 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48114,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:18.899 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48115,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:19.622 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48116,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:20.092 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48117,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:20.527 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48118,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:20.567 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48119,"src":"280024310133","method":"EMData.GetStatus","params":{"id":0}}
25-01-20 08:34:21.069 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - recv <= {"id":48120,"src":"280024310133","method":"EM.GetStatus","params":{"id":0}}
25-01-20 08:34:21.087 DEBUG uni-meter.websocket.e0eadcb8-52d8-431b-9b49-faa34475035c - send => {"id":48120,"src":"shellypro3em-b827eb364242","dst":"280024310133","result":{"id":0,"a_current":0.4,"a_voltage":230,"a_act_power":93,"a_aprt_power":93,"a_pf":1,"a_freq":50,"b_current":0.4,"b_voltage":230,"b_act_power":93,"b_aprt_power":93,"b_pf":0.98,"b_freq":50,"c_current":0.4,"c_voltage":230,"c_act_power":93,"c_aprt_power":93,"c_pf":0.98,"c_freq":50,"total_current":1.2,"total_act_power":279,"total_aprt_power":279}}
2
```

I already pushed that change in the branch to this feature request. Can you please test if it fits your needs?

StPinkert, 2025-01-21:
OK, i'll try to test it out as soon as possible (probably next weekend). Let's hope for some sun.

sdeigm, 2025-01-22:
I keep my fingers crossed for the sun!

StPinkert, 2025-01-25:
Well, there was only a short period with sun today - not enough to load the battery substantially, but i charged it a bit using grid power to test it out. 

Short answer: It works as expected.

Long answer: I tested several settings by increasing the value step by step. Started with 500ms and increased it by 500ms up until 2500ms. The lower the throttle value the higher the MS-A2 overshoot (at 500ms +/- 300W around zero). With 2000ms and higher i get acceptable values (+/-30W around zero). So i ended up at 2500ms that seem to me a good balance between a smooth but yet fast enough regulation.

If you're interested i attached a log file. With every restart the throttle value has been increased. (500ms, 1000ms, 1500ms, 2000ms and finally 2500ms)

[uni-meter-throttle.log](https://github.com/user-attachments/files/18547238/uni-meter-throttle.log)

StPinkert, 2025-01-25:
Graphical view with some annotations

[image]

sdeigm, 2025-01-26:
Hi, thanks for your extensive testing and the nice chart showing your findings!!

Good to hear, that it is possible to use MQTT as input using these settings. Interessting that the behavior using MQTT is so different compared to directly contacting the IR read head. The small additional latency seems to have a big impact.

I will close this issue and update the Readme regarding the Throttling.

closed 2025-01-26 by sdeigm

peerfiet, 2026-04-14:
Just for other people passing by here as I was also fighting with this for quite some time (uni-meter in combination with Marstek Venus E 3.0). A different solution seems to be to set the "scale" parameter of all phases in uni-meter to 0.5 and disable throttling. This slows down the process of getting back to "0" a bit but keeps the system very stable (+10/-10 Watt) around "0" especially on smaller changes or in quasi-steady state (change +50/-50 Watt). Here is an example of a power change of -1000 Watt and getting back to zero in around 20-30 seconds.  

[image]

Current figures a suggesting that - also it does not look like this at a first glance looking at above graph - this gives better results with respect to minimizing daily grid feed-in (my guess at the moment is that the system is most of the time in a quasi steady-state and the control loop is working very good in that case with these parameters).

If you want the control loop to be more aggressive choose numbers above 0.5 for scale (and <= 1.0  of course ;-)), but this will bring oscillation back to some extent.

coltc50, 2026-08-30:
I'm just here to say thank you for the pro tip with marstek venus e 3 @peerfiet