sdeigm/uni-meter #16: Getting it to run on ESP32.

martind40, 2025-01-23:
Is it planned to run the uni-meter on an ESP32? The Raspberry is like "casting pearls before swine", isn't it?

sdeigm, 2025-01-23:
Hi,  that would be more or less an issue of running a java application on an ESP32. A quick internet search brought up this:

https://esp32.com/viewtopic.php?t=41884 and using https://github.com/FlintVN/FlintESPJVM

I don't know if that would work with a relative modern Java 17 application. So perhaps you can give it a try?

For this project I used Java and reactive programming techniques since that's the stuff I am using every day and I am familar with. A lot of highlevel 3rd Party libraries are used: Pekko, Websockets, MQTT. So porting it to another language like for instance C++ would be a complete rewrite.

Anyhow the core is realtive simple: Hoymiles queries the Shelly through its websocket interface.

TheRealMoeder, 2025-01-31:
This seems to be a perfect task for an esp32 to me as well.

@martind40 For my use case an esp32 running mDNS, websocket and handling mqtt input should be enough. With some AI help this should be a quick project. I'll try setting something up, as my programming skills are limited, help is aways welcome.

You will find my work here https://github.com/TheRealMoeder/MQTT2Shelly_ESP32[https://github.com/TheRealMoeder/MQTT2Shelly_ESP32](https://github.com/TheRealMoeder/MQTT2Shelly_ESP32)

TheRealMoeder, 2025-01-31:
@sdeigm can you elaborate on the minimum requirements the MS-A2 expects from the shelly emulator? Is ist just the HTTP request found in ShellyPro3EM.java or do we also need the V1 API or anything else I overlooked from a quick glance through your code?

sdeigm, 2025-02-01:
@TheRealMoeder its even less then the stuff I implemented. Unfortunately I don't remember it exactly, but initially, when you connect the Hoymiles storage to Shelly, the Storage issues a GetDeviceInfo request to check the device. Afterward, only EM.GetStatus and EMData.GetStatus requests are send via Websocket JSON RPC. 

So in fact, I don't know any more if there are any HTTP requests used at all or if the entire communication is done via Websocket.

You can easily trace that if you use my software using a DEBUG output for "uni-meter.output" and connect it to your storage. Since yesterday there is a docker container available and a separate project uni-meter-docker here on github, containing docker-compose instructions. So you can test it without installing at all.

TheRealMoeder, 2025-02-03:
Thank you for the info, this should be enough to get started.

TheRealMoeder, 2025-02-10:
Just a small update to whom it may concern:

Esp8266 is now running WiFi Manager for easy setup, announces itself to the Hoymiles MS-A2 via mDNS, reads incoming MQTT power in Tasmota Json style and can handle the rpc queries via websocket.
I still have to do some more testing and code cleanup but will try to publish it on GitHub within a few days. Beware it will be a rudimentary implementation and far from sdeigms excellent solution.
I will try to implement SMA Multicast Energy data as a second step, as that's what I personally need.

sdeigm, 2025-02-11:
Great to hear! In which progamming language is this done? C/C++?

TheRealMoeder, 2025-02-11:
Yes, with huge amounts of copy and paste from the library examples. I'm sure the code will be really bad from a programmers point of view, but as long as it works...;)

TheRealMoeder, 2025-02-12:
Running into some trouble unfortunately. mDNS broadcast works fine, the storage queries Shelly.GetDeviceConfig via http and upgrades to websocket connection, but from there on nothing's seems to happen... Some more debugging needed...

sdeigm, 2025-02-12:
I was fighting with similar problems. While I am not 100% sure, I supspected that Hoymiles does not use a real JSON Parser, where the order of the properties can be arbitrary. I suspected that the order of properties matters for them. Therefore I added these @JsonPropertyOrder Annotations. 

 @JsonInclude(JsonInclude.Include.NON_NULL)
  @JsonIgnoreProperties(ignoreUnknown = true)
  @JsonPropertyOrder({"name", "id", "mac", "slot", "model", "gen", "fw_id", "ver", "app", "auth_en", "auth_domain", "profile"})
  public record GetDeviceInfoResponse(
        @JsonProperty("name") String name,
        @JsonProperty("id") String id,
        @JsonProperty("mac") String mac,
        @JsonProperty("slot") int slot,
        @JsonProperty("model") String model,
        @JsonProperty("gen") long gen,
        @JsonProperty("fw_id") String fw_id,
        @JsonProperty("ver") String ver,
        @JsonProperty("app") String app,
        @JsonProperty("auth_en") boolean auth_en,
        @JsonProperty("auth_domain") String auth_domain,
        @JsonProperty("profile") String profile
  ) implements Response {
    @Override
    public String toString() {
      try {
        return objectMapper.writeValueAsString(this);
      } catch (JsonProcessingException e) {
        throw new RuntimeException(e);
      }
    }
  }

Perhaps you should give this a try ...

TheRealMoeder, 2025-02-12:
A manual websocket connection gives exactly the same results as uni-meter, so I don't think this will be the problem. I think the Esp8266 websocket (or my program) can't handle the fast queries the hoymiles fires at it. Tcpdump on the uni-meter machine shows extremely frequent websocket queries.

TheRealMoeder, 2025-02-14:
Well it finally works. Please find the preliminary version for ESP8266 here: https://github.com/TheRealMoeder/Energy2Shelly_ESP

martind40, 2025-02-15:
Great to read..... try to test it next weekend.

TheRealMoeder, 2025-02-15:
Great, I'll improve some more in the coming days and also test build for esp32, although I don't have one for testing in the moment. Maybe we should move discussions to my repository as well and @sdeigm could provide a link in his readme.

TheRealMoeder, 2025-02-17:
added preliminary support for SMA energy meters

TheRealMoeder, 2025-02-23:
@sdeigm ESP is up and running, should we close this issue ? Again, maybe you could link my repository in your readme for anyone interested.

sdeigm, 2025-02-23:
Ok, when no further discussions around porting the stuff are necessary, we can close this.

I added a link to your repository in the new preliminary remarks section of the documentation.

closed 2025-02-23 by sdeigm