sdeigm/uni-meter #22: Making Tasmota configurable

StPinkert, 2025-01-26:
At the moment you assume, that the values are named "currW","totalKwh" and "exportTotalKwh". That's not always the case and depends on the sml-script that is used inside tasmota. (See [https://tasmota.github.io/docs/Smart-Meter-Interface/#abb-b23-modbus](url)

{"StatusSNS":{"Time":"2025-01-26T13:07:25","Main":{"power":504,"counter_pos":1360.768,"counter_neg":149.851}}}

sdeigm, 2025-01-26:
Ok, good point, didn't know that since I do not own such a device. 

I will make that configurable using JSON path. Is there I kind of standard, that is mostly used? Just asking that to provide this as default.

closed 2025-01-26 by sdeigm

StPinkert, 2025-01-26:
You can have a look at the linked tasmota sml documentation. There are a lot of ready to use scripts for dozens of different power meter. From what i've seen so far there is not "standard name scheme" yet.

StPinkert, 2025-02-06:
Sorry for the late answer. The Tasmota-Input is working as intended. 

Regarding my oscillation problem - it makes no difference, if i use tasmota or mqtt. The attached analysis was created using the tasmota input device. It looks very similar to my previous tests using MQTT input device with the same throttle settings.

[image]

P.S. Apart from sporadic errors (mostly "failure while processing websocket notification: com.fasterxml.jackson.core.io.JsonEOFException: Unexpected end-of-input in field name") it's running flawless since two weeks or so.