sdeigm/uni-meter #31: Add Shelly.GetStatus

TheRealMoeder, 2025-02-03:
While I'm still working on the esp32 solution, I also tried using uni meter as a Shelly device in Openhab 4. Unfortunately, uni meter throws an error when being queried Shelly.GetStatus. Do you plan on implementing more RPC services for full compatibility?

sdeigm, 2025-02-05:
I guess I will never reach full compatibility since there are a lot of methods to query and configure the device as well as thinks like webhooks. But I will add enhancements to reach compatibility with certain devices or software tools. So if you need Shelly.GetStatus I will add it :-)

Since I do not have a real shelly I can only implement that functionality based on the information I will find in the internet.

TheRealMoeder, 2025-02-05:
API Docs für m the Shelly seem pretty complete, so I don't see why it couldn't work. I currently run a python script reading a sma energy meter so I can't concurrently use uni-meter on the same IP, so I would instead use Uni-meter to the data as a Shelly device into Openhab. The other and probably more universal option would be to add an additional MQTT output option to uni-meter.

sdeigm, 2025-02-05:
I don't say it will not work, I am just saying it takes a lot of diligence to reach and complain full compatibility. So I am adding the parts that are needed on demand. Now I am adding Shelly.GetStatus and then we will see what's needed next instead of taking the specifiction and adding function by function.

If you can send me some information (or point me to the right website) how the Shelly writes its data into mqtt topic, I can also add this.

TheRealMoeder, 2025-02-05:
Great, I will help with testing of course.

Concerning MQTT, I didn't mean Uni-meter should necessarily copy Shelly output but rather offer a config option to add an Mqtt output server and topic to which it could publish energy data.

sdeigm, 2025-02-05:
Let's discuss the MQTT stuff in a new issue, so that I can close this one.

closed 2025-02-05 by sdeigm