sdeigm/uni-meter #32: MQTT output

sdeigm, 2025-02-05:


TheRealMoeder, 2025-02-05:
I suggest an additional configurable MQTT output using the same syntax as for the input configurations.

TheRealMoeder, 2025-02-05:
I think having multiple output devices in one process would be the nicer solution. Taking my case I think two processes couldn't register to the same energy meter broadcast on the same interface.

RichiCaneloni, 2025-06-19:
Hallo, ich bekomme das Uni-meter in HA nicht zum Laufen. Ich möchte den Shelley 3M pro emulieren. Am Zähler sitzt ein IR-Lesekopf Tasmota. Dieser funktioniert über MQTT wunderbar in HA. Nun will ich die Hoymiles MS-A2 damit regeln. Uni-meter addon ist in HA installiert. Die unimeter.conf ist im addonverzeichnis. Zusätzlich ist auch pyscript installiert. Im Protokoll von uni-meter kommt der Fehler **_25-06-19 16:53:21.360 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-06-19 16:53:21.498 ERROR uni-meter.input          - MQTT stream failed: Not authorized to connect_**. Was mach ich denn falsch?