sdeigm/uni-meter #34: Adding Shelly device to Openhab throws errors

TheRealMoeder, 2025-02-05:
Some more debugging seems necessary...

Unimeter Error:
`INFO  org.apache.pekko.actor.DeadLetterActorRef - Message [com.deigmueller.uni_meter.application.WebsocketInput$Ack] to Actor[pekko://uni-meter/deadLetters] was not delivered. [1] dead letters encountered. If this is not an expected behavior then Actor[pekko://uni-meter/deadLetters] may have terminated unexpectedly. This logging can be turned off or adjusted with configuration settings 'pekko.log-dead-letters' and 'pekko.log-dead-letters-during-shutdown'.`

Openhab Error:
2025-02-05 19:25:15.609 [DEBUG] [shelly.internal.ShellyHandlerFactory] - Shelly Pro 3EM: Create new thing of type shelly:shellypro3em using ShellyRelayHandler

2025-02-05 19:25:15.643 [DEBUG] [shelly.internal.ShellyHandlerFactory] - Thing handler for uid shelly:shellypro3em:3d3c76e9fb added, total things = 1

2025-02-05 19:25:17.650 [DEBUG] [y.internal.handler.ShellyBaseHandler] - shellypro3em-3d3c76e9fb: Using default userId admin from binding config

2025-02-05 19:25:17.650 [DEBUG] [y.internal.handler.ShellyBaseHandler] - shellypro3em-3d3c76e9fb: Using default password from bindingConfig (userId=admin)

2025-02-05 19:25:17.650 [DEBUG] [y.internal.handler.ShellyBaseHandler] - shellypro3em-3d3c76e9fb: Config: Device address=192.168.17.1, HTTP user/password=admin/***, update interval=60
Events: Button: false, Switch (on/off): false, Push: false, Roller: trueSensor: true, CoIoT: true
Blu Gateway=false, Range Extender: true

2025-02-05 19:25:17.651 [DEBUG] [y.internal.handler.ShellyBaseHandler] - shellypro3em-3d3c76e9fb: Start initializing for thing Shelly Pro 3EM, type shellypro3em, Device address 192.168.17.1, Gen2: true, isBlu: false, alwaysOn: true, hasBattery: false, CoIoT: true

2025-02-05 19:25:18.739 [DEBUG] [g.shelly.internal.api2.Shelly2ApiRpc] - shellypro3em-3d3c76e9fb: Connect Rpc Socket (discovery = false)

2025-02-05 19:25:18.739 [DEBUG] [helly.internal.api2.Shelly2RpcSocket] - Shelly Pro 3EM: Connect WebSocket, URI=ws://192.168.17.1/rpc

2025-02-05 19:25:18.903 [DEBUG] [y.internal.handler.ShellyBaseHandler] - shellypro3em-3d3c76e9fb: Unexpected API result: 500/Internal Server Error`

TheRealMoeder, 2025-02-05:
I changed my data flow in the meantime, so I have no urgent need for this. But I can still assist in testing if your interested in completing the feature.

harvey637, 2025-02-20:
Hi,
getting things up a little ... jon output of the emulator is working and reflects the input data - thank you very much!

But using the "original" hoymiles ms-a2 to fetch the data from the emulator
- finds the emulator using (add) (add shelly device found)
- but then the emulator outputs: exception in HTTP server: java.lang.IllegalArgumentException: unhandled RPC method 'EM.GetConfig'
- and the hoymiles ms-a2 says "data could not be received" (latest firmware from 25.01.2025)

 So in my opinion some dummy "EM.GetConfig" should return some dummy configuration to make the ms-a2 happy
thank you and
cu
Harvey

sdeigm, 2025-02-20:
Hi, thanks for reporting this. 

This must be new in the latest firmware version since I (and others)  already sucessfully added the Shelly to the Hoymiles storage using an older firmware version.

So, I would now rate this as a bug that I have to fix soon. Hopefully I find some time in the evening to fix this.

Stefan

TheRealMoeder, 2025-02-20:
EM.GetConfig is pretty simple according to the API:

```
{
  "id": 0,
  "name": null,
  "blink_mode_selector": "active_energy",
  "phase_selector": "a",
  "monitor_phase_sequence": true,
  "reverse": {
    "b": true
  },
  "ct_type": "120A"
}
```

sdeigm, 2025-02-21:
@harvey637 and @TheRealMoeder, yesterday I was a little bit confused about which RPC method is missing, but I finally sorted it out with the help of @TheRealMoeder :-)

So this ticket has its old title again, the reported missing EM.GetConfig method was fixed in another [issue](https://github.com/sdeigm/uni-meter/issues/52) and a new v1.0.9 version is released.

@harvey637 can you please test if you can now add the Shelly to the hoymiles storage and give feedback in the other ticket?

harvey637, 2025-02-21:
Hallo Stefan,
I tested adding the shelly emulator 1.0.9 to my hoymiles ms-a2 version V01.04.01 (the one with support added for ecotracker 2025-01-24) : 
and **SUCCESS**!

[image]

I see the same info as in the emulator, everything looks good. Now I have to wait for a situation with more sunshine than consumption ....

By the way, there might be a typo somewhere with tri-phase, as ONLY the "default-power-l1" was use in ALL phases, as I removed temporarily the generic-http input, so the defaults kicked in after forget-interval. "default-l2" and "default-l3" was not used?!? 

Also the implementation of the shelly-tri-phase in the hoymiles firmware is not yet complete, as I cannot select, on which pase THIS output is used, but this is problem og hoymiles ....

Thank you for your great help! (also the documentation and samples)
cuz
Harvey

harvey637, 2025-02-21:
As things work I will now test with tri-phase and ONLY read the correct phase from my smartmeter (I have all three phases plus total in my smartmeter).
Because I think that I only want to use the battery (which is single phase) to regulate exactly THIS phase, not the two other phases.
So with the battery feeding into phase L1 I want to read real phase L1 and use this reading for "power-l1" and ZERO the the two other phases.
May be I have to use dummy json to simulate "0" for this usage.
Also there might be the idea, to use ONE shelly emulator with THREE Outputs for the three phases to make THREE batteries each on a single phase happy. But this is far future! 
thank you!
Harvey

sdeigm, 2025-03-20:
Implemented in version 1.1.3

closed 2025-03-20 by sdeigm