sdeigm/uni-meter #37: (Tasmota) Default Wert

twitnic, 2025-02-08:
Hey,

ich hab ein Tasmota IR-Lesekopf angebunden.
Da ich aber keine direkte Stromquelle im Zählerschrank habe, hab ich den ESP mit Tasmota an einer Powerbank. Die kann u.a. nach einer Dauer X leer sein.

Wäre es ggf möglich, für den Fall, dass via Tasmota nichts mehr kommt, man einen Default-Wert einstellen kann?

Vielen Dank
Dominic

sdeigm, 2025-02-09:
Hallo Dominic, 

das sollte kein Problem sein. Heute fällt das uni-meter wenn es eine Minute keinen Werte mehr erhält auf den Default-Wert 0 Watt zurück. Das werde ich konfigurierbar machen.

Den Fall, daß das Input-Device sozusagen ab und zu sozusagen "geplant" ausfällt wie wenn bei dir die Powerbank leer ist, hatte ich so noch nicht bedacht. Dann wird ja auch das Log ziemlich mit Fehlermeldungen zugemüllt. Das müsste ich in dem Zusammenhang dann irgendwie auch noch mal verbessern.

Gruß
Stefan

closed 2025-02-09 by sdeigm

sdeigm, 2025-02-09:
In der Version 1.0.7 kann man das nun folgendermassen konfigurieren:

```
uni-meter {
  output-devices {
    shelly-pro3em {
      forget-interval = 1m

      default-power-total = 0

      default-power-l1 = 0
      default-power-l2 = 0
      default-power-l3 = 0
    }
  }
}
```