sdeigm/uni-meter #38: Not getting values from MQTT to the virtual Shelly

CountZ74, 2025-02-08:
Hi, I am publishing the power value from Home Assistant. MQTT Explorer on Windows and mosquitto-sub on the RPi are able to subscribe and see the values. Uni-meter output says MQTT stream connected, but all the Shelly values are 0 or the dummy values. I guess it is not a problem with uni-meter, but rather my lack of ability to get the configuration right.

TheRealMoeder, 2025-02-08:
Show us your configuration file

CountZ74, 2025-02-09:
Here it is:

`uni-meter {
       
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.mqtt"
  
  output-devices {
    shelly-pro3em {
      udp-port = 1010
      udp-interface = "0.0.0.0"
      device {
        mac = "A827EB364242"
        hostname = "shellypro3em-a827eb364242"
        
        
      }
    }
  }

  input-devices {
    mqtt {
      url = "tcp://192.168.xxx.xxx:1883"
      username = "user"
      password = "password"

      power-phase-mode = "mono-phase"
#      energy-phase-mode = "mono-phase"

      channels = [{
        type = "value"
       	topic = "homeassistant/sensor/power-total"
        channel = "power-total"
        #json-path = "$..power"
        scale = 1.0 # default, can be omitted
      }]
#,{
#        type = "value"
#        topic = "homeassistant/sensor"
#        channel = "gesamtenergie"
        #json-path = "$..counter_pos"
#      }]
#,{
#        type = "json"
#        topic = "tele/smlreader/SENSOR"
#        channel = "energy-production-total"
#        json-path = "$..counter_neg"
#      }]
    }
  }
}`

sdeigm, 2025-02-09:
Hm, that looks ok if your data is stored in the topic homeassistant/sensor/power-total as direct value (not using JSON).

Unfortunately I don't have any debug output in the code that shows the data read from MQTT. I will have to improve that.

But using the attached logback.xml file it should be possible to check if there is some data ariving in uni-meter at all. Can you please use the attache logback.xml file and and then look for entries 

```Mqtt.onTopicData()```

in the log file? That would show that the uni-meter gets notifications about changed topic data. Would be good if you can send me that debug output

[logback.zip](https://github.com/user-attachments/files/18723963/logback.zip)

CountZ74, 2025-02-09:
I am running uni-meter manually right now, until I get it working. Here is a file containing the trace output:

[log.txt](https://github.com/user-attachments/files/18724023/log.txt)

sdeigm, 2025-02-09:
That also looks good, the topic data is delivered to the uni-meter. As said before unfortunately I do not trace the actual topic content.

Can you tell me the content value of the topic homeassistant/sensor/power-total?

CountZ74, 2025-02-09:
This is a MQtt Explorer screenshot:
[image]

sdeigm, 2025-02-09:
Ok, die denke die " um den Wert sind das Problem. Damit kommt der "value" channel nicht klar, er erwartet eine Zahl.

Kannst du das mal als json Channel konfigurieren und als Json-Path nur das $ Zeichen verwenden?

CountZ74, 2025-02-09:
Ok, ich schau mal was in Homeassistant möglich ist...

CountZ74, 2025-02-09:
Oh Mann, manchmal ist es schon schwer. Ich hab die Anführungszeichen mit in das Payloadfeld von Homeassistant kopiert...
Anführungszeichen gelöscht und schon funktioniert es.
Danke für deine Hilfe!

sdeigm, 2025-02-10:
Switching back to english :-)

 Anyhow if the problem is solved, I improved two things:

- In the future ' and " characters are stripped from the strings so that such a value would be parsed successfully
- Improved logging so that the debug logs contain information about the data read from MQTT

closed 2025-02-10 by sdeigm