sdeigm/uni-meter #45: Getting error: unhandled HTTP path: cm

markus-hi, 2025-02-15:
Hi,

i'm trying to setup a Shelly Pro3EM emulation. This is my actually config:

`uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.tasmota"

  input-devices {
      generic-http {
      url = "http://10.10.1.72/cm?cmnd=status%2010"
      #username = "username"
      #password = "password"

      power-phase-mode = "tri-phase"
      energy-phase-mode = "mono-phase"

      channels = [{
        type = "json"
        channel = "energy-consumption-total"
        json-path = ".StatusSNS.LK13BE.Power_curr"
        scale = 1
      },{
        type = "json"
        channel = "energy-production-total"
        json-path = ".StatusSNS.LK13BE.Power_total_in"
        scale = 1
      },{
        type = "json"
        channel = "power-l1"
        json-path = ".StatusSNS.LK13BE.Power_L1_curr"
      },{
        type = "json"
        channel = "power-l2"
        json-path = ".StatusSNS.LK13BE.Power_L2_curr"
      },{
        type = "json"
        channel = "power-l3"
        json-path = ".StatusSNS.LK13BE.Power_L3_curr"
      }]
    }
}
output-devices {
    shelly-pro3em {
      device {
        mac = "A827BC364242"
        hostname = "shellypro3em-a827bc364242"
      }
      udp-port = 1010
      udp-interface = "0.0.0.0"
    }
  }
}`

I'm getting this error at the command line:

`25-02-15 21:12:39.066 DEBUG uni-meter.output.http    - unhandled HTTP path: cm
25-02-15 21:12:39.076 DEBUG uni-meter.input          - status response: The requested resource could not be found but may be available again in the future.`

No values were shown at http://10.10.1.72/status. A request on the command line via curl is successfully:

root@srv-unimeter:~# curl -s "http://10.10.1.72/cm?cmnd=status%2010" | jq .StatusSNS.LK13BE.Power_curr
271

I'm using Version 1.0.8, do you have an Idea whats wrong there?

Markus

sdeigm, 2025-02-16:
You trying to use the configuration of the generic-http input device, but you still have configured the system to use the tasmota input device. Please change the 4th line in your configuration to:

``input = "uni-meter.input-devices.generic-http"``

then it should work as expected

markus-hi, 2025-02-16:
Thank you very much! That solved the problem🙏

closed 2025-02-16 by markus-hi