sdeigm/uni-meter #46: Option for minimum values

tmtudy, 2025-02-15:
As the inverter for batteries is quite inefficient for low power values, could you help with setting a threshold under which a 0 is returned instead of the read value?
Example: if power is between +150 and -200 (random numbers) return 0. 
Thanks

tmtudy, 2025-02-19:
Well, i actually look at this in a more simplistic way, something like "threshold value for sending real consumption". I am not saying there should always be a (example) 200w delta, but that when actual power is below a threshold (let.s also say 200w) the battery should not intervene. Example: actual value: 120w, value seen by battery/returned by the emulator: 0w; actual value 220w, value seen by battery: 220w This would make for a worse zero on meter result, but a better overall efficiency. Regarding efficiency at low power here is a forum link with some tests from other users explaining my reasoning (in dutch but numbers are easy to follow i hope): https://gathering.tweakers.net/forum/list_messages/2262054?data%5Bfilter_keywords%5D=Charge%20100

sdeigm, 2025-02-20:
First of all, thanks for the link with the good information about the efficiency in different charging situations!

I think I understand your simplistic way and I was not suggesting something different. I just wanted to make clear that in my opinion it does not solve the problem you want to solve. Perhaps my example was not clear enough. Look at the following situation:

1.) The sun comes out and we are able to send 2000W to the grid. The smart meter has a reading of -2000W. That is above your +-200W threshold and will be send to the storage.
2.) The storage gets the -2000W information and starts to charge with 2000W (at a very efficient charging point). Now we have a smart meter reading of 0 since the storage consumes everything from the solar panel.
3.) Now we are getting some clouds and the solar power goes back to 100W. Since the battery still charges with 2000W we have a smart meter reading of 1900W which is above the threshold and will be send to the storage.
4.) The storage now reduces its charging by 1900W to just 100W, a very inefficient charging point. The smart meater reading goes back to 0.
5) Now we are getting some more solar power, let's say 199W. The smart meter reading is below your threshold and will not be send to the battery. The charging stays at just 100W, we are sending 199W to the grid. Even if we could charge in a slightly more efficient power point, we are staying at the inefficient 100W.

Even if your +-200W threshold was considered at every point, the battery still ends in an inefficient charging point. Without the current charging/discharging information from the battery we can never control the charging point just by providing some special smart meter readings. That is not possible.

The threshold just leads to a "staircase effect", the current charging point only changes in steps greater than your threshold:

1) Sunny evening, no clouds, we are charging the battery with 2000W, smart meter reading 0, the sun starts sinking, the solar power continously goes down to 0. 
2) We are still charing at 2000W, the energy consumption from the grid (and with that the smart meter readings) slowly goes up 1w...199w
3) At 200w we are above the threshold, the battery starts charging at 1800w and the smart meter goes back to 0w. This continues: 1600w charging, 1400w, ... A staircase effect.

Or did I missunderstood something and you are suggesting something different?

tmtudy, 2025-02-20:
You are indeed correct, i only thoght about startup values, not at variations during the use. Any idea that would achieve what i am looking for? I guess it should be a combination of actual battery power and to/from grid power (like if battery value is below 10w and house value is below 200w do not start to discharge), but maybe you have a simpler idea?

sdeigm, 2025-02-21:
No, I have to admit, I don't have any idea. I think, that without getting the current charging/discharging information from the storage, it is not possible to achive the desired behaviour.

tmtudy, 2025-02-21:
Thanks for thinking along, i have a way to get thosr values myself (socket with api for measuring) but i think that is not really the scope for the emulator. So unless you are interested in developing that logic as well, i will close this

sdeigm, 2025-02-21:
Ok, if you have a way to get the values from the storage, I would be interested to add such a logic :-)

You have a Marstek storage, correct? Would be great if you can provide me a link/information about the interface. 

And you have to support the testing :-)

tmtudy, 2025-02-21:
I will get the battery delivered on monday, as i know my meter is not yet supported, i was preparing the emulator. I now have a p1 meter from homewizard (already integrated in your meter as a generic http input) and plan to also use a plug from them for battery measuring (important as it gives positive and negative values, not all do that). For coding purposes, you would only need to have a second input as generic http (or any of the other supported inputs) and take care of the logic. I can of course also test.

sdeigm, 2025-02-21:
Ok, I understand, you are just using another smart meter to gather the charging/discharging information. That makes sense. 

I think I will close this issue now and will create a new one regarding the "back-channel" from the storage.