sdeigm/uni-meter #47: IoBroker data points as input

schorschis, 2025-02-17:
I hope I've come to the right place with my question, but I haven't found another suitable forum. 
 I am currently reading my electricity meter via an IR read head and the IoBroker smart meter adapter and also a samrtmeter via the ioBroker Modbus adapter.
Is it possible to use data points from IoBroker as input to the uni-meter?

schorschis, 2025-02-18:
Thank you! It is certainly feasible.

sdeigm, 2025-02-20:
I guess it must be

``json-path = "$[0].val"``

You can perfectly test this [here](https://jsonpath.com/).

And a wonderful documentation can be found [here](https://jsontostring.com/jsonpath/)

harvey637, 2025-02-20:
WORKS PERFECT!
Thank you very much for the very helpfull and superfast answer!
Harvey

harvey637, 2025-02-20:
Just as info to be used in readme.md as example for iobroker - adjust to your gusto :-)

```
    generic-http {

      # alles vom Stromzähler lesen (in iobroker über den Adapter "smartmeter" angeschlossen) aktualisiert alle 5 Minuten

      url = "http://192.168.x.x:8082/getBulk/smartmeter.0.1-0:1_8_0__255.value,smartmeter.0.1-0:2_8_0__255.value,smartmeter.0.1-0:16_7_0__255.value/?json"

    # output: [{"id":"smartmeter.0.1-0:1_8_0__255.value","val":16464.7379,"ts":1740054549023,"ack":true},{"id":"smartmeter.0.1-0:2_8_0__255.value","val":16808.0592,"ts":1740054549029,"ack":true},{"id":"smartmeter.0.1-0:16_7_0__255.value","val":4.9,"ts":1740054549072,"ack":true}]

      #username = "username"
      #password = "password"

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"

      channels = [{
        type = "json"
        channel = "energy-consumption-total"
        json-path = "$[0].val"
        scale = 0.001
      },{
        type = "json"
        channel = "energy-production-total"
        json-path = "$[1].val"
        scale = 0.001
      },{
        type = "json"
        channel = "power-total"
        json-path = "$[2].val"
      }]
    }

```

I find the mono-phase "power-total" as "total_act_power" and "total_aprt_power", and a third of the value for each of the phases. Current is calculated accordingly.
The readings of the counters is not visible?!?

Testing with the url given above for the input, testing with http://192.168.x.x/rpc/EM.GetStatus?id=0 for the output

this is just a "give back" for your very good help! Thank you
Harvey

sdeigm, 2025-02-21:
Thanks for the feedback. I will add a section to the documentation and if I find some time create a directory with sample configurations!

closed 2025-02-21 by sdeigm