sdeigm/uni-meter #56: No energy data from SRHDZM

martind40, 2025-02-21:
Anything wrong in my config?

Output:
{"wifi_sta":{"connected":true,"ssid":"my-wifi","ip":"192.168.1.42","rssi":-50},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"19:28","unixtime":1740162520,"serial":1,"has_update":false,"mac":"B827EB364242","ram_total":50648,"ram_free":38376,"ram_lwm":32968,"fs_size":233681,"fs_free":174194,"uptime":74,"temperature":28.08,"overtemperature":false,"tmp":{"tC":28.08,"tF":82.54,"is_valid":true},"emeters":[{"power":0.0,"pf":1.0,"current":0.0,"voltage":230.0,"is_valid":true,"total":0.0,"total_returned":0.0},{"power":0.0,"pf":1.0,"current":0.0,"voltage":230.0,"is_valid":true,"total":0.0,"total_returned":0.0},{"power":0.0,"pf":1.0,"current":0.0,"voltage":230.0,"is_valid":true,"total":0.0,"total_returned":0.0}],"total_power":0.0,"fs_mounted":true}

Config:
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"
  
  input = "uni-meter.input-devices.shrdzm"

  input-devices {
    shrdzm {
      port = 9522
      interface = "0.0.0.0"
    }
  }
}

SHRDZM IP: 
192.168.43.30

sdeigm, 2025-02-22:
I think you configuration looks ok. I implemented the support for the SRHDZM within this [feature request](https://github.com/sdeigm/uni-meter/issues/5).

I think you have to enable then sending of the data on your SRHDZM device as shown in the issue from above. There you have to provide the IP address of your virtual Shelly and the port 9522. The virtual shelly is passive in this scenario, the SHRDZM device is actively sending the data.

Perhaps the SRHDZM device has also a rest interface (I am not sure about that). In that case you could use uni-meter's generic-http input device to actively gather the data using the rest API.

martind40, 2025-02-22:
UDP activated at SHRDZM. Works!

sdeigm, 2025-02-23:
Great to hear, I added your configuration as sample :-)

closed 2025-02-23 by sdeigm