sdeigm/uni-meter #57: Support of mulitple storages (shared load)

sdeigm, 2025-02-23:


TheRealMoeder, 2025-02-28:
How can you add two storages with one energy meter in the hoymiles app?

closed 2025-03-01 by sdeigm

sdeigm, 2025-03-01:
You have to create two systems, each one with a single storage. Than you have to announce the same uni-meter two times via mDNS with two different Shelly-Ids (mac addressess). 

```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.vz-logger"

  output-devices {
    shelly-pro3em {
      min-sample-period = 5000ms
      udp-port = 1010

      client-contexts = [{
        address = "192.168.178.195"
        mac = "b827eb364242"
        power-factor = 0.5
      },{
        address = "192.168.178.92"
        mac = "b827eb365252"
        power-factor = 0.5
      }]
    }
  }
  # ...
}
```

[image]