sdeigm/uni-meter #62: Docker: Listener collides on port 80 on startup

MuellBin2, 2025-03-02:
Moin sdeigm,

thank you very much for the software!
I pulled your  docker and run it in briged network mode. I figured out which storage to attach to the host system (opt) and what settings to write. Default web URL port:80

when I fire up the test run, the listener cannot use port 80.

```
# /opt/uni-meter/bin/uni-meter.sh
25-03-02 19:38:16.184 INFO  uni-meter                - ##################################################################
25-03-02 19:38:16.187 INFO  uni-meter                - # Universal electric meter converter 1.1.0 (2025-03-01 09:54:53) #
25-03-02 19:38:16.187 INFO  uni-meter                - ##################################################################
25-03-02 19:38:16.188 INFO  uni-meter                - initializing actor system

25-03-02 19:38:16.903 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-03-02 19:38:17.241 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-03-02 19:38:17.257 INFO  uni-meter.controller     - creating TibberPulse input device
25-03-02 19:38:18.517 ERROR uni-meter.http.port-80   - failed to bind HTTP server: Bind failed because of java.net.BindException: [/0.0.0.0:80] Address already in use
25-03-02 19:38:18.548 ERROR org.apache.pekko.io.TcpListener - Bind failed for TCP channel on endpoint [/0.0.0.0:80]
java.net.BindException: [/0.0.0.0:80] Address already in use
```

what else is running in the container that blocks the port 80? 

Something is answering on port 80 "_The requested resource could not be found but may be available again in the future._"

sdeigm, 2025-03-03:
Hi, to be honest I don't fully understand your failure report. You are talking about the docker container but from the output above it seems as if you want to start the uni-meter manually. Have you for the output above execed into the docker container and than executed ``/opt/uni-meter/bin/uni-meter.sh``? Then the output would be pretty normal since the uni-meter is already running in the docker container and therefore the 'Address already in use' failure message.

The 'Something' answering on port 80 ist is the uni-meter. The / is currently not bound to anything since normally there would be the configuration UI of the real shell. Please try either http://<your-ip>/status or http://<your-ip>/rpc/EM.GetStatus?id=0 to access the power data from the virtual shelly. I think that should already work in your setup.

MuellBin2, 2025-03-03:
Thank you for the quick reply, it works indeed.
The power-values are zero, so the "Ersatzwerte" are not taken into consideration. Time to dig into that direction.

Have a good day!

closed 2025-03-03 by MuellBin2