sdeigm/uni-meter #64: Feature Request: Add Websocket subscription support

MuellBin2, 2025-03-03:
We should add an active output websockets as used by APSystems' EZHI.
As I described the actions are the following:

1. Binding
2. Connecting
3. Sending data

The Uni-meter needs to answer its details on `/shelly`
```
  "id": "shellypro3em-b827eb364242",
         "mac": "b827eb364242",
        "model": "SPEM-003CEBEU120",
        "gen": 2,
        "fw_id": "20210720-153353/0.6.7-gc36674b",
        "ver": "0.6.7",
        "app": "FourPro",
        "auth_en": "false",
         "auth_domain": "shellypro4pm-b827eb364242"
```

Websocket is called at `/rpc/Ws.SetConfig`
config = request.args.get('config', type=str)
  "status": "success",
 "message": f"Config set successfully: {config}"

`/rpc/Shelly.Reboot` is called to finalize the binding process and the Uni-meter needs to take over the IP adress of the binding process
```
{         "code": 200,
            "message": "OK",
            "headers": {
        "Connection": "close",
        "Content-Length": "4",
        "Content-Type": "application/json",
        "Server": "ShellyHTTP/1.0.0"
  },
 + "body": "null"
```

finally, the Uni-meter pushes the consumption values to the new subscriber at `/meter`:
                "time": int(time.time()),  # UNIX timestamp
                "freq": 50,  # Frequency
                "total_power":  a b and c values  # Total system power
                "emeters": a b and c values  # Three phases

MuellBin2, 2025-03-05:
```
import org.springframework.boot.SpringApplication;
import org.springframework.boot.autoconfigure.SpringBootApplication;
import org.springframework.scheduling.annotation.Scheduled;
import org.springframework.web.bind.annotation.*;
import org.springframework.web.socket.TextMessage;
import org.springframework.web.socket.WebSocketSession;
import org.springframework.web.socket.config.annotation.*;
import org.springframework.web.socket.handler.TextWebSocketHandler;

import java.io.IOException;
import java.util.*;
import java.util.concurrent.CopyOnWriteArrayList;

@SpringBootApplication
@RestController
public class ShellyEmulatorApplication {

    public static void main(String[] args) {
        SpringApplication.run(ShellyEmulatorApplication.class, args);
    }

    private static final List<WebSocketSession> sessions = new CopyOnWriteArrayList<>();

    @GetMapping("/rpc/Ws.SetConfig")
    public Map<String, Object> setWsConfig(@RequestParam(required = false) String config) {
        if (config == null || config.isEmpty()) {
            return Map.of("status", "error", "message", "Config parameter is missing");
        }
        return Map.of("status", "success", "message", "Config set successfully: " + config);
    }

    @GetMapping("/rpc/Shelly.Reboot")
    public Map<String, Object> reboot() {
        return Map.of(
                "code", 200,
                "message", "OK",
                "headers", Map.of(
                        "Connection", "close",
                        "Content-Length", "4",
                        "Content-Type", "application/json",
                        "Server", "ShellyHTTP/1.0.0"
                ),
                "body", "null"
        );
    }

//At this point the live power data needs to be imported, I just used dummy values for my test.
    @GetMapping("/rpc/Shelly.GetStatus")
    public Map<String, Object> getStatus() {
        return Map.of("emeters", List.of(
                Map.of("power", 10, "voltage", 230, "current", 0.4),
                Map.of("power", 9, "voltage", 231, "current", 0.3),
                Map.of("power", 11, "voltage", 229, "current", 0.4)
        ));
    }

    @GetMapping("/shelly")
    public Map<String, Object> getDeviceInfo() {
        return Map.of(
                "id", "shellypro3em-b827eb364242",
                "mac", "b827eb364242",
                "model", "SPEM-003CEBEU120",
                "gen", 2,
                "fw_id", "20210720-153353/0.6.7-gc36674b",
                "ver", "0.6.7",
                "app", "FourPro",
                "auth_en", "false",
                "auth_domain", "shellypro4pm-b827eb364242"
        );
    }

    // Sending the data out for every 2000ms, contains dummy data yet. May need some closer debugging, this part did not work yet.
    @Scheduled(fixedRate = 2000)
    public void sendMeterData() {
        if (sessions.isEmpty()) return;

        Map<String, Object> data = Map.of(
                "time", System.currentTimeMillis() / 1000,
                "freq", 50,
                "total_power", 30,
                "emeters", List.of(generateEmeterData(), generateEmeterData(), generateEmeterData())
        );

        String jsonData = new com.fasterxml.jackson.databind.ObjectMapper().writeValueAsString(data);
        for (WebSocketSession session : sessions) {
            try {
                session.sendMessage(new TextMessage(jsonData));
            } catch (IOException e) {
                e.printStackTrace();
            }
        }
    }

    private Map<String, Object> generateEmeterData() {
        return Map.of(
                "voltage", 230,
                "current", 0.4,
                "power", 10,
                "reactive_power", 0,
                "apparent_power", 0,
                "pf", 0.99,
                "total", 100
        );
    }

    @Configuration
    @EnableWebSocket
    public static class WebSocketConfig implements WebSocketConfigurer {
        @Override
        public void registerWebSocketHandlers(WebSocketHandlerRegistry registry) {
            registry.addHandler(new ShellyWebSocketHandler(), "/meter").setAllowedOrigins("*");
        }
    }

    public static class ShellyWebSocketHandler extends TextWebSocketHandler {
        @Override
        public void afterConnectionEstablished(WebSocketSession session) {
            sessions.add(session);
        }

        @Override
        public void afterConnectionClosed(WebSocketSession session, org.springframework.web.socket.CloseStatus status) {
            sessions.remove(session);
        }
    }
}

```

I tried to convert my Python into Java, hope it may help.

sdeigm, 2025-03-10:
Hi, it took some time but I have build a first version including outbound websocket support :-)

You can either download the github packet: [uni-meter-1.1.1-OUTBOUND.tgz](https://github-registry-files.githubusercontent.com/898342116/0605c700-fd8e-11ef-8763-83e31b2bc08e?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20250310%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250310T080324Z&X-Amz-Expires=300&X-Amz-Signature=6ee009dd994dc04e1ec51cd9497bb4eee8afec20f5d37e1f8f0f2c5a80c6072f&X-Amz-SignedHeaders=host&response-content-disposition=filename%3Duni-meter-1.1.1-OUTBOUND.tgz&response-content-type=application%2Foctet-stream)

Or use a docker package:

sdeigm/uni-meter-1.1.1-OUTBOUND

Since I have neither a real Shelly Pro3EM nor your EZHI device, the version is just based on synthetic tests I have created myself. So I think it will be a little bit of ping-pong and takes some iterations between you and me until it finally works. Would be good if you can trace some debug logs containing the problems.

You can find the source code of the changes in branch 64-feature-request-add-websocket-subscription-support

MuellBin2, 2025-03-11:
Alright, I pulled the Docker and set it up from the old configuration.
EZHI just enabled "SHEM-3" next to Shelly Pro 3EM. Just a side Info.

The Uni-Meter is not being found out of the box, I will go deeper into the settings now and check what I can find out.

sdeigm, 2025-03-11:
I think there are multiple things wrong in your setup. First your are not pulling the docker container that supports outbound websockets. You should modify your docker-compose files as below:

[image]

Then the logging output should look like (also with a version 1.1.1-OUTBOUND in the header line)

```
 uni-meter Pulled
 Container uni-meter  Recreate
 uni-meter Published ports are discarded when using host network mode
 Container uni-meter  Recreated
Attaching to uni-meter
uni-meter  | 25-03-11 11:27:11.998 INFO  uni-meter                - ###########################################################################
uni-meter  | 25-03-11 11:27:12.000 INFO  uni-meter                - # Universal electric meter converter 1.1.1-OUTBOUND (2025-03-10 08:59:19) #
uni-meter  | 25-03-11 11:27:12.000 INFO  uni-meter                - ###########################################################################
uni-meter  | 25-03-11 11:27:12.000 INFO  uni-meter                - initializing actor system
uni-meter  | 25-03-11 11:27:12.275 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
uni-meter  | 25-03-11 11:27:12.390 INFO  uni-meter.controller     - creating ShellyPro3EM output device
```

The last thing: you have configured your system to use a VzLogger input device. If you want to use a Tibber Pulse bridge, please use a config file like I have just provided unter samples: [uni-meter.conf](https://github.com/sdeigm/uni-meter/blob/main/samples/tibber/uni-meter.conf)

sdeigm, 2025-03-11:
If that works as expected, it would be good if you could trace the problems using the following logging setup:

[image]

Looking forward to your results and findings :-)

MuellBin2, 2025-03-11:
Hm strange should be 1.1.1. Also it is not reading my config file. I think I need to reacreate the Docker from screatch, it doesn't contain what it says:

```
{
AppArmorProfile:"docker-default"
Args:[]
Config:{
AttachStderr:false
AttachStdin:false
AttachStdout:false
Domainname:""
Entrypoint:[
]
Env:[
]
Hostname:"shellypro3em-B827EB364242"
**Image:"sdeigm/uni-meter:1.1.1-OUTBOUND"**
Labels:{}
MacAddress:"B8:27:EB:36:42:42"
OpenStdin:true
StdinOnce:false
Tty:true
User:""
WorkingDir:""
}
Created:"2025-03-11T10:58:13.666784682Z"
```

sdeigm, 2025-03-11:
No, there must be something wrong with your docker compose file. I tested it now with both versions x64 and arm64. The correct version from hub.docker.com should print 1.1.1-OUTBOUND in the second line of the log:

[image]

Perhaps there is an old container running in your environment, then please use ``docker ps``  and ``docker kill`` to get rid of that container.

And there should also be a message "creating TibberPulse input device" if you configured the uni-meter to use Tibber:

[image]

MuellBin2, 2025-03-11:
[image]

I really dont belive this anymore 😄 . I have the two images locally. I deleted the 1.1.0 Image, then I run the 1.1.1 Image. Still it shows me version 1.1.0 when I start it up. I think my docker is the probleme here oO, lemme fix that ...

MuellBin2, 2025-03-11:
Ok I found the issue, i was mounting /opt with all its sub folder obviously containing version 1.1.0 .. typicall level 8 mistake ;)

so its now up and running, values are correct. Logging is on TRACE.

Maybe I need to configure logs longer. But it start up good, let me put down the interval so we dont have too many tibber massages inside the trace file.

MuellBin2, 2025-03-11:
https://www.dropbox.com/scl/fi/kj9ytd4lx7ynivt8jt42o/20250311-TraceFile_uni-meter.txt?rlkey=0s9pvomumzktt39avs8g668i4&dl=0

MuellBin2, 2025-03-11:
I manually pinged the docker container on the lookup:
/shelly

<html><body>

type | "SPEM-003CEBEU120"
-- | --
mac | "B827EB364242"

</body>
</html>

In my implementation I have:

<html><body>
<div class="toolbar"><div class="devtools-separator"></div><div class="devtools-searchbox"><input class="searchBox devtools-filterinput" placeholder="Filter JSON" value=""></div></div><div class="panelContent" id="json-scrolling-panel" tabindex="0">
  |  
-- | --
app | "FourPro"
auth_domain | "shellypro4pm-b827eb364242"
auth_en | "false"
fw_id | "20210720-153353/0.6.7-gc36674b"
gen | 2
id | "shellypro3em-b827eb364242"
mac | "b827eb364242"
model | "SPEM-003CEBEU120"
ver | "0.6.7"

</div>
</body>
</html>

That auth information etc needs to be added

sdeigm, 2025-03-11:
I think you are right, the EZHI is not happy with the response from the http://<shelly-ip>/shelly command. In you log I can see, that this funtion is triggered once:

```
25-03-11 12:49:16.737 TRACE uni-meter.output         - Shelly.onShellyGet()
```

If I am using a normal HTTP browser and send this request to my local uni-meter, I am getting the following answer:

```
{
    "type": "SPEM-003CEBEU120",
    "mac": "B827EB364242",
    "auth": false,
    "fw": "20241011-114455/1.4.4-g6d2a586",
    "discoverable": true,
    "longid": 1,
    "num_outputs": 0,
    "num_meters": 3
}
```

Can you please send this request ```http://<shelly-ip>/shelly``` to your implementation and post the received JSON here so that we can see what is different?

MuellBin2, 2025-03-11:
> ```
> <html><body>
> <div class="toolbar"><div class="devtools-separator"></div><div class="devtools-searchbox"><input class="searchBox devtools-filterinput" placeholder="Filter JSON" value=""></div></div><div class="panelContent" id="json-scrolling-panel" tabindex="0">
>   |  
> -- | --
> type | "SPEM-003CEBEU120"
> mac | "B827EB364242"
> auth | false
> fw | "20241011-114455/1.4.4-g6d2a586"
> discoverable | true
> longid | 1
> num_outputs | 0
> num_meters | 3
> 
> </div>
> </body>
> </html>
> ```

yes, thats what I get

sdeigm, 2025-03-11:
Ok, seems to be different from the Shelly [specification](https://shelly-api-docs.shelly.cloud/gen1/#shelly), but perhaps that has changed. I will adjust that accordingly.

MuellBin2, 2025-03-11:
> Ok, seems to be different from the Shelly [specification](https://shelly-api-docs.shelly.cloud/gen1/#shelly), but perhaps that has changed. I will adjust that accordingly.

Be careful, ChatGTP created it using the specifications, but unnecessary data fields don't hurt

sdeigm, 2025-03-11:
Perhaps the Pro3EM implements already this [specification](https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/Shelly). That is closer to your output

sdeigm, 2025-03-11:
There is now an updated container [1.1.1-OUTBOUND2](https://hub.docker.com/layers/sdeigm/uni-meter/1.1.1-OUTBOUND2/images/sha256-9bc69a57038728bdbb39a6055210089c3dbbb97a79cc1ace8eba684af0a1c3c8) on dockerhub.

That version should provide the following info on http://.../shelly:

```
{
    "id": "shellypro3em-b827eb364242",
    "mac": "B827EB364242",
    "slot": 1,
    "model": "SPEM-003CEBEU",
    "gen": 2,
    "fw_id": "20241011-114455/1.4.4-g6d2a586",
    "ver": "1.4.4",
    "app": "Pro3EM",
    "auth_en": false,
    "auth_domain": null,
    "profile": "triphase"
}
```

MuellBin2, 2025-03-11:
yes, it apperas. But "binding failed" after selecting it. Let me traces this for you:

https://www.dropbox.com/scl/fi/z4h97it5mdhis13jr79ti/20250311-TraceFile_uni-meter.txt?rlkey=p2ejf3sl7sqko29unjoizfs0a&dl=0

MuellBin2, 2025-03-11:
Uni Meter seems to send now already, but binding was not accepted at EZHI. Did you react on the "restart" command?

MuellBin2, 2025-03-11:
Yes, lets add that. I may now receive 0 values at the EZHI

sdeigm, 2025-03-11:
Do you have to failure message regarding the missing Shelly.Reboot from the log?

MuellBin2, 2025-03-11:
https://www.dropbox.com/scl/fi/a31pfiq4bvr96x65644ne/20250311-TraceFile_uni-meter2.txt?rlkey=p0ga0tqn4qjpvobwqqb9i0m2j&st=huex617l&dl=0

The app always shows UniMeter 2x, even after restarting.

MuellBin2, 2025-03-11:
Again, I am stupid. My own device has the same name, thats why i was able to bind one of them.
With only Uni-meter, binding still fails. Let me recreate that trace. sorry

MuellBin2, 2025-03-11:
https://www.dropbox.com/scl/fi/uqidg3d4mmokrg7lssjss/20250311-TraceFile_uni-meter3.txt?rlkey=sof3e9iswqdcxyd4grz6bpo46&dl=0

Latest binding trace

sdeigm, 2025-03-11:
So, there is again a new container 1.1.1-OUTBOUND3. At least the Shelly.Reboot method is in there. I am not sure, that the response is 100% ok, since I can not do a real reboot and close the connection.

In the traces I can already see, that the uni-meter notifies the power data to the EZHI:

```
25-03-11 15:17:54.673 DEBUG uni-meter.output.outbound-websocket - send => {"src":"shellypro3em-b827eb364242","method":"NotifyStatus","params":{"ts":1741706274.67,"em:0":{"id":0,"a_current":0.24,"a_voltage":230,"a_act_power":54.85,"a_aprt_power":54.85,"a_pf":1,"a_freq":50,"b_current":0.24,"b_voltage":230,"b_act_power":54.85,"b_aprt_power":54.85,"b_pf":1,"b_freq":50,"c_current":0.24,"c_voltage":230,"c_act_power":54.85,"c_aprt_power":54.85,"c_pf":1,"c_freq":50,"total_current":0.72,"total_act_power":164.55,"total_aprt_power":164.55}}}
```

But for some reason it seems as if the EZHI is echoing that data:

```
25-03-11 15:17:54.827 TRACE uni-meter.output         - incoming websocket connection cbccbde9-01ac-412e-92e3-2baaf07620f2 received message: {"src":"shellypro3em-b827eb364242","method":"NotifyStatus","params":{"ts":1741706274.67,"em:0":{"id":0,"a_current":0.24,"a_voltage":230,"a_act_power":54.85,"a_aprt_power":54.85,"a_pf":1,"a_freq":50,"b_current":0.24,"b_voltage":230,"b_act_power":54.85,"b_aprt_power":54.85,"b_pf":1,"b_freq":50,"c_current":0.24,"c_voltage":230,"c_act_power":54.85,"c_aprt_power":54.85,"c_pf":1,"c_freq":50,"total_current":0.72,"total_act_power":164.55,"total_aprt_power":164.55}}}
```

sdeigm, 2025-03-11:
I have not changed this (at least I think). And I also think it is correct according to this spec:

[Sys.SetConfig](https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/Sys/#syssetconfig-example)

Is it now not working anymore?

MuellBin2, 2025-03-11:
https://www.dropbox.com/scl/fi/uppbt6c1mc3u969gadubr/20250311-TraceFile_uni-meter.txt?rlkey=dmpptbiiuiqcnpcqy1mpbyvhf&dl=0

sdeigm, 2025-03-11:
It seems as if the  EZHI is just echoing the NotifyStatus messages from the uni-meter.

Do you have a trace how this websocket communication looks like in a successful case?

MuellBin2, 2025-03-11:
Maybe it is doing something. Now it stopped mirroring the massages back i think.

I have a consumption from grid of 100W, and the EZHI injects 100W though being on 225W upper limit. 
Still not working OK, it shoud inject 100W more to cancel out grid consumption. But there seems to be happening something rn.

MuellBin2, 2025-03-11:
I am going to increase 		min-sample-period = 2500ms and wait for an effect.
What happens if I restart Uni-Meter? The connection is gone, right? I will have to bind again?

sdeigm, 2025-03-11:
Normally the outbound websocket connection should be automatically reopened again, when it is closed.

sdeigm, 2025-03-11:
Yes, the uni-meter does not store the request for the outbound-websocket. You have to bind again when you restart.

The min-sample-period is not evaluated in the context of the outbound websocket. Currently whenever the uni-meter gets new data from the Tibber Bridge, it is immediately notified on the outbound websocket connection.

MuellBin2, 2025-03-11:
It is working somehow. Maybe it has a huge lag due to the implementation inside EZHI.

The battery is starting to charge now after having -80w for some time. I think I need to fiddle around with it now a little bit.

sdeigm, 2025-03-11:
Ok, I have to figure out, why I am getting this "echo" and if that is normal. Then I can just ignore it.

Regarding the connection: as said nothing is stored. So when you restart the container, you have to bind again. But if the connection is just dropped for whatever reason (network, EZHI not available) the uni-meter should normally try to build it up again. I will again test this in my environment.

Just keep me updated about you findings :-)

MuellBin2, 2025-03-11:
[image]

This chart appeared when Shelly was bind. It shows that data comes through, now it fell to 0 W (= lost connection?). It would be really good to have a stable connection, i.e. put the IP adress of the device into the config to re-send data when Uni Meter restarts.

I will try a slower output again, every 5s.

sdeigm, 2025-03-11:
I have fixed the problem with the unstable connection. Whenever the connection is dropped it should be automatically recreated after a few seconds. The test version for this is 1.1.1-OUTBOUND4.

But this does not handle a restart of the uni-meter. When you restart the uni-meter you still have to bind again.

I will also improve the behavior so that the min-sample-frequency is also taken into account for the outbound websocket connection. But that takes some time, not today ...

MuellBin2, 2025-03-11:
I run several tests now. The EZHI is always reducing power to 0 after a certain amount of time, but always in the same pattern. 
It seems like the power values are not accepted by the EZHI?

Ill try version 4 now.

[image]

sdeigm, 2025-03-11:
The values send to the EZHI can be seen in this log message:

```
25-03-11 16:29:43.591 DEBUG uni-meter.output.outbound-websocket - send => {"src":"shellypro3em-b827eb364242","method":"NotifyStatus","params":{"ts":1741710583.59,"em:0":{"id":0,"a_current":-0.08,"a_voltage":230,"a_act_power":-19.35,"a_aprt_power":-19.35,"a_pf":1,"a_freq":50,"b_current":-0.08,"b_voltage":230,"b_act_power":-19.35,"b_aprt_power":-19.35,"b_pf":1,"b_freq":50,"c_current":-0.08,"c_voltage":230,"c_act_power":-19.35,"c_aprt_power":-19.35,"c_pf":1,"c_freq":50,"total_current":-0.24,"total_act_power":-58.05,"total_aprt_power":-58.05}}}
```

MuellBin2, 2025-03-11:
Yes, but as you said, its just a mirror of what is sent. 
The Graph that the app displays seems to be correct, it matches more or less the tibber app.

Maybe the EZHI looks for voltage and frequency as well? I have 234 49,96 Hz on L2 and 233 and 49,99 Hz on the same line according to my hoymiles.

sdeigm, 2025-03-11:
No, the "send => " direction is what uni-meter sends out. For whatever reason the EZHI echos that and it comes back as "recv <=". But the values are the ones received from the Tibber Bridge. Voltage and frequency are just default values, but they are in the correct range and should be accepted by the EZHI (if it is using them at all).

It's currently not really clear for me what's going wrong. Is the EZHI not adjusting its output correctly?

MuellBin2, 2025-03-11:
The EZHI adjusts its output slowly gen zero. It does not react properly on the received values. I found a guy with a real Shelly installed. I will interview him how his EZHI behaves.

MuellBin2, 2025-03-11:
I also contacted APSystems. Let's see if they have an answer. The data is been sent to their backend.

closed 2025-03-12 by sdeigm