sdeigm/uni-meter #65: Q: how to setup mqtt topics in uni-meter.conf file

SpaceTeddy, 2025-03-05:
Hi,
I would like to ask a question regading setup correct mqtt json tpics in configuration file
My file looks like the following:
```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  output-devices {
    shelly-pro3em {
      device {
        mac = "B827EB364242"
        hostname = "shellypro3em-b827eb364242"
      }
    }
  }
  input = "uni-meter.input-devices.mqtt"

  input-devices {
    mqtt {
      url = "tcp://192.168.0.202:1883"
      username = "xxxx"
      password = "xxxx"

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"

      channels = [{
        type = "json"
        topic = "tele/tasmota_65F460/SENSOR"
        channel = "power-total"
        json-path = "$..16_7_0"
        scale = 1.0 # default, can be omitted
      },{
        type = "json"
        topic = "tele/tasmota_65F460/SENSOR"
        channel = "energy-consumption-total"
        json-path = "$..ESL"
      },{
        type = "json"
        topic = "tele/tasmota_65F460/SENSOR"
        channel = "energy-production-total"
        json-path = "$..ESL"
      }]
    }
  }
}

my json string looks like the following:

```
12:52:13.286 MQT: tele/tasmota_65F460/SENSOR = {"Time":"2025-03-05T12:52:13","SM":{"1_8_0":2817.98268,"16_7_0":-404,"36_7_0":-553,"56_7_0":38,"76_7_0":111,"32_7_0":234.2,"52_7_0":234.1,"72_7_0":234.3,"96_1_0":"1EBZ0102007534"},"ESL":{"ESL":400.48}}

Question:
How to setup the following 
power-total, shall be "16_7_0"
    json-paths:  
energy-consumption-total, shall be "16_7_0" as well??
energy-production-total, shall be "ESL":{"ESL":400.48}

Any idea how to setup the paths?

the status shows the following , but only the "16_7_0" value as "total power"

[image]

thx

sdeigm, 2025-03-05:
The JSON Path for 

power-total should be: "$.SM['16_7_0']"
energy-consumption-total should be "$.SM['1_8_0']"
energy-production-total should be "$.ESL['ESL']" (if that is the production value ...)

Hope that helps. 

A good place to test this is https://jsonpath.com/

SpaceTeddy, 2025-03-05:
Seems to be working!

Tahnaks a lot

closed 2025-03-05 by sdeigm