sdeigm/uni-meter #66: Question: Would the Shelly App find the virtual device in the same LAN?

tmb80c, 2025-03-05:
Hi,

great work. My plan is to link a SolarEdge EM which provides values through TCP modbus with the Anker Solarbank AC (if I get it right then th is at the moment not yet supported).

I'm trying the docker version of uni-meter under Truenas Scale (have to find a way to assign an IP to the docker so that I can map/expose the port 80). Once configured correctly should the official Shelly App find this virtual Shelly EM in the same network or is the implementation to limited for the official APP?

Thanks

sdeigm, 2025-03-05:
The scan of the Shelly App will find the virtual Shelly, but you can not add it to the App/Cloud. For that to work additional functions would have to be implemented.

For the Shelly to be discoverable, it is not sufficient just to expose the Port 80. Also the avahi daemon on Port 5353 has to be exposed. Dependening on your configuration either the avahi daemon within the docker container or the avahi daemon of the guest system could be used.

Regarding the Solar Edge EM: I alread have one Modbus device included in the uni-meter. If you can provide me the register information of the Solar Edge Device, it is perhaps not to difficult to add a Solar Edge SM as input ...

tmb80c, 2025-03-05:
> Regarding the Solar Edge EM: I alread have one Modbus device included in the uni-meter. If you can provide me the register information of the Solar Edge Device, it is perhaps not to difficult to add a Solar Edge SM as input ...

Solaredge implemented the SunSpec Protocol.

Codes are published here, please search "SolarEdge Inverters - SunSpec Logging - Technical Note"

thomluther, 2025-03-10:
Question:
Would the Anker Solarbank be able to use the virtual Shelly? AFAIK, Anker tries to register the device through the Shelly cloud. 
Is there already a positive or negative confirmation whether the uni meter can be used with the Solarbank?

tmb80c, 2025-03-11:
> thomluther

I’m able to register the uni-meter docker with the Shelly Cloud.

Then in the Anker App I could also add the uni-meter (enable device control).

My uni-meter is off-line since my physical powermeter (Solaredge Modbus) is not yet supported in uni-meter. So I can’t test further but would be happy to volunteer.

￼￼￼

sdeigm, 2025-03-11:
Ok, interesting, that you can register the device within the Shelly Cloud. I wasn't aware that this already works. I am just wondering if this "offline" is really related to the uni-meter not receiving any data from the Solaredge device. I fear that it is more related to the uni-meter not sending any data up into the cloud.

Anyhow I will provide a first version which hopefully can at least read some data localy from the Solaredge device using the interface description you provided above.

tmb80c, 2025-03-11:
Is this  helpful: "Device Information" for the uni-meter in the Shelly App looks like this (see screenshot). Interesting that it does not provide the IP address (Device-IP/Geräte-IP). 

[image]

Geräte-IP).

tmb80c, 2025-03-11:
> Ok, interesting, that you can register the device within the Shelly Cloud. I wasn't aware that this already works. I am just wondering if this "offline" is really related to the uni-meter not [...]

AFAIK, the Anker should communicate with the Shelly locally, if in the same LAN. Let's cross the fingers that the Anker reads the Shelly EM and then the Anker SB communicates to the cloud from where the APP receives the data.

> Anyhow I will provide a first version which hopefully can at least read some data localy from the Solaredge device using the interface description you provided above.

Cool!

letsrock303, 2025-03-16:
Hello, 
just a hint from my side....maybe it is helpful

The shelly muss be configured for "cloud = true" befor you can add the device to the App.
I did a short test with the uni-meter and on shelly dimmer device.

Maybe you can add the cloud configuration to your configuration file.
So as i understand your code, you have hard-coded this or? ("cloud":{"enabled":false,"connected":false},)

so I think when I see whats in the output from  "http://<ip>/status" - the information is not vaild for the app

sdeigm, 2025-03-18:
I have prepared a new uni-meter test version [uni-meter-1.1.2-CLOUD.tgz](https://github-registry-files.githubusercontent.com/898342116/d448bd80-0396-11f0-9805-e2c6a536551d?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20250318%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250318T065558Z&X-Amz-Expires=300&X-Amz-Signature=b8f68c81224530a5cbe48ed95bc963a822131e5329eb48b440542f628c7f3e83&X-Amz-SignedHeaders=host&response-content-disposition=filename%3Duni-meter-1.1.2-CLOUD.tgz&response-content-type=application%2Foctet-stream) as github package as well as docker container sdeigm/uni-meter:1.1.2-CLOUD.

That version successfully answers all requests send out be the Shelly App when scanning the local network and adding a Shelly. Anyhow it ends with the screenshot @tmb80c added above: the uni-meter IP address is not known in the cloud. That must be information that is send by the shelly when reporting its state to the cloud. That interface is nowhere documented as far as I know and its not possible to trace that communication using for instance wireshark since it is SSL communication.

So if the Zendure device needs this information from the Shelly cloud, then I think we are stuck. Anyhow I still think/hope that it is not necessary to provide the information in the cloud. In one of the issues here there was already a trace where the Zendure device contacted the Shelly. So perhaps you all try again to connect the Zendure device with the shelly.

@letsrock303: the new version also provides the cloud status as {"cloud": {"enabled":true, "connected":true}}.

MuellBin2, 2025-03-18:
APSystem just did an update of their app changing the bind process. Now they ask for a shelly account, it seems it is the same feature as tmb80c describes. 
I will test that once shelly cloud is integradetd&working

sdeigm, 2025-03-20:
Hm, I think that is bad new since I am not able to update the device's data in the Shelly cloud. So the APSystem App will not be able to get the IP address of the virtual shelly from the cloud data.

sdeigm, 2025-04-13:
Closing this with the final outcome: Shelly App finds the device, but it will always stay "offline" since it is not possible to send data from the virtual Shelly into the cloud.

closed 2025-04-13 by sdeigm