sdeigm/uni-meter #68: QUESTION: Is there a possibility to use the Uni-Meter with Zendure App (Hyper 2000)

Milvus11, 2025-03-12:
I have the Zendure Hyper 2000 with Battery here, which would run perfectly with a Pro3EM. However, I can't attach one in the rental flat. What would need to be done to connect them, is there already any idea/info about it?

I run a Tibber Pulse for the data for the Uni-Meter

Thanks a lot in advance!

floriandrott, 2025-03-12:
I have the same issue/question. As mentioned in #66 I was able to get the uni-meter into the Shelly app. But it seems like that the Shelly Cloud is waiting for some data of the meter device.

Within the Zendure App I have authorized the Shelly Cloud, but there are no devices listed which I can add to Zendure. Any help would be very much appreciated.

Milvus11, 2025-03-12:
Thank you for your fast response and willingness to look into this!
According to a German entry at zendure forum, a local connection without cloud should also be possible:

https://www.zendure.de/blogs/news/shelly-zendure-hyper-lokale-kommunikationsbetriebsanleitung?srsltid=AfmBOoq8YcsrrzLDJ9BinZSssoWlgPs3cbbFomvIVCxOA5JhXBAzvdDX

Would be the nicest option, to leave the cloud out and get it running with the uni-meter!

sdeigm, 2025-03-12:
I don't have a Zendure device available for testing. But if you have such a device available, can you please if it is possible to connect the uni-meter just locally without the cloud as described in the document from above? In theory the uni-meter in the version 1.1.1 should provide (almost) all possible APIs that could be used to connect a device with a shelly/uni-meter locally (mDNS, UDP, OutboundWebsocket). If something missing we can figure this out using uni-meters debug log.

Milvus11, 2025-03-12:
I will try to, though I cannot guarantee, I am coding-wise good enough. I do know, that my raspi 3b+ with uni-meter works, as I have the data in Home assistant. I will sit down and try to find out, what exactly i need to do. Just by trying to find the shelly in the app does not work for me so far. If you have any step by step order I should try, I will follow that!

sdeigm, 2025-03-12:
So if you already have a running uni-meter the most stuff is already done. Have you also set up the avahi Daemon as described in the Readme here on github? The avahi Daemon provides the mDNS information that is most likely used by Zendure to look up the Shelly locally. If the uni-meter ist not found when the avahi Daemon is running as described in the Readme, we have to look into the uni-meter logs if you have some entries like "unknown RPC method XYZ'" which indicates that Zendure is using a method currently not yet implemented.

Milvus11, 2025-03-12:
So I try to find the Shelly via the Zendure App. So far nothing to be found. Here is the log of uni-meter. Does this help you anyhow? If you have any things I should do, I will try!:

```
25-03-12 20:22:40.291 INFO  uni-meter                - ##################################################################
25-03-12 20:22:40.315 INFO  uni-meter                - # Universal electric meter converter 1.1.0 (2025-03-01 09:54:53) #
25-03-12 20:22:40.316 INFO  uni-meter                - ##################################################################
25-03-12 20:22:40.317 INFO  uni-meter                - initializing actor system
25-03-12 20:22:42.997 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-03-12 20:22:44.411 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-03-12 20:22:44.466 INFO  uni-meter.controller     - creating TibberPulse input device
25-03-12 20:22:44.507 INFO  uni-meter.output         - using total default power of 150.0
25-03-12 20:22:46.590 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-03-12 20:22:49.042 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-03-12 20:23:20.256 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-03-12 20:24:11.039 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-03-12 20:24:42.274 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-03-12 20:25:23.885 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-03-12 20:26:02.111 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-03-12 20:26:33.311 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-03-12 20:27:31.685 ERROR uni-meter.input          - Failed to parse status response: Timeout
```

sdeigm, 2025-03-13:
Good morning, first of all it seems as if your TibberPulse input has some problems since I can see these

`25-03-12 20:23:20.256 ERROR uni-meter.input          - Failed to parse status response: Timeout`

failure messages. Does your uni-meter provide valid power values at all? You can check this using the following URL in your webbrowser:

`http://<uni-meter-ip>/rpc/EM.GetStatus?id=0`

That should give you an output like

[image]

Above you started the uni-meter manually using the command `sudo /opt/uni-meter/bin/uni-meter.sh`. That's ok, but I think you uni-meter service is still running in the background. Therefore you are getting the 

`25-03-12 20:59:04.611 ERROR uni-meter.http.port-80 - failed to bind HTTP server: Bind failed because of java.net.BindException: [/0.0.0.0:80] Address already in use`

failure message. There can only be one uni-meter running. So if you want to run it manually as above please execute `sudo systemctl stop uni-meter` before. That stops the service. You can check if there is only one running uni-meter using the following command. If there is only one uni-meter running you should just see the 2 output lines:

```
sdeigm@raspberrypi:~ $ ps -eaf | grep uni-meter.sh
root     1403115       1  0 Mär12 ?       00:00:00 /bin/bash /opt/uni-meter/bin/uni-meter.sh
sdeigm   1463683 1463522  0 09:34 pts/0    00:00:00 grep --color=auto uni-meter.sh
sdeigm@raspberrypi:~ $
```

Attached you will find a logback.xml file which enables the logging more fine grained, just for the output device and the websocket connection. Please use this logback.xml file (replace the /opt/uni-meter/config/logback.xml file with this file, remove the *.txt at the end) and let's see if we are getting better logging results with these settings

[logback.xml.txt](https://github.com/user-attachments/files/19225206/logback.xml.txt)

Milvus11, 2025-03-13:
Unfortunately I wasn't able to get into it today, I get on it as soon as possible! Thank you already for the input!

Milvus11, 2025-03-14:
Good evening! So following your Feedback the following response:

I do get an output by using your provided link:

`{
  "id": 0,
  "a_current": 0.32,
  "a_voltage": 230,
  "a_act_power": 72.67,
  "a_aprt_power": 72.67,
  "a_pf": 1,
  "a_freq": 50,
  "b_current": 0.32,
  "b_voltage": 230,
  "b_act_power": 72.67,
  "b_aprt_power": 72.67,
  "b_pf": 1,
  "b_freq": 50,
  "c_current": 0.32,
  "c_voltage": 230,
  "c_act_power": 72.67,
  "c_aprt_power": 72.67,
  "c_pf": 1,
  "c_freq": 50,
  "total_current": 0.96,
  "total_act_power": 218.01,
  "total_aprt_power": 218.01
}`

The values do also change when I refresh the page. I have no clue why the error occurs.

2nd, thank you for clarification. I was trying out various different things, including starting the service everytime I changed something in the config-file etc. I took care of that now, the output looks like the one you provided.

I used your provided logback file and got the following log. only saw two weird things, but you might be knowing what you are looking for :-D 

Several times:

25-03-14 19:24:57.079 TRACE uni-meter.output         - Shelly.onShellyGet()
25-03-14 19:24:57.099 TRACE uni-meter.output         - Shelly.onShellyGet()
25-03-14 19:24:57.111 TRACE uni-meter.output         - Shelly.onSettingsGet()
25-03-14 19:24:57.126 TRACE uni-meter.output         - ShellyPro3EM.onStatusGet()
25-03-14 19:24:57.126 TRACE uni-meter.output         - using default power factor 1.0 for /192.178.188.70
25-03-14 19:25:26.616 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-03-14 19:25:57.793 ERROR uni-meter.input          - Failed to parse status response: Timeout

also: 

25-03-14 17:50:44.167 DEBUG uni-meter.output.http    - unhandled HTTP path: 

[uni-meter.log](https://github.com/user-attachments/files/19252935/uni-meter.log)

I played around a bit with opening a port on my FritzBox router for the Raspi, but nothing happened in the Zendure App, so I closed all ports on my router again.

sdeigm, 2025-03-14:
Hi, I think you already extracted the relevant parts. It seems as if the Zendure is already trying to contact the uni-meter.

In the last day I added support for the EZHI micro inverter. Getting this connection to work needed exactly changes in the methods that are called by the Zendure system in your logfile.

So I think you should try the version 1.1.1 I just released 3 days ago. The changes in that version might already be sufficient for the Zendure support. If not please provide a logfile with the same settings as you used before.

Milvus11, 2025-03-14:
Thank you for the suggestions. Unfortunately no luck. I also tried to set the Raspi on 2.4 Ghz like the Hyper, so that this wouldn't be the issue. 
Here is the log file:

[uni-meter.log](https://github.com/user-attachments/files/19254826/uni-meter.log)

sdeigm, 2025-03-15:
I can not find any entries like

Shelly.onShellyGet()

in the new logfile anymore. So during this run of the uni-meter there was no connection request from the outside.

Did you again try to connect your system with the uni-meter as described in your [Zendure Documentation](https://www.zendure.de/blogs/news/shelly-zendure-hyper-lokale-kommunikationsbetriebsanleitung?srsltid=AfmBOoq8YcsrrzLDJ9BinZSssoWlgPs3cbbFomvIVCxOA5JhXBAzvdDX) you linked above?

In the older log you provided there was a connection from 192.178.188.70. Which device in your network has this IP address?

Milvus11, 2025-03-15:
Good Morning, 
I checked, that IP belongs to my Home Assistant. Interestingly I did not knowingly try to connect them atm. Sorry for that confusion. I tried to connect like stated in the manual, but it does not find / reach out to it. The log shows nothing.

sdeigm, 2025-03-15:
But in the trace before you already had connection requests from the 192.178.188.70 IP address. In the last trace there is no connection request at all. Isn't it possible to trigger the search for Shelly from within the Zendure device's App? For me it seems as if there is a menu item to trigger the search:

[image]

Somehow we again have to get these

```
25-03-14 19:24:57.099 TRACE uni-meter.output - Shelly.onShellyGet()
25-03-14 19:24:57.111 TRACE uni-meter.output - Shelly.onSettingsGet()
```

calls from the Zendure device against the uni-meter.

Milvus11, 2025-03-15:
That trigger does at least not show anything at all in the log-file. That is why I assumed it might need an initial connection via the cloud to receive the Shelly Infos. However, I played around on all connection possibilities I can think of and were in the Zendure app and did not receive anything. 

Following is the only thing I got out, trying to connect the Raspi to the Shelly App:

`25-03-15 11:29:25.373 DEBUG uni-meter.output.http    - unhandled HTTP path: rpc/
25-03-15 11:29:25.467 TRACE uni-meter.output         - ShellyPro3EM.onShellyGet()
25-03-15 11:29:25.467 TRACE uni-meter.output         - Shelly.rpcGetDeviceInfo()
25-03-15 11:29:25.467 TRACE uni-meter.output         - ShellyPro3EM.rpcGetDeviceInfo(): {"id":"shellypro3em-abcdefghijkl","mac":"abcdefghijkl","slot":1,"model":"SPEM-003CEBEU","gen":2,"fw_id":"20241011-114455/1.4.4-g6d2a586","ver":"1.4.4","app":"Pro3EM","auth_en":false,"auth_domain":null,"profile":"triphase"}
25-03-15 11:29:29.049 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-03-15 11:29:35.381 DEBUG uni-meter.output.http    - unhandled HTTP path: rpc/
25-03-15 11:30:00.224 ERROR uni-meter.input          - Failed to parse status response: Timeout
`

sdeigm, 2025-03-15:
Whatever you did, it was helpful :-)

The 

```25-03-15 11:29:35.381 DEBUG uni-meter.output.http    - unhandled HTTP path: rpc/ ```

was helpful. Zendure is using the JSON RPC API via HTTP POST request. All other devices before used the HTTP GET request. I will add support for the HTTP POST request.

You just have to remember what you did to trigger this action ;-)

sdeigm, 2025-03-15:
Support will be added in this ticket: https://github.com/sdeigm/uni-meter/issues/70

Milvus11, 2025-03-15:
okay, :-D tbh., I think, as stated, that it was from the Shelly-App. But nevertheless, as Shelly and Zendure somehow communicate, one can hope, that they kind of use the same methods. I am definitely ready to give it a shot!

Thank you so much for all your help!

iroc-fabi, 2025-03-15:
Mein Zendure Hyper wird nächste Woche geliefert. Bin dann auch bereit zum testen.

burpeenerd, 2025-03-20:
@iroc-fabi has your Zendure arrived? 
#70 has been implemented and can be tested via docker container using https://hub.docker.com/layers/sdeigm/uni-meter/1.1.3-SNAPSHOT/images/sha256-b9e31206e5072a501d1c8fa86b45d661064811b822092f4a5a0224a970b07694

iroc-fabi, 2025-03-20:
> [@iroc-fabi](https://github.com/iroc-fabi) has your Zendure arrived? [#70](https://github.com/sdeigm/uni-meter/issues/70) has been implemented and can be tested via docker container using [...]

Leider noch nicht geliefert.

lpetflo, 2025-03-23:
Will test it as soon as my Zendure System arrives. Thank you so much for your amazing work!

iroc-fabi, 2025-03-23:
Soll das so sein?

[image]

burpeenerd, 2025-03-23:
@iroc-fabi: Hast Du eine (random) neue mac-adresse für das virtuelle Shelly vergeben?
Es sollte möglich sein, das virtuelle Shelly hinzuzufügen - aber dann bleibt es bei "ausstehende Verbindung" in der Liste stehen.

Milvus11, 2025-03-23:
@iroc-fabi, which version do you use? I couldn't get the cloud version from github somehow..

iroc-fabi, 2025-03-23:
> [@iroc-fabi](https://github.com/iroc-fabi), which version do you use? I couldn't get the cloud version from github somehow..

The actually version 1.1.3

iroc-fabi, 2025-03-23:
> [@iroc-fabi](https://github.com/iroc-fabi): Hast Du eine (random) neue mac-adresse für das virtuelle Shelly vergeben? Es sollte möglich sein, das virtuelle Shelly hinzuzufügen - aber dann bleibt es bei "ausstehende Verbindung" in der Liste stehen.

Nein noch nicht. Ist noch der Standard aus der Installation. Teste eine andere MAC heute Abend.

iroc-fabi, 2025-03-23:
Ich bin einen Schritt weiter. 
So alles richtig, damit ich mit Zendure verbinden kann?

[image]

burpeenerd, 2025-03-23:
Genau. So weit sind wir auch (alle). Die Frage ist nun, ob das nun für jemanden mit Zendure Hardware reicht, um den virtuellen Shelly zu verbinden, oder nicht. 
Ohne Hardware funktioniert es jedenfalls nicht.

iroc-fabi, 2025-03-23:
> Genau. So weit sind wir auch (alle). The question is now whether that is enough for someone with Zendure hardware to connect the virtual Shelly or not.Without hardware, it't work it not to be.

In der Zendure App kann man den Shelly nicht hinzufügen. 

[image]

Milvus11, 2025-03-24:
Maybe another hint I found in this forum, regarding the phase mode of the shelly:

https://forum.zendure.com/d/4264/8

[image]

Could that be an issue?

lpetflo, 2025-03-24:
Changing the MAC didn't work for me. Is there a different way than changing the uni-meter.conf file? I still get the "Das Gerät gehört bereits einem anderen Account/Nutzer" message and the automatically detected shelly still shows the default MAC, not my custom one. Maybe that's the problem?  

```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.tibber-pulse"
  
  output-devices {
    shelly-pro3em {
	  port = 80
      udp-port = 1010
      mac = "B827EB12xxxx"
      hostname = "shellypro3em-b827eb12xxxx"
    }
  }

    input-devices {
    tibber-pulse {
      url = "http://192.168.xxx.xxx"
      node-id = 2
      user-id = "admin"
      password = "xxxx-xxxx"
    }
  }
}
```

Did I miss something?
Adding the device to the Zendure App wasn't successful either. However I got the hardware yesterday and will try it again tomorrow.

burpeenerd, 2025-03-24:
> Did I miss something? 

Did you modify the two avahi service files as per the documentation? https://github.com/sdeigm/uni-meter?tab=readme-ov-file#configuring-the-shelly-device-id

lpetflo, 2025-03-24:
> Did you modify the two avahi service files as per the documentation?

Oh no, I actually missed the second part of adding the MAC to the service files. :(
However after changing the MAC in both files (4 occurrences, I know) I still only find the Shelly with the default MAC and while it started adding the device now, it still failed with the same error message as before.

Milvus11, 2025-03-25:
> > Did you modify the two avahi service files as per the documentation?
> 
> Oh no, I actually missed the second part of adding the MAC to the service files. :( However after changing the MAC in both files (4 occurrences, I know) I still only find the Shelly with the default MAC and while it started adding the device now, it still failed with the same error message as before.
> 
> EDIT: Ok sorry, I managed to add the MAC to
> 
> ```
> uni-meter {
>   output = "uni-meter.output-devices.shelly-pro3em"
> 
>   input = "uni-meter.input-devices.tibber-pulse"
>   
>   output-devices {
>     shelly-pro3em {
>       mac = "B827EB12xxxx"
>       hostname = "shellypro3em-b827eb12xxxx"
>   }
> }
> ```
> 
> instead of
> 
> ```
> uni-meter {
>   output = "uni-meter.output-devices.shelly-pro3em"
> 
>   input = "uni-meter.input-devices.tibber-pulse"
>   
>   output-devices {
>     shelly-pro3em {
>       device {
>         mac = "B827EB12xxxx"
>         hostname = "shellypro3em-b827eb12xxxx"
>       }
>     }
>   }
> }
> ```
> 
> Now I was able to add the emulated Shelly to the Shelly App, but Zendure won't recognize it yet...

Hey, is your second code part anyhow different than the first? I am not able to add the shelly emulator to the app and would like to understand, what I do wrong. Maybe you have the hint with it.

lpetflo, 2025-03-25:
I noticed an additional mistake I did while initializing uni-meter. I wanted to use the docker based way of starting it and thus forgot to add avahi to the docker-compose.yaml
After adding avahi, I still don't see the uni-meter in the Zendure App, but at least I'm (sometimes) getting this error in the log:

`uni-meter  | 25-03-25 20:04:35.069 ERROR uni-meter.output         - failure while processing websocket notification: unhandled RPC method 'Shelly.GetComponents'`

In my naive view this looks like the Zendure App is trying to find a shelly device and the uni-meter isn't able to answer? However I can't reproduce this error consistently with searching again a second time in the app.

lpetflo, 2025-03-25:
@Milvus11 it seems to be important do follow the uni-meter/output-devices/shelly-pro3em/device/mac structure (I forgot to add the device keyword in my first attempt)

lpetflo, 2025-03-25:
As I'm not sure if that request actually comes from the Zendure System and I'm not to firm with networking topics I provide the log with added TRACE for uni-meter.websocket and uni-meter.input

[log.txt](https://github.com/user-attachments/files/19456632/log.txt)

.138 is the uni-meter device (a local vm on my NAS - connected via Ethernet, which might potentially be an issue, but changing this to a 2.4GHz device would be quite some hassle...)
.78 is the tibber device
192.168.178.99 seems to be the default IP set by @sdeigm ? Could it be, that the different IP-range 66 vs. 178 is a problem? The SSID "kabellos" isn't correct either, but I guess that's expected...

Milvus11, 2025-03-25:
> [@Milvus11](https://github.com/Milvus11) it seems to be important do follow the uni-meter/output-devices/shelly-pro3em/device/mac structure (I forgot to add the device keyword in my first attempt)

What structure is that? Where do I find it? My shelly app stops at 

[image]

sdeigm, 2025-03-25:
@Milvus11 I think your current problem is, that you configured an invalid MAC address. It should be a 12 character long hexadecimal string. That means it must only contain digits from 0 to 9 or letters from a to f. x is not allowed

lpetflo, 2025-03-25:
@Milvus11 the structure is in the [uni-meter.conf](https://github.com/sdeigm/uni-meter/blob/main/docker/uni-meter.conf) file.

Milvus11, 2025-03-25:
> [@Milvus11](https://github.com/Milvus11) I think your current problem is, that you configured an invalid MAC address. It should be a 12 character long hexadecimal string. That means it must only contain digits from 0 to 9 or letters from a to f. x is not allowed

A to F. That was my mistake. Thank you so much!

sdeigm, 2025-03-25:
@lpetflo I am also not sure, if the request are coming from the zendure device. But anyhow it is an interesting finding, that Shelly.GetComponents is called and not answered. I will add the method in the next version.

Most of the values uni-meter sends as answer in the request are configurable. There is a file [reference.con](https://github.com/sdeigm/uni-meter/blob/main/src/main/resources/reference.conf) which contains all default values used by uni-meter. You can override each of these values in your uni-meter.conf file using exactly the structure you can see in the reference.conf file.

So if you want to experiment with different settings, just override these values in your uni-meter.conf.

Regarding the websocket connection: unfortunately my debug trace does not contain the remote ip address anymore. I will also improve that in the next version. In the meantime you can perhaps use a tool like wireshark to check which device sends the request?

lpetflo, 2025-03-26:
I’ll take a closer look at who is trying to connect with whom, as well as whether I can identify any search requests coming from the Zendure system. 
Thanks for your efforts, and feel free to message me if you need any insights regarding Tibber/Zendure. :)

lpetflo, 2025-03-26:
I'm currently trying to setup uni-meter on my old Raspberry Pi (in desperate hope, that being connected to the same Wifi Network might help). While waiting for it to boot I took a look into Wireshark.

- Most of the time the communication is between the Zendure System and some kind of [Amazon AWS Device](https://ipinfo.io/18.195.156.44) in Frankfurt (possibly the Zendure Cloud, I guess?)

`18.195.156.44	192.168.66.135	TCP	54	1883 → 51952 [ACK] Seq=1 Ack=1 Win=65535 Len=0`

- When trying to connect a new shelly device through Profile/Shelly-Icon/AddDevice I get the following log:
68556	109.418390	18.195.156.44	192.168.66.135	MQTT	180	Publish Message [iot/gDa3tb/70jKB755/properties/read]

with the packet content being:
`����  0| #iot/gDa3tb/70jKB755/properties/read{"messageId":11,"deviceId":"70jKB755","timestamp":"1743015975","properties":["getAll"]}`

sdeigm, 2025-04-03:
@ChristianR82 from the trace I can again see that some device calls the RPC Method Shelly.GetComponents which is not yet implemented. I really should add that method soon. 

Can you figure out which device in your network has the IP address 192.168.188.80? Is that the Zendure device?

ChristianR82, 2025-04-03:
Hi sdeigm,
indeed, this is the setup:
192.168.188.31 - Input source: Fronius PV-Inverter with Modbus-connected 3 phase-power meter.
192.168.188.80 - RasperryPi with Homeassistant, running Uni-Meter (intended for productive use)
192.168.188.45 - Windows-PC with WSL, presently running another instance of uni-meter (for easier testing/analyzing)
192.168.188.52 - Zendure Hyper 2000 Battery Inveter with AB200S Battery
192.168.188.1   - Router

ChristianR82, 2025-04-11:
@sdeigm 
Thank you for adding the GetComponents rpc. The previous error is gone now. 
Unfortunately: 
- In the shelly app, as before, the uni-meter is found but then reported  to be offline.
- More important: in the Zendure App the uni-meter remains undetected.
However, there is somethinig suspicious in the log: 
`- ShellyPro3EM.rpcShellyGetStatus(): ShellyGetStatusResponse[wifi_sta=WiFiStatus[connected=true, ssid=kabellos, ip=192.168.178.99, rssi=-83]`
All my devices are in 192.168.1**7**8.xxx address range, including a physical Shelly 2pm at 192.168.1**8**8.99. But I have no idea how the 192.168.**178.99** may have come into the log message. 
Full log is attached.

[Log_uni-meter-1.1.5.txt](https://github.com/user-attachments/files/19714495/Log_uni-meter-1.1.5.txt)

sdeigm, 2025-04-12:
The wifi information provided by uni-meter is static but confiurable (all values from this file [reference.conf](https://github.com/sdeigm/uni-meter/blob/main/src/main/resources/reference.conf) can be used in your uni-meter.conf):

```
uni-meter {
  output-devices {
    shelly-pro3em {
      wifi-status {
        connected = true
        ssid = "kabellos"
        ip = "192.168.178.99"
        rssi = "-83"
      }
    }
  }
}
```

But I don't think the wifi information has any effect on the lookup of the device. The methods in your provided logfile are from an home assistant instance that automatically detected the uni-meter via mDNS (I can see this from the "src":"aios-546750005856"" RPC address which is used by home assistant). I think there are absolutely no calls from the zendure device.

If Zendure just gets the IP address of the shelly just from the cloud, I don't think there is any chance to solve this. It is not possible for an emulator to send up any information into the cloud (non-documented ssl interface) and therefore that status of the virtual shelly will always be "offline" there.

lpetflo, 2025-06-17:
Are there any updates regarding Zendure and the ZENKi implementation?

Derich367, 2025-08-08:
Just for your information:
First i tried to link my zendure system (solarflow 2400 ac) with this addone to a tasmota sensor, which was not working for me.

Then i tried to use the zendure ha integration (https://github.com/Zendure/Zendure-HA), which directly supports any home assistant entity as a smart meter. So you don't have to link anything into zendure app, it is just working directly in home assistant.

fatfox511, 2026-08-16:
> > That could work and is worth a try. If the Zendure system is totaly cloud-independent after the initial pairing, you can do the initial pairing using a real Shelly. Afterward make sure that the uni-meter gets exactly the same local IP Address as the real Shelly during the pairing process and configure the uni-meter's device id to the one of the original Shelly.
> 
> [@sdeigm](https://github.com/sdeigm) Ich habe den Versuch nun gemacht, hab mir einen Shelly pro 3EM gekauft und in Betrieb genommen, in der Shelly cloud registriert, in der Zendure App eingerichtet und hinzugefügt. Das hat funktioniert, ist ja auch so vorgesehen. Dann hab ich den Shelly ausgeschaltet und stattdessen mit derselben IP meinen Raspberry Pi mit der Uni- Meter Shelly Emulation drauf gestartet. Schwupps, der Shelly wurde in der Zendure App als nicht erreichbar und somit offline markiert. Den via Uni-meter simulierten Shelly hat das System hartnäckig ignoriert. Dieses Vorgehen funktioniert damit also leider auch nicht. Keine Ahnung ob es was gebracht hätte, vor der Shelly Ausserbetriebsetzung das Internet auszuschalten, den Zendure Speicher abzuschalten und dann erst die Veränderungen im Netzwerk durchzuführen - egal, war mir dann schon zu blöd.
> 
> Ich hab mich entschlossen die Dinge selbst in die Hand zu nehmen und hab daher in den letzten Wochen ein kleines Python-Projekt zur lokalen Steuerung des Zendure SolarFlow 2400 AC+ Speichers über MQTT gebaut um das Problem, den Zendure-Speicher mit meinem SMA Home Manager 2.0 und einem über Uni-Meter emulierten Shelly zu betreiben und dabei möglichst sauber auf Netzbezug bzw. Einspeisung zu reagieren. Das Script läuft bei mir auf einem Raspberry Pi und nutzt u. a. MQTT, eine Shelly-kompatible Messwertquelle, eine Weboberfläche, Live-Graphen, CSV-Logging und verschiedene Schutz-/Fallback-Mechanismen.
> 
> Grober Funktionsumfang:
> 
>     * automatische Lade-/Entladeregelung anhand der Netzleistung
> 
>     * komplett lokale Steuerung ohne Cloud, im Zendure System muss lediglich MQTT aktiviert werden
> 
>     * dadurch komplette Steuerung des Zendure über MQTT
> 
>     * Weboberfläche mit Status, Settings, großem Graph, CSV-Export und Diagnosefunktionen
> 
>     * manueller Modus für Stop/Hold, feste Ladung und feste Entladung
> 
>     * Nachtmodus mit konstanter Entladung
> 
>     * rollierendes CSV-Logging
> 
>     * Headless Mode
> 
>     * optionaler Guard für Berücksichtigung einer zweiten Batterie im System, bei mir z. B. der SMA Sunny Island über durch auslesen derer DAten über die bei mir zur e-Auto-Ladung eh schon vorhandene OpenSource-Software EVCC (auch über MQTT), um unerwünschtes Batterie-zu-Batterie-Laden zu vermeiden
> 
>     * umfangereiche Optionen, um das System an die eigenen Bedürfnisse anzupassen
> 
>     * ausführliches Benutzer-, Installations- und Administrationshandbuch
> 
> 
> Das Projekt löst bei mir insbesondere das Problem, dass der Zendure-Speicher nicht nur „blind“ lädt oder entlädt, sondern abhängig von der gemessenen Netzleistung geregelt wird. Zusätzlich wird berücksichtigt, dass in einem System mit zweitem Speicher besondere Regelungsprobleme auftreten können, die die Software sehr gut löst. Um einen Eindruck zu vermitteln wie das ganze aussieht hab ich ein Screenshots beigefügt.
> 
> Warum mach ich das? Weil ich so lang nach einer Lösung für die Kombi Zendure / SMA / zweite Batterie und das alles lokal gesucht habe, aber nichts gefunden hab und damit anderen wie mir einfach helfen will.
> 
> Das Ganze ist kein kommerzielles Produkt, sondern ein privates Projekt für technisch versierte Anwender. Nutzung natürlich auf eigenes Risiko.
> 
> Falls jemand Interesse am Script bzw. an der Dokumentation hat, schreibt mir gerne eine private Nachricht.
> <img alt="Image" width="1587" height="972" src="https://private-user-images.githubusercontent.com/111155318/592612978-b5d7760d-51bc-4c7b-a08d-02252a0c9f4f.jpg?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODY4OTQ1MTIsIm5iZiI6MTc4Njg5NDIxMiwicGF0aCI6Ii8xMTExNTUzMTgvNTkyNjEyOTc4LWI1ZDc3NjBkLTUxYmMtNGM3Yi1hMDhkLTAyMjUyYTBjOWY0Zi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwODE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDgxNlQxNTMwMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNzhmMjIyYzg2YWZmNjY4OTc4YWY2YWU5NGZkOTMyOTk2OTY2YzU1YzFmMTFkODk4N2I1NWRiM2I3YWZmYzQ0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZyZXNwb25zZS1jb250ZW50LXR5cGU9aW1hZ2UlMkZqcGVnIn0.XJlSGCv5NK8GAXX1Cvr8-Q0qGrb0jDEG-KWY7ljjV-I"> <img alt="Image" width="1582" height="713" src="https://private-user-images.githubusercontent.com/111155318/592613214-69994fd6-7f3c-40ce-b572-188843158b08.jpg?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODY4OTQ1MTIsIm5iZiI6MTc4Njg5NDIxMiwicGF0aCI6Ii8xMTExNTUzMTgvNTkyNjEzMjE0LTY5OTk0ZmQ2LTdmM2MtNDBjZS1iNTcyLTE4ODg0MzE1OGIwOC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwODE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDgxNlQxNTMwMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xOWYxZDZhMDdlNTNmZDEzY2YyMGI2ZGNjY2I0MGM4YzYwNTA0ZTRhODBlZWRjMDRlM2Q2YjNmOWQxZDUxZGE3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZyZXNwb25zZS1jb250ZW50LXR5cGU9aW1hZ2UlMkZqcGVnIn0.upxbiBXJNCBBCY9rb0TkGsqOgi2hrEh-IZA9peK7a5o">

Hey Eddy,

ich kann dich nicht privat anschreiben. Würdest du dieses Projekt mit mir teilen?

Beste Grüße

TTXaser2, 2026-08-21:
> 
> Hey Eddy,
> 
> ich kann dich nicht privat anschreiben. Würdest du dieses Projekt mit mir teilen?
> 
> Beste Grüße

Hi @fatfox511 ,

hab Dich auf mein Repo freigeschaltet, probier es gern aus. Software ist aktuell noch in Entwicklung, steuert aber bereits seit mehreren Monaten meinen SolarFlow 2400AC+ im perfekten Zusammenspiel mit meinem 13kW "Altanlagen" Speicher SMA Sunny Island. Repo findest Du hier: https://github.com/TTXaser2/zendure-energy-controller 

VG, Eddy

OneCreek, 2026-11-12:
@TTXaser2 same here. wäre Dir auch über eine Freischaltung dankbar :)

TTXaser2, 2026-11-15:
> [@TTXaser2](https://github.com/TTXaser2) same here. wäre Dir auch über eine Freischaltung dankbar :)

@OneCreek  Einladung ist raus, bitte gib mir gern Feedback ob der Zugriff klappt. Entwicklung läuft gerade noch, aber alle Releases sollten stabile Versionen sein die grundsätzlich laufen. Ich hab den Leistungsumfang in den letzten Monaten massiv erweitert, die Konfiguration ist recht umfangreich aber von der Hilfe her mittlerweile gut brauchbar. Wenn Du die Installation geschafft hast (Start des Installers mit der Option --fresh-install), gib mir gern über Discussions Feedback, auch wenn was nicht läuft.