sdeigm/uni-meter #70: Support JSON RPC via HTTP POST for Zendure support

sdeigm, 2025-03-15:


sdeigm, 2025-03-20:
All methods called by the Shelly App are now implemented. Anyhow, the final update in the Shelly cloud is not done since that interface is not documented and not traceable

closed 2025-03-20 by sdeigm

Milvus11, 2025-03-20:
Once again thank you for your efforts! I installed 1.1.2, it is found with the shelly app, but as you said, not implementable. Zendure App does not find it so far. I have the issue, that the tibber integration shows nothing in the .log file, but did not put it in trace mode yet, to find out more.

burpeenerd, 2025-03-20:
I have not (yet) bought a Zendure, due to this "issue", so I will keep watching this issue.
Currently, all my power readings are in mqtt, and I successfully ran uni-meter 1.1.3 docker (built today). All values are populated in /status endpoint. 
But unfortunately, in Shelly App, I only "see" the device, but without any values or the IP (exactly like in the screenshot from the other GitHub issue). I also tried Zendure App (without any physical Zendure) to try to find and connect to the virtual Shelly EM Pro, but without any success 😓