sdeigm/uni-meter #73: Persistent Outbound connection

MuellBin2, 2025-03-16:
Since Version 1.1.1 it is possible to bind devices to the outbound connection. However, Uni-Meter needs a new connection to be set up every time the software is restarted. In contrast, the binding stays active on the connected devices. So it would be possible, after a restart, to automatically start sending the values again.

In praxis, the connected device (APSystems EZHI) needs following steps to reconnect to Uni-Meter after the software has been restarted:
- delete the old connection
- reconnect to uni meter
- name the new connection
- do a phase detect, where several 800w pulses are sent into the grid (=energy loss)
- enable intelligent connection again 

Instead of an IP adress, the network name could be used to overcome changing IP adresses. The settings should be done in the config file.

sdeigm, 2025-03-17:
Yes, that is a good point. I will add a configuration parameter for the outbound websocket URL within the uni-meter.conf file.

closed 2025-03-18 by sdeigm

sdeigm, 2025-03-20:
You can now test the persistent outbound connection feature in the [uni-meter-1.1.3-SNAPSHOT](https://github-registry-files.githubusercontent.com/898342116/c38c7900-0549-11f0-974b-a74665e7488c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20250320%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250320T042031Z&X-Amz-Expires=300&X-Amz-Signature=f4b9383516628b514d5d8b31fe3645bbcb2373d9aa29f2405094d229cae896ae&X-Amz-SignedHeaders=host&response-content-disposition=filename%3Duni-meter-1.1.3-20250320.041049-1.tgz&response-content-type=application%2Foctet-stream) snapshot version which I have just build. A docker container sdeigm/uni-meter:1.1.3-SNAPSHOT is also available.

The outbound websocket connection is configured like below:

```
uni-meter {
  
  # ...
  
  output-devices {
    shelly-pro3em {
      # ...
	  ws = {
        enabled = true
        server = "ws://localhost:4242/websocket"
        ssl_ca = "*"
	  }
    }
  }
}
```

sdeigm, 2025-03-21:
Are you sure the Websocket server on the other side is running?

You can test this with this tool ```https://websocket.tech/```

MuellBin2, 2025-03-21:
Yes they did. First they realeased a new app that would only bind using shelly cloud, now the firmware for the apsystems followed that threw out the websocket i think.
I dont know if we can integrate uni-meter in the shelly cloud, but anyhow, if the connection is not possible locally in the end this is a bad design.

tmb80c, 2025-03-21:
> if the connection is not possible locally in the end this is a bad design.

agree

looks like neither Zendure, Anker nor APSystems provide (currently) the shelly EM local as we would like to have it. I have less hope for Anker that this will change, they even don't support the Shelly plugs. Anker's shelly support is more or less marketing. Hope APsystems will do better.