sdeigm/uni-meter #75: Integrate uni-meter as extension in homeassitant

meier-rene, 2025-03-17:
Hi currently I'm trying to integrate uni-meter as an extension to homeassitant, because I dont want to run an additional system. I forked your repo to https://github.com/meier-rene/uni-meter and already adjusted the Dockerfile to include all required build tools, because the environment in homeassitant is very limited.

At the moment I don't know if it is possible to include it and I want to ask: Are you aware of any attempts to include uni-meter in homeassitant? Is there a discussion somewhere?

Best, Rene

sdeigm, 2025-03-18:
Hi Rene, I am currently not aware of any attempts to include the uni-meter into homeassistant.

To be honest, I don't really know what it means to include the uni-meter into homeassistant and what has to be done therefore. But I am open to discuss this here and adjust the software if needed. 

Perhaps you can clarify what steps are necessary to include the uni-meter and what does that mean in the end?

Best regards
Stefan

meier-rene, 2025-03-18:
Because you have propably the best overview whats going on around your project I was just asking if I missed something and would redo work which already has been done.

About running uni-meter in homeassitant:
Because I already have a homeassitant system running and I need uni-meter to provide data for the hoymiles ms-a2, I thought it would be a good idea to run this all on one system.

The dataflow should be hichi IR -> MQTT Broker in HA -> uni-meter -> hoymiles

I dont know if this will work but I will find out. The homeassitant operating system itself is very limited. There is no java and no maven. There is only docker. Thats why all applications which should run on HA need to be packaged in a Dockerfile. I modified your Dockerfile to be able to build without any external toolchain required.

Now I need to write a extension description file for HA and I can install it as a tool in HA. I will provide a PR with the working solution if I manage to put things together.

sdeigm, 2025-03-18:
Ok, that sounds good. 

Just as info: the hoymiles storage expects the Shelly to run on port 80, this seems to be hardcoded. So if the homeassistant system itself uses that port, that might lead to a conflict. If that's the case you have for instance to setup Nginx to split the traffic between the homeassistant and uni-meter.

sdeigm, 2025-03-26:
Hi Rene,

just to inform you, I got three requests regarding a homeassistant addon in the last days. So out of curiosity I installed homeassistant myself and started to develop an addon myself. My current results are pushed in the branch linked to this issue. I am already able to start the unimeter within homeassistant. Now I am working on the configuration part and the ahavi daemon integration. The latter is a little bit tricky I guess - if at all possible. I think I have to use the mDNS of the homeassistant itself if possible.

Regards
Stefan

ChristianR82, 2025-03-27:
Hello,
making uni-meter runable in home assistant would be great. 
Concerning the point of getting the mDNS running: As I understand, it is needed to resolver a hostname into in ip-address. In case this is not possible with the mDNS service of home assistant OS, there could be an alternative:
The DNS-Server (Filter) 'Pi-hole' is available as an [home assistant add-on](https://www.home-assistant.io/integrations/pi_hole/). While in general, it is used in private networks for blocking-ads, it also allows to configure local hostnames and their corresponding IP addresses. I presently use this for the [YCast-Addon](https://github.com/casperklein/homeassistant-addons/tree/master/ycast).

sdeigm, 2025-03-28:
mDNS and DNS are not exactly the same. 

DNS is used to convert a known hostname into its IP address.

mDNS is used to automatically discover a service in the local network using UDP broadcasts. In our case here the uni-meter has to be announced, so that it is automatically discovered by the Hoymiles storage. All other storages do not use mDNS, so the addon should already be working.

sdeigm, 2025-03-28:
The mDNS stuff is solved, but I am currently struggeling to add https://github.com/sdeigm/uni-meter as repository. Adding the repository succeeds but it does not show up in the repository stores (even after I refreshed the browser window).

Any idea from somebody with more homeassistant experience then me?

thomluther, 2025-03-28:
> Any idea from somebody with more homeassistant experience then me?

Hi, you need to modify your `repository.yaml`
Add something like:
```yaml
addons:
  uni_meter:
    repository: ha_addon
```

sdeigm, 2025-03-28:
Thanks! I also tried this and removed it again since it does not seem to be necessary.

It is now working. My problem was that the image in the was not matching a certain regular expression. It has to be

``<image-registry>/<user>/<image>``

without a version. So for me it is

``docker.io/sdeigm/uni-meter-addon``

You can now install the addon, so if you want to test it feel free :-)

The mDNS stuff is solved like in this post [mDNS in Home Assistant](https://community.home-assistant.io/t/mosquitto-broker-announcing-advertising-mqtt-service-via-mdns/343190/14) using HACS and Pyscript integration. But mDNS is only necessary for the users with Hoymiles storages. All the others can live without it ...

sdeigm, 2025-03-28:
I just added a small documentation section regarding Home Assistant which has to be improved for sure ...

meier-rene, 2025-03-28:
Thanks a lot for all your effort. I could install the extension and start it. It looks ok from the logs:
```
[21:01:54] INFO: using custom configuration file: uni-meter.conf
25-03-28 21:01:56.131 INFO  uni-meter                - ##################################################################
25-03-28 21:01:56.146 INFO  uni-meter                - # Universal electric meter converter 1.1.3 (2025-03-21 18:55:40) #
25-03-28 21:01:56.147 INFO  uni-meter                - ##################################################################
25-03-28 21:01:56.148 INFO  uni-meter                - initializing actor system
25-03-28 21:01:57.644 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-03-28 21:01:58.459 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-03-28 21:01:58.496 INFO  uni-meter.controller     - creating MQTT input device
25-03-28 21:01:59.250 INFO  uni-meter.input          - subscribing to topic: tele/tasmota_F7F3B8/SENSOR
25-03-28 21:02:00.295 INFO  uni-meter.input          - MQTT stream connected
25-03-28 21:02:01.752 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
```
Nevertheless I can not connect to port 80:
```
curl http://192.168.178.100:80
The requested resource could not be found but may be available again in the future.
```
`docker ps` does not list any connected ports
```
docker ps                                     
CONTAINER ID   IMAGE                                                           COMMAND                  CREATED          STATUS          PORTS                                                                                                                          NAMES
93e931100e05   sdeigm/uni-meter-addon:1.1.3                                    "/init /run.sh"          35 minutes ago   Up 35 minutes                                                                                                                                  addon_663b81ce_uni_meter
```
although your config correctly lists the required ports:

[image]

Any obvious things I missed?

sdeigm, 2025-03-29:
Your curl command already talks to the uni-meter. I can see that from the returned message. I admit it is a little bit confusing and has to be improved. Normally a real Shelly would provide its web UI on the ``/`` path. Uni-meter does not return anything under this path and therefore your message from above. Please try

``http://192.168.178.100/rpc/EM.GetStatus?id=0`` and
``http://192.168.178.100/rpc/EMData.GetStatus?id=0``

and you will see the current meter readings.

meier-rene, 2025-03-29:
Do you mind to provide your pyscript for the mDNS? If I understand you correctly, you have a working solution. I try to set up the whole system in a remote place with vpn and I just can not scan the remote network. My local operator of the remote network has limited skills and it would be really helpful, if I would knew the mDNS is correct. 

Thats what I have atm:

```
from zeroconf import ServiceInfo, Zeroconf
import socket
import asyncio
from concurrent.futures import ThreadPoolExecutor
import logging

# Set up logging
logger = logging.getLogger("pyscript.mdns_entry")

def publish_mdns_service(service_type, service_name, port, txt_records):
    logger.debug(f"Publishing mDNS service: {service_name}.{service_type} on port {port}")
    
    service_info = ServiceInfo(
        service_type,
        f"{service_name}.{service_type}",
        addresses=[socket.inet_aton("192.168.178.100")],  # Replace with your device's IP address
        port=port,
        properties=txt_records,
        server=f"{service_name}.local.",
    )
    
    zeroconf = Zeroconf()
    zeroconf.register_service(service_info)
    logger.debug(f"Service {service_name}.{service_type} registered successfully")

async def announce_services():
    loop = asyncio.get_event_loop()
    executor = ThreadPoolExecutor()

    # Announce shelly service
    shelly_txt_records = {
        "gen": "2",
        "id": "shellypro3em-b827eb364242",
        "arch": "esp8266",
        "fw_id": "20241011-114455/1.4.4-g6d2a586"
    }
    await loop.run_in_executor(executor, publish_mdns_service, "_shelly._tcp.local.", "shellypro3em-b827eb364242", 80, shelly_txt_records)

    # Announce HTTP service
    http_txt_records = {
        "gen": "2",
        "id": "shellypro3em-b827eb364242",
        "arch": "esp8266",
        "fw_id": "20241011-114455/1.4.4-g6d2a586"
    }
    await loop.run_in_executor(executor, publish_mdns_service, "_http._tcp.local.", "shellypro3em-b827eb364242", 80, http_txt_records)

# Call the function to announce services
await announce_services()
```

sdeigm, 2025-03-29:
That's the pyscript:

```
import socket
from homeassistant.components import zeroconf
from zeroconf.asyncio import AsyncServiceInfo

@service
def register_service(type, name, ip, port):
    aiozc = zeroconf.async_get_async_instance(hass)
    info = AsyncServiceInfo(
        type_=type + "._tcp.local.",
        name=name + "." + type + "._tcp.local.",
        addresses=[socket.inet_aton(ip)],
        port=port,
        properties={}
    )
    aiozc.async_register_service(info) 
```

and somewhere I manually called the service with (I don't find that place anymore, I am Homeassistant beginner)

type=shelly
name=shellypro3em-b827eb365555 
ip=<homeassistant-ip>
port=80

The name has to be adjusted to the name you configured in uni-meter.conf (see Readme on  github) 

Then the uni-meter is announced via mDNS. I somehow have to integrate that service call info the uni-meter docker addon container.

Hope that helps for the start ....

Gitsarry, 2025-03-30:
Does this work also with Home Assistant running in a qemu KVM, hosted on Debian GNU/Linux 12 "Bookworm"? Any chance to temporarily to a workaround and e.g. change the Avahi daemon configuration manually via root SSH on the HA VM to get the Hoymiles MS-A2 discover the "Shelly" ? After it has been added by the Hoymiles Add, I guess, the original HA Avahi Daemon config can be restored?

closed 2025-04-01 by sdeigm