sdeigm/uni-meter #78: Feature Request: Solaredge meter / Modbus

tmb80c, 2025-03-18:
Would be great if the uni-meter can also read from solaredge meter in the near future.

Solaredge implemented the SunSpec Protocol.

Codes are published here, please search "SolarEdge Inverters - SunSpec Logging - Technical Note"

harvey637, 2025-03-18:
Hi, 
as reading the feature request and having a solaredge system also:
I read solaredge meter via modbus into my iobroker system and also a smartmeter/volkszähler via smartmeter.
Then I feed the smartmeter readings via http and iobroker api into this shelly emulator as documented in the examples.
This is working quite nice, but:
- there is an time intervall reading smartmeter (and solaredge)
- the value is instable/moving (flipping plus/minus with some larger spikes, e.g. switching on coffee machine)
So I did a small blockly script on my ioborker to flatten the spikes a bit an write the new value to a variable.
This variable is then fed into shelly emu via iobroker api request as written in documentation. 
This seems to work FOR ME !
Just my experience.

PS: possibly the smoothing of the value could also be done in shelly enu .... with variable tau ...
cu
Harvey

sdeigm, 2025-03-20:
There is now a SNAPSHOT version [uni-meter-1.1.3-SNAPSHOT](https://github-registry-files.githubusercontent.com/898342116/c38c7900-0549-11f0-974b-a74665e7488c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20250320%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250320T042031Z&X-Amz-Expires=300&X-Amz-Signature=f4b9383516628b514d5d8b31fe3645bbcb2373d9aa29f2405094d229cae896ae&X-Amz-SignedHeaders=host&response-content-disposition=filename%3Duni-meter-1.1.3-20250320.041049-1.tgz&response-content-type=application%2Foctet-stream) which provides initial support for Solaredge (just the power value). There is also a docker container available sdeigm/uni-meter:1.1.3-SNAPSHOT

To configure the device use the following uni-meter.conf:

```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.solaredge"
  
  input-devices {   
	solaredge {
	  address = "192.168.178.125"
	  port = 1502
	  unit-id = 0
	  power-register = 40083
	}  
  }  
}
```

The id of the power register is configurable. 40083 is most likely, but also 40084 might be a value depending on the configuration.

Initializing DEBUG output in the logback.xml file should give you hopefully debug output like

```
25-03-20 05:28:24.291 DEBUG uni-meter.input          - power 172.59 W (value=17259, scale=-2)
```

Everything is totaly untested, since I don't have a Solaredge device :-)

tmb80c, 2025-03-20:
Ok I tested the initial version. My Modbus is on port 502, that's just a config thing not an issue.

uni-meter connects to my modbus but can't read the power register if I understand the log correctly.

I tested both registers: 400083 and 40084 and see  the same error.

-03-20 15:14:29.886 INFO  uni-meter                - ###########################################################################
2025-03-20 15:14:29.887898+00:0025-03-20 15:14:29.887 INFO  uni-meter                - # Universal electric meter converter 1.1.3-SNAPSHOT (2025-03-20 05:10:50) #
2025-03-20 15:14:29.887910+00:0025-03-20 15:14:29.887 INFO  uni-meter                - ###########################################################################
2025-03-20 15:14:29.887921+00:0025-03-20 15:14:29.887 INFO  uni-meter                - initializing actor system
2025-03-20 15:14:30.174326+00:0025-03-20 15:14:30.174 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
2025-03-20 15:14:30.298288+00:0025-03-20 15:14:30.298 INFO  uni-meter.controller     - creating ShellyPro3EM output device
2025-03-20 15:14:30.309193+00:0025-03-20 15:14:30.307 INFO  uni-meter.controller     - creating Solaredge input device
2025-03-20 15:14:30.623952+00:0025-03-20 15:14:30.623 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
2025-03-20 15:14:30.874804+00:0025-03-20 15:14:30.874 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
2025-03-20 15:14:35.487381+00:0025-03-20 15:14:35.482 ERROR uni-meter.input          - failed to read input registers at address 40083, quantity 2
2025-03-20 15:14:35.487484+00:00java.util.concurrent.CompletionException: java.util.concurrent.TimeoutException: request timed out after 5000ms
2025-03-20 15:14:35.487514+00:00at java.base/java.util.concurrent.CompletableFuture.encodeThrowable(CompletableFuture.java:332)
2025-03-20 15:14:35.487568+00:00at java.base/java.util.concurrent.CompletableFuture.completeThrowable(CompletableFuture.java:347)
2025-03-20 15:14:35.487594+00:00at java.base/java.util.concurrent.CompletableFuture$UniApply.tryFire(CompletableFuture.java:636)
2025-03-20 15:14:35.487617+00:00at java.base/java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:510)
2025-03-20 15:14:35.487650+00:00at java.base/java.util.concurrent.CompletableFuture.completeExceptionally(CompletableFuture.java:2162)
2025-03-20 15:14:35.487674+00:00at com.digitalpetri.modbus.client.ModbusTcpClient.lambda$sendBufferAsync$3(ModbusTcpClient.java:126)
2025-03-20 15:14:35.487696+00:00at com.digitalpetri.modbus.TimeoutScheduler.lambda$fromScheduledExecutor$0(TimeoutScheduler.java:47)
2025-03-20 15:14:35.487728+00:00at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
2025-03-20 15:14:35.487752+00:00at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
2025-03-20 15:14:35.487774+00:00at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
2025-03-20 15:14:35.487806+00:00at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
2025-03-20 15:14:35.487828+00:00at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
2025-03-20 15:14:35.487850+00:00at java.base/java.lang.Thread.run(Thread.java:840)
2025-03-20 15:14:35.487884+00:00Caused by: java.util.concurrent.TimeoutException: request timed out after 5000ms
2025-03-20 15:14:35.487908+00:00at com.digitalpetri.modbus.client.ModbusTcpClient.lambda$sendBufferAsync$3(ModbusTcpClient.java:127)
2025-03-20 15:14:35.487931+00:00... 7 common frames omitted

sdeigm, 2025-03-20:
Ok, I am getting exactly the same error when I am accessing a Modbus register that does not exist on my SMD120 modbus device. Interestingly in the documentation of that device [SDM120](https://eastroneurope.com/images/uploads/products/protocol/SDM120-MODBUS_Protocol.pdf) the register also have numbers like 30013 for Power, but in the program code I have to use the documented number - 30001 (for whatever reason).

Can you also please check the number 83, 84 and 82?

tmb80c, 2025-03-20:
In my Homeassistant instance I'm using the Solaredge Modbus Integration: 

[WillCodeForCats](https://github.com/WillCodeForCats)
[solaredge-modbus-multi](https://github.com/WillCodeForCats/solaredge-modbus-multi)

May be this is helpful.

sdeigm, 2025-03-20:
I really meant the plain registers 82, 83 and 84 without the 40000. Can you please check them?

jfraf, 2025-03-20:
> I really meant the plain registers 82, 83 and 84 without the 40000. Can you please check them?

same error messages as with the ones in the 40000 range.

sdeigm, 2025-03-20:
Ok, can you please do another test? We can change the unit-id which is currently 0 to 1 and 2.

For every unit id please test the registers 82, 83, 84 and 40082, 40083 and 40084

```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.solaredge"
  
  input-devices {   
	solaredge {
	  address = "192.168.178.125"
	  port = 502
	  unit-id = 0
	  power-register = 40083
	}  
  }  
}
```

If that is not successful, I have to prepare another version which reads Holding registers instead of Input registers

jfraf, 2025-03-20:
> In my Homeassistant instance I'm using the Solaredge Modbus Integration:
> 
> [WillCodeForCats](https://github.com/WillCodeForCats) [solaredge-modbus-multi](https://github.com/WillCodeForCats/solaredge-modbus-multi)
> 
> May be this is helpful.

The SolarEdge implementation of openWB might also be helpful: https://github.com/snaptec/openWB/tree/master/packages/modules/devices/solaredge (especially inverter.py and meter.py). 

Also, I think that one of the meter-specific registers would be better suited (e.g., 40206) than the ones for the inverter which only give information about the solar power and do not consider the imported/exported power (if I got it right).

tmb80c, 2025-03-20:
This is the output of the new version above:

2025-03-20 21:40:27.929007+00:0025-03-20 21:40:27.927 INFO  uni-meter                - ###########################################################################
2025-03-20 21:40:27.929089+00:0025-03-20 21:40:27.928 INFO  uni-meter                - # Universal electric meter converter 1.1.3-SNAPSHOT (2025-03-20 21:57:37) #
2025-03-20 21:40:27.929136+00:0025-03-20 21:40:27.929 INFO  uni-meter                - ###########################################################################
2025-03-20 21:40:27.929187+00:0025-03-20 21:40:27.929 INFO  uni-meter                - initializing actor system
2025-03-20 21:40:28.224396+00:0025-03-20 21:40:28.224 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
2025-03-20 21:40:28.351919+00:0025-03-20 21:40:28.351 INFO  uni-meter.controller     - creating ShellyPro3EM output device
2025-03-20 21:40:28.359167+00:0025-03-20 21:40:28.358 INFO  uni-meter.controller     - creating Solaredge input device
2025-03-20 21:40:28.709175+00:0025-03-20 21:40:28.708 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
2025-03-20 21:40:28.934595+00:0025-03-20 21:40:28.934 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
2025-03-20 21:40:37.242059+00:0025-03-20 21:40:37.208 WARN  com.digitalpetri.modbus.client.ModbusTcpClient - No pending request for response frame: ModbusTcpFrame[header=MbapHeader[transactionId=331, protocolId=0, length=19, unitId=2], pdu=0310303030332e3235333700000000000000]
2025-03-20 21:40:40.526218+00:0025-03-20 21:40:40.524 WARN  com.digitalpetri.modbus.client.ModbusTcpClient - No pending request for response frame: ModbusTcpFrame[header=MbapHeader[transactionId=332, protocolId=0, length=83, unitId=2], pdu=0350006700320000000000000000fffe000000000000091009140909ffff000000001386fffe000000000000000000000000037b7eb80000ffff80000000ffff000000008000000080008000fffe00020000]
2025-03-20 21:40:43.246149+00:0025-03-20 21:40:43.241 WARN  com.digitalpetri.modbus.client.ModbusTcpClient - No pending request for response frame: ModbusTcpFrame[header=MbapHeader[transactionId=332, protocolId=0, length=83, unitId=2], pdu=0350006700320000000000000000fffe000000000000090d09140909ffff000000001386fffe000000000000000000000000037b7eb80000ffff80000000ffff000000008000000080008000fffe00020000]
2025-03-20 21:40:46.446768+00:0025-03-20 21:40:46.445 WARN  com.digitalpetri.modbus.client.ModbusTcpClient - No pending request for response frame: ModbusTcpFrame[header=MbapHeader[transactionId=332, protocolId=0, length=83, unitId=2], pdu=0350006700320000000000000000fffe00000000000009110918090cffff000000001386fffe000000000000000000000000037b7eb80000ffff80000000ffff000000008000000080008000fffe00020000]
2025-03-20 21:40:49.775908+00:0025-03-20 21:40:49.769 WARN  com.digitalpetri.modbus.client.ModbusTcpClient - No pending request for response frame: ModbusTcpFrame[header=MbapHeader[transactionId=334, protocolId=0, length=175, unitId=2], pdu=03ac00000000000100000003000000c800000001000000003f8000000000000000000000000000004248000042c8000042c8000042c800003f8000003f8000003f803336bf733336bf733336bf73fffe41ef000042b4000042beffff42d1000042dcfffe42effffec26f0000c22c00000000000000000000422cfffe426f00004000000042c80001000000000000333342480000423e0000437d800043430000436681db4139cccd4248cccd424

tmb80c, 2025-03-20:
what does unit-id mean? 

I have a Master inverter and a slave inverter and the slave inverted is connected to the meter.

When I increase unit-id to 2 then the log looks different:

2025-03-20 21:56:25.394593+00:0025-03-20 21:56:25.392 INFO  uni-meter                - ###########################################################################
2025-03-20 21:56:25.394733+00:0025-03-20 21:56:25.394 INFO  uni-meter                - # Universal electric meter converter 1.1.3-SNAPSHOT (2025-03-20 21:57:37) #
2025-03-20 21:56:25.394816+00:0025-03-20 21:56:25.394 INFO  uni-meter                - ###########################################################################
2025-03-20 21:56:25.394929+00:0025-03-20 21:56:25.394 INFO  uni-meter                - initializing actor system
2025-03-20 21:56:25.681619+00:0025-03-20 21:56:25.681 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
2025-03-20 21:56:25.803308+00:0025-03-20 21:56:25.803 INFO  uni-meter.controller     - creating ShellyPro3EM output device
2025-03-20 21:56:25.810551+00:0025-03-20 21:56:25.809 INFO  uni-meter.controller     - creating Solaredge input device
2025-03-20 21:56:26.128182+00:0025-03-20 21:56:26.127 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
2025-03-20 21:56:26.373602+00:0025-03-20 21:56:26.373 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
2025-03-20 21:56:30.969207+00:0025-03-20 21:56:30.967 ERROR uni-meter.input          - failed to read input registers at address 40210, quantity 1
2025-03-20 21:56:30.969259+00:00java.util.concurrent.CompletionException: java.util.concurrent.TimeoutException: request timed out after 5000ms
2025-03-20 21:56:30.969270+00:00at java.base/java.util.concurrent.CompletableFuture.encodeThrowable(CompletableFuture.java:332)
2025-03-20 21:56:30.969292+00:00at java.base/java.util.concurrent.CompletableFuture.completeThrowable(CompletableFuture.java:347)
2025-03-20 21:56:30.969300+00:00at java.base/java.util.concurrent.CompletableFuture$UniApply.tryFire(CompletableFuture.java:636)
2025-03-20 21:56:30.969307+00:00at java.base/java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:510)
2025-03-20 21:56:30.969319+00:00at java.base/java.util.concurrent.CompletableFuture.completeExceptionally(CompletableFuture.java:2162)
2025-03-20 21:56:30.969327+00:00at com.digitalpetri.modbus.client.ModbusTcpClient.lambda$sendBufferAsync$3(ModbusTcpClient.java:126)
2025-03-20 21:56:30.969334+00:00at com.digitalpetri.modbus.TimeoutScheduler.lambda$fromScheduledExecutor$0(TimeoutScheduler.java:47)
2025-03-20 21:56:30.969345+00:00at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
2025-03-20 21:56:30.969352+00:00at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
2025-03-20 21:56:30.969360+00:00at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
2025-03-20 21:56:30.969371+00:00at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
2025-03-20 21:56:30.969379+00:00at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
2025-03-20 21:56:30.969386+00:00at java.base/java.lang.Thread.run(Thread.java:840)
2025-03-20 21:56:30.969398+00:00Caused by: java.util.concurrent.TimeoutException: request timed out after 5000ms
2025-03-20 21:56:30.969406+00:00at com.digitalpetri.modbus.client.ModbusTcpClient.lambda$sendBufferAsync$3(ModbusTcpClient.java:127)
2025-03-20 21:56:30.969413+00:00... 7 common frames omitted
2025-03-20 21:56:32.125351+00:0025-03-20 21:56:32.124 ERROR uni-meter.input          - failed to read input registers at address 40210, quantity 1
2025-03-20 21:56:32.125456+00:00java.util.concurrent.CompletionException: com.digitalpetri.modbus.exceptions.ModbusResponseException: 0x03 [READ_HOLDING_REGISTERS] generated exception response 0x02 [ILLEGAL_DATA_ADDRESS]
2025-03-20 21:56:32.125487+00:00at java.base/java.util.concurrent.CompletableFuture.encodeThrowable(CompletableFuture.java:332)
2025-03-20 21:56:32.125524+00:00at java.base/java.util.concurrent.CompletableFuture.completeThrowable(CompletableFuture.java:347)
2025-03-20 21:56:32.125549+00:00at java.base/java.util.concurrent.CompletableFuture$UniApply.tryFire(CompletableFuture.java:636)
2025-03-20 21:56:32.125570+00:00at java.base/java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:510)
2025-03-20 21:56:32.125604+00:00at java.base/java.util.concurrent.CompletableFuture.completeExceptionally(CompletableFuture.java:2162)
2025-03-20 21:56:32.125627+00:00at com.digitalpetri.modbus.client.ModbusTcpClient.onFrameReceived(ModbusTcpClient.java:171)
2025-03-20 21:56:32.125649+00:00at com.digitalpetri.modbus.tcp.client.NettyTcpClientTransport$ModbusTcpFrameHandler.lambda$channelRead0$0(NettyTcpClientTransport.java:163)
2025-03-20 21:56:32.125683+00:00at com.digitalpetri.modbus.internal.util.ExecutionQueue$Task.run(ExecutionQueue.java:110)
2025-03-20 21:56:32.125706+00:00at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
2025-03-20 21:56:32.125729+00:00at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
2025-03-20 21:56:32.125761+00:00at java.base/java.lang.Thread.run(Thread.java:840)
[... cut ...] DING_REGISTERS] generated exception response 0x02 [ILLEGAL_DATA_ADDRESS]
2025-03-20 21:56:32.125807+00:00... 6 common frames omitted
2025-03-20 21:56:33.317247+00:0025-03-20 21:56:33.316 ERROR uni-meter.input          - failed to read input registers at address 40210, quantity 1
2025-03-20 21:56:33.317314+00:00java.util.concurrent.CompletionException: com.digitalpetri.modbus.exceptions.ModbusResponseException: 0x03 [READ_HOLDING_REGISTERS] generated exception response 0x02 [ILLEGAL_DATA_ADDRESS]

sdeigm, 2025-03-21:
> what does unit-id mean?

On the modbus each device has a unit-id. A register value on the modbus is addressed by the unit-id of the device and the address of the register. I think unit-id 1 is correct in your setup. In the second trace when you used unit-id 2 we are just getting timeout exceptions. There is no device with the unit-id 2 that sends an answer.

Regarding the first test run you did with the new version: I can only see these 

```
2025-03-20 21:40:46.446768+00:0025-03-20 21:40:46.445 WARN com.digitalpetri.modbus.client.ModbusTcpClient - No pending request for response frame: ModbusTcpFrame[header=MbapHeader[transactionId=332, protocolId=0, length=83, unitId=2], pdu=0350006700320000000000000000fffe00000000000009110918090cffff000000001386fffe000000000000000000000000037b7eb80000ffff80000000ffff000000008000000080008000fffe00020000]
```

warnings. This warning is printed out by the underlying modbus library, when it is not able to correlate a received answer to a send request. The correlation is done by the transaction id of the request.

But I don't see any error messages in your trace. So I am wondering if it perhaps already running with these settings. How does your logback configuration for that test looked like? Did you enable the DEBUG logging for the input device? If not can you please do another run with the following logback.xml settings:

```
<?xml version="1.0" encoding="UTF-8"?>
<Configuration status="WARN">
    <appender name="STDOUT" class="ch.qos.logback.core.ConsoleAppender">
        <encoder>
            <pattern>%d{yy-MM-dd HH:mm:ss.SSS} %-5level %-24logger - %msg%n</pattern>
        </encoder>
    </appender>

    <root level="INFO">
        <appender-ref ref="STDOUT" />
    </root>

    <logger name="uni-meter.input"  level="DEBUG" additivity="false">
        <appender-ref ref="STDOUT" />
    </logger>
</Configuration>
```

From the testing until now I would say, that unit-it 1 and power-register 40206 or 206 and power-scale-register 40210 or 210 are the settings to use.

jfraf, 2025-03-21:
> So I prepared a new version which now read HoldingRegisters. There are now two registers to configure, the power-register and the power-scale-register:

Looks promising so far! I will check if the values are consistant with the other apps that use the meter's data.

```
 ⠿ Container uni-meter  Created                                                                                                                               0.0s
Attaching to uni-meter
uni-meter  | Process 9 died: No such process; trying to remove PID file. (/run/avahi-daemon//pid)
uni-meter  | 25-03-21 05:50:36.411 INFO  uni-meter                - ###########################################################################
uni-meter  | 25-03-21 05:50:36.412 INFO  uni-meter                - # Universal electric meter converter 1.1.3-SNAPSHOT (2025-03-20 21:57:37) #
uni-meter  | 25-03-21 05:50:36.412 INFO  uni-meter                - ###########################################################################
uni-meter  | 25-03-21 05:50:36.412 INFO  uni-meter                - initializing actor system
uni-meter  | 25-03-21 05:50:36.700 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
uni-meter  | 25-03-21 05:50:36.826 INFO  uni-meter.controller     - creating ShellyPro3EM output device
uni-meter  | 25-03-21 05:50:36.833 INFO  uni-meter.controller     - creating Solaredge input device
uni-meter  | 25-03-21 05:50:36.845 DEBUG io.netty.util.internal.logging.InternalLoggerFactory - Using SLF4J as the default logging framework
uni-meter  | 25-03-21 05:50:36.846 DEBUG io.netty.channel.MultithreadEventLoopGroup - -Dio.netty.eventLoopThreads: 8
uni-meter  | 25-03-21 05:50:36.852 DEBUG io.netty.util.concurrent.GlobalEventExecutor - -Dio.netty.globalEventExecutor.quietPeriodSeconds: 1
uni-meter  | 25-03-21 05:50:36.856 DEBUG io.netty.util.internal.InternalThreadLocalMap - -Dio.netty.threadLocalMap.stringBuilder.initialSize: 1024
uni-meter  | 25-03-21 05:50:36.856 DEBUG io.netty.util.internal.InternalThreadLocalMap - -Dio.netty.threadLocalMap.stringBuilder.maxSize: 4096
uni-meter  | 25-03-21 05:50:36.891 DEBUG io.netty.util.internal.PlatformDependent0 - -Dio.netty.noUnsafe: false
uni-meter  | 25-03-21 05:50:36.891 DEBUG io.netty.util.internal.PlatformDependent0 - Java version: 17
uni-meter  | 25-03-21 05:50:36.892 DEBUG io.netty.util.internal.PlatformDependent0 - sun.misc.Unsafe.theUnsafe: available
uni-meter  | 25-03-21 05:50:36.892 DEBUG io.netty.util.internal.PlatformDependent0 - sun.misc.Unsafe base methods: all available
uni-meter  | 25-03-21 05:50:36.892 DEBUG io.netty.util.internal.PlatformDependent0 - sun.misc.Unsafe.storeFence: available
uni-meter  | 25-03-21 05:50:36.892 DEBUG io.netty.util.internal.PlatformDependent0 - java.nio.Buffer.address: available
uni-meter  | 25-03-21 05:50:36.893 DEBUG io.netty.util.internal.PlatformDependent0 - direct buffer constructor: unavailable: Reflective setAccessible(true) disabled
uni-meter  | 25-03-21 05:50:36.893 DEBUG io.netty.util.internal.PlatformDependent0 - java.nio.Bits.unaligned: available, true
uni-meter  | 25-03-21 05:50:36.894 DEBUG io.netty.util.internal.PlatformDependent0 - jdk.internal.misc.Unsafe.allocateUninitializedArray(int): unavailable: class io.netty.util.internal.PlatformDependent0$7 cannot access class jdk.internal.misc.Unsafe (in module java.base) because module java.base does not export jdk.internal.misc to unnamed module @1e96a13e
uni-meter  | 25-03-21 05:50:36.895 DEBUG io.netty.util.internal.PlatformDependent0 - java.nio.DirectByteBuffer.<init>(long, {int,long}): unavailable
uni-meter  | 25-03-21 05:50:36.895 DEBUG io.netty.util.internal.PlatformDependent - sun.misc.Unsafe: available
uni-meter  | 25-03-21 05:50:36.895 DEBUG io.netty.util.internal.PlatformDependent - -Dio.netty.tmpdir: /tmp (java.io.tmpdir)
uni-meter  | 25-03-21 05:50:36.895 DEBUG io.netty.util.internal.PlatformDependent - -Dio.netty.bitMode: 64 (sun.arch.data.model)
uni-meter  | 25-03-21 05:50:36.895 DEBUG io.netty.util.internal.PlatformDependent - -Dio.netty.maxDirectMemory: -1 bytes
uni-meter  | 25-03-21 05:50:36.895 DEBUG io.netty.util.internal.PlatformDependent - -Dio.netty.uninitializedArrayAllocationThreshold: -1
uni-meter  | 25-03-21 05:50:36.896 DEBUG io.netty.util.internal.CleanerJava9 - java.nio.ByteBuffer.cleaner(): available
uni-meter  | 25-03-21 05:50:36.896 DEBUG io.netty.util.internal.PlatformDependent - -Dio.netty.noPreferDirect: false
uni-meter  | 25-03-21 05:50:36.897 DEBUG io.netty.channel.nio.NioEventLoop - -Dio.netty.noKeySetOptimization: false
uni-meter  | 25-03-21 05:50:36.897 DEBUG io.netty.channel.nio.NioEventLoop - -Dio.netty.selectorAutoRebuildThreshold: 512
uni-meter  | 25-03-21 05:50:36.901 DEBUG io.netty.util.internal.PlatformDependent - org.jctools-core.MpscChunkedArrayQueue: available
uni-meter  | 25-03-21 05:50:36.929 DEBUG com.digitalpetri.modbus.client.ChannelFsm - S(NotConnected)          x E(Connect)               = S'(Connecting)          
uni-meter  | 25-03-21 05:50:36.938 DEBUG com.digitalpetri.modbus.tcp.client.NettyTcpClientTransport - onStateTransition: NotConnected -> Connecting via Connect
uni-meter  | 25-03-21 05:50:36.951 DEBUG io.netty.channel.DefaultChannelId - -Dio.netty.processId: 11 (auto-detected)
uni-meter  | 25-03-21 05:50:36.959 DEBUG io.netty.util.NetUtil    - -Djava.net.preferIPv4Stack: false
uni-meter  | 25-03-21 05:50:36.959 DEBUG io.netty.util.NetUtil    - -Djava.net.preferIPv6Addresses: false
[... cut ...]
uni-meter  | 25-03-21 05:50:39.357 DEBUG uni-meter.output         - power phase 0: PowerData[power=50.33, apparentPower=50.33, powerFactor=1.0, current=0.22, voltage=230.0, frequency=50.0]
uni-meter  | 25-03-21 05:50:39.358 DEBUG uni-meter.output         - power phase 1: PowerData[power=50.33, apparentPower=50.33, powerFactor=1.0, current=0.22, voltage=230.0, frequency=50.0]
uni-meter  | 25-03-21 05:50:39.358 DEBUG uni-meter.output         - power phase 2: PowerData[power=50.33, apparentPower=50.33, powerFactor=1.0, current=0.22, voltage=230.0, frequency=50.0]
uni-meter  | 25-03-21 05:50:40.448 DEBUG uni-meter.input          - power scale 0.0
```

sdeigm, 2025-03-21:
Ok, that looks good! Which settings did you use? uni-id = 1, 40206 and 40210?

If the values are ok, I will try to figure out the other registers so that we also get the voltage, frequence, data per phase etc.

jfraf, 2025-03-21:
This is my config:

```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.solaredge"
  
  input-devices {   
	solaredge {
	  address = "192.168.178.88"
	  port = 502
	  unit-id = 1
          power-register = 40206   # 206
          power-scale-register = 40210   # 210
	}  
  }  
}
```

And here, for comparison, the GUI of my wallbox. The values do not match exactly since the values displayed are only updated every 10s.

[image]

jfraf, 2025-03-21:
> If the values are ok, I will try to figure out the other registers so that we also get the voltage, frequency, data per phase etc.

Thank you for your great work! I will gladly do some more testing if you have any updates in the code.

As a side note, just in case it is not already obvious: positive values mean that power is fed into the grid while negative numbers mean an import of power.

tmb80c, 2025-03-21:
> But I don't see any error messages in your trace. So I am wondering if it perhaps already running with these settings. How does your logback configuration for that test looked like? Did you enable [...]

Wow - you are the man. I enabled debug and set unit-id = 1 and my output is now similar to @jfraf:

Application Name:uni-meterContainer ID:3042eed18b8761789a279cf95eb75cd860fa6c92ac4079ee717731147a05946f
2025-03-21 07:54:51.424481+00:0025-03-21 07:54:51.421 INFO  uni-meter                - ###########################################################################
2025-03-21 07:54:51.424519+00:0025-03-21 07:54:51.423 INFO  uni-meter                - # Universal electric meter converter 1.1.3-SNAPSHOT (2025-03-20 21:57:37) #
2025-03-21 07:54:51.424527+00:0025-03-21 07:54:51.423 INFO  uni-meter                - ###########################################################################
2025-03-21 07:54:51.424537+00:0025-03-21 07:54:51.423 INFO  uni-meter                - initializing actor system
2025-03-21 07:54:51.705074+00:0025-03-21 07:54:51.704 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
2025-03-21 07:54:51.829241+00:0025-03-21 07:54:51.829 INFO  uni-meter.controller     - creating ShellyPro3EM output device
2025-03-21 07:54:51.837809+00:0025-03-21 07:54:51.835 INFO  uni-meter.controller     - creating Solaredge input device
2025-03-21 07:54:52.133695+00:0025-03-21 07:54:52.133 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:54:52.209452+00:0025-03-21 07:54:52.209 DEBUG uni-meter.input          - power 565.8000000000001 W (value=5658, scale=-1.0)
2025-03-21 07:54:52.222347+00:0025-03-21 07:54:52.222 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
2025-03-21 07:54:52.473244+00:0025-03-21 07:54:52.473 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
2025-03-21 07:54:53.273694+00:0025-03-21 07:54:53.273 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:54:53.331511+00:0025-03-21 07:54:53.330 DEBUG uni-meter.input          - power 621.1 W (value=6211, scale=-1.0)
2025-03-21 07:54:54.401387+00:0025-03-21 07:54:54.400 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:54:54.453518+00:0025-03-21 07:54:54.452 DEBUG uni-meter.input          - power 623.7 W (value=6237, scale=-1.0)
2025-03-21 07:54:55.757262+00:0025-03-21 07:54:55.756 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:54:55.803228+00:0025-03-21 07:54:55.802 DEBUG uni-meter.input          - power 623.1 W (value=6231, scale=-1.0)
2025-03-21 07:54:56.915636+00:0025-03-21 07:54:56.914 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:54:56.964229+00:0025-03-21 07:54:56.963 DEBUG uni-meter.input          - power 626.0 W (value=6260, scale=-1.0)
2025-03-21 07:54:58.045836+00:0025-03-21 07:54:58.045 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:54:58.065317+00:0025-03-21 07:54:58.064 DEBUG uni-meter.input          - power 619.6 W (value=6196, scale=-1.0)
2025-03-21 07:54:59.148351+00:0025-03-21 07:54:59.147 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:54:59.196982+00:0025-03-21 07:54:59.196 DEBUG uni-meter.input          - power 615.1 W (value=6151, scale=-1.0)
2025-03-21 07:55:00.248252+00:0025-03-21 07:55:00.247 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:55:00.277302+00:0025-03-21 07:55:00.276 DEBUG uni-meter.input          - power 612.7 W (value=6127, scale=-1.0)
2025-03-21 07:55:01.328166+00:0025-03-21 07:55:01.327 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:55:01.347944+00:0025-03-21 07:55:01.347 DEBUG uni-meter.input          - power 602.6 W (value=6026, scale=-1.0)
2025-03-21 07:55:02.431314+00:0025-03-21 07:55:02.430 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:55:02.478479+00:0025-03-21 07:55:02.477 DEBUG uni-meter.input          - power 590.7 W (value=5907, scale=-1.0)
2025-03-21 07:55:03.532860+00:0025-03-21 07:55:03.532 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:55:03.561995+00:0025-03-21 07:55:03.561 DEBUG uni-meter.input          - power 626.9000000000001 W (value=6269, scale=-1.0)
2025-03-21 07:55:04.672478+00:0025-03-21 07:55:04.671 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:55:04.731477+00:0025-03-21 07:55:04.730 DEBUG uni-meter.input          - power 631.7 W (value=6317, scale=-1.0)
2025-03-21 07:55:05.771982+00:0025-03-21 07:55:05.771 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:55:05.801067+00:0025-03-21 07:55:05.800 DEBUG uni-meter.input          - power 634.7 W (value=6347, scale=-1.0)
2025-03-21 07:55:06.903126+00:0025-03-21 07:55:06.902 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:55:06.977110+00:0025-03-21 07:55:06.976 DEBUG uni-meter.input          - power 638.9000000000001 W (value=6389, scale=-1.0)
2025-03-21 07:55:08.034783+00:0025-03-21 07:55:08.034 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:55:08.123432+00:0025-03-21 07:55:08.123 DEBUG uni-meter.input          - power 631.1 W (value=6311, scale=-1.0)
2025-03-21 07:55:09.000751+00:0025-03-21 07:55:08.987 WARN  com.digitalpetri.modbus.client.ModbusTcpClient - No pending request for response frame: ModbusTcpFrame[header=MbapHeader[transactionId=23671, protocolId=0, length=83, unitId=2], pdu=03500067003204e601a101a201a2fffe0f8b0f8a0f9408fb08f608e8ffff7026ffff1386fffe70b2ffff9110fffed922fffe037b876800000f3afffd [... cut ...]
2025-03-21 07:55:12.750611+00:0025-03-21 07:55:12.749 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:55:13.109181+00:0025-03-21 07:55:13.108 DEBUG uni-meter.input          - power 616.3000000000001 W (value=6163, scale=-1.0)
2025-03-21 07:55:14.210995+00:0025-03-21 07:55:14.210 DEBUG uni-meter.input          - power scale -1.0
2025-03-21 07:55:14.250116+00:0025-03-21 07:55:14.249 DEBUG uni-meter.input          - power 638.4000000000001 W (value=6384, scale=-1.0)
2025-03-21 07:55:15.532007+00:0025-03-21 07:55:15.531 DEBUG uni-meter.input          - power scale -1.0

sdeigm, 2025-03-21:
Ok, I once again updated the 1.1.3-SNAPSHOT github package and the container on docker hub. This version now reads all meter data for all phases. You now just have to configure the unit-id. The power-register and power-register-scale configuration parameters have been removed. In theory there is a base-register-address configuration value which defaults to 40190 but you should not modify that.

In the debug output you should see a summary like:

```
25-03-21 09:11:40.309 DEBUG uni-meter.input          - current: 0.0
25-03-21 09:11:40.309 DEBUG uni-meter.input          - voltage: 0.0
25-03-21 09:11:40.309 DEBUG uni-meter.input          - frequency: 0.0
25-03-21 09:11:40.309 DEBUG uni-meter.input          - real power: 0.0
25-03-21 09:11:40.309 DEBUG uni-meter.input          - apparent power: 0.0
25-03-21 09:11:40.309 DEBUG uni-meter.input          - power factor: 0.0
25-03-21 09:11:40.309 DEBUG uni-meter.input          - real energy exported: 0.0
25-03-21 09:11:40.309 DEBUG uni-meter.input          - real energy imported: 0.0
```

Internally the data of the single phases is used. Since I can not test it with the real device there are most likely still some bugs in the usage of the values. Especially the scale of the energy values might be wrong (Wh or kWh)

For the voltage values I am not sure which one to use. Currently I am using the "Line to Neural" voltage value. But there are also "Line to Line" voltage values which I think are irrelevant for us. But I can correct this, if my assumption is wrong :-)

sdeigm, 2025-03-21:
Just forgot to mention: you can check the values for the different phases using your browser and the URL

```http://<uni-meter-ip>/rpc/EM.GetStatus?id=0```

tmb80c, 2025-03-21:
updated docker:

2025-03-21 08:46:43.740245+00:0025-03-21 08:46:43.739 DEBUG uni-meter.input          - current: -19.71
2025-03-21 08:46:43.740348+00:0025-03-21 08:46:43.739 DEBUG uni-meter.input          - voltage: 230.6
2025-03-21 08:46:43.740384+00:0025-03-21 08:46:43.739 DEBUG uni-meter.input          - frequency: 49.98
2025-03-21 08:46:43.740411+00:0025-03-21 08:46:43.739 DEBUG uni-meter.input          - real power: 4561.0
2025-03-21 08:46:43.740438+00:0025-03-21 08:46:43.739 DEBUG uni-meter.input          - apparent power: 4562.0
2025-03-21 08:46:43.740464+00:0025-03-21 08:46:43.739 DEBUG uni-meter.input          - power factor: -99.45
2025-03-21 08:46:43.740491+00:0025-03-21 08:46:43.740 DEBUG uni-meter.input          - real energy exported: 831663.0
2025-03-21 08:46:43.740517+00:0025-03-21 08:46:43.740 DEBUG uni-meter.input          - real energy imported: 423128.4

URL-response:

{"id":0,"a_current":-10.35,"a_voltage":230.8,"a_act_power":2385,"a_aprt_power":2386,"a_pf":-99.58,"a_freq":49.98,"b_current":-4.76,"b_voltage":229.9,"b_act_power":1087,"b_aprt_power":1087,"b_pf":-99.12,"b_freq":49.98,"c_current":-3.54,"c_voltage":229.7,"c_act_power":820,"c_aprt_power":820,"c_pf":-99.38,"c_freq":49.98,"total_current":-18.65,"total_act_power":4292,"total_aprt_power":4293}

looks good.

In the shelly app I can't see anything accept that the uni-meter is registered but doesn't report a connection. I think this is by design as you wrote in the other thread. I still can adapt the uni-meter in the Anker app but it reports no connection, obviously.

sdeigm, 2025-03-21:
The power factory is not "Shelly-like", it is normally in the range from 0 to 1, here it seems to be 0 to 100% and its negative. 

Can you please also check the energy counters:

```http://<uni-meter-ip>/rpc/EMData.GetStatus?id=0```

tmb80c, 2025-03-21:
> http://<uni-meter-ip>/rpc/EMData.GetStatus?id=0

{"id":0,"a_total_act_energy":106.85,"a_total_act_ret_energy":485.42,"b_total_act_energy":1527.33,"b_total_act_ret_energy":753.87,"c_total_act_energy":1532.38,"c_total_act_ret_energy":1480.94,"total_act":3166.56,"total_act_ret":2720.23}

just noticed, when I start the uni-meter docker then my second inverter (slave) connected to the SolarEdge meter disconnects from Homeassistant. The master inverter is not affected. If uni-meter and the homeassistant SolarEdge modbus can't run in parallel then unfortunately I can't use uni-meter (for my Anker).

harvey637, 2025-03-21:
Hi, only one slave is allowed by solaredge I guess.
My workflow is, that the readings of the solaredge-meter is read by iobroker via modbus adapter. So one clients reads all the data.
Than I fetch the contents of these variables via the api of iobroker, not conflict with reading the solaredge modbus.

My findings: the solaredge readings are a bit jumpy, also the three phases are not so helpful (for me). In the end I read the smartmeter, which is the "offical" meter for my power to/from the grid. But even this data is jumpy if accidantly measured at the moment of a small spike. So even this data is smoothed like:
calculate delta last and new data, delta * tau (I use 0.1 or 0.15)), old_value plus delta as new data.
Make a smooth curve :-)

sdeigm, 2025-03-21:
I am closing this ticket now for version 1.1.3. But feel free to add further comments or create a new ticket if necessary.

closed 2025-03-21 by sdeigm

jfraf, 2025-03-21:
> just noticed, when I start the uni-meter docker then my second inverter (slave) connected to the SolarEdge meter disconnects from Homeassistant. The master inverter is not affected. If uni-meter and [...]

You could use tcp-multiplexer as I wrote in https://github.com/sdeigm/uni-meter/issues/78#issuecomment-2741534672 to allow multiple devices simultaneous access to the SolarEdge meter. I have this running smoothly for a couple of weeks now.

jfraf, 2025-05-14:
My Marstek Venus E finally arrived, so I could finally test uni-meter's Solaredge implementation in a real-life scenario.

However, I noticed that the Marstek considers exported power as imported an vice versa (uni-meter v.1.1.9). 
When I changed src/main/java/com/deigmueller/uni_meter/input/device/modbus/solaredge/Solaredge.java so that the realPower and apparentPower values (lines 160ff.) were multiplied by -1, it works correctly.

I cannot rule out a misconfiguration of my Solaredge setup (however, openWB and SE inverter seem to read/interpret the values correctly), but maybe it is a general issue with uni-meter's SE implementation. I guess it would be helpful if a "flip the sign" option were available for the SE input device in uni-meter's configuration.

sdeigm, 2025-05-14:
Ok, thanks for that information. I will check the definition of the registers in the Solaredge documentation and adding such a negation parameter might be a good idea. In the meanwhile I think you can help yourself by specifying the following parameter (which is not intended for that purpose but can be misused):

```
uni-meter {
    output-devices {
      shelly-pro3em {
        default-client-power-factor = -1.0
     }
  }
}
```

sdeigm, 2025-05-14:
One more question you are saying the openWB is interpreting the values correctly. Are you using the Solaredge device as EVU module or PV?

jfraf, 2025-05-14:
> Are you using the Solaredge device as EVU module or PV?

Both (in openWB v.1.9, which I am still on; might have changed in v2)

sdeigm, 2025-05-16:
@jfraf, there is now a snapshot version available for testing. See this [comment](https://github.com/sdeigm/uni-meter/issues/151#issuecomment-2885629333)

hendrikmuhs, 2025-05-16:
> > just noticed, when I start the uni-meter docker then my second inverter (slave) connected to the SolarEdge meter disconnects from Homeassistant. The master inverter is not affected. If uni-meter and the homeassistant SolarEdge modbus can't run in parallel then unfortunately I can't use uni-meter (for my Anker).
> 
> You could use tcp-multiplexer as I wrote in [#78 (comment)](https://github.com/sdeigm/uni-meter/issues/78#issuecomment-2741534672) to allow multiple devices simultaneous access to the SolarEdge meter. I have this running smoothly for a couple of weeks now.

FYI for anyone finding this issue: There is also a small python tool called "modbus proxy". Works probably the same way: connect modbus-proxy to SE, connect all clients to the proxy. I have 3 clients: HA, uni-meter and the heat-pump connector.