sdeigm/uni-meter #79: Integration Problem with Homilies MS A2

klausmuehle, 2025-03-21:
Hi Stefan,
first of all: great SW -- I wanted to integrate a battery system with my existing PV system and found and purchased a Hoymiles MS A2. As I already have an intelligent power meter with IR readout, uni-meter ist exactly the missing link for such an integration!

i have integrated uni-meter on a Raspi system and it seems to be running fine (according to local log). However the problem is that the Hoymiles MS A2 does not find / integrate the uni-meter device. 

When I try to add the device manually (by IP address), I see that the uni-meter emulation receives two rpc requests from Hoymiles, one rpcGetDeviceInfo,  then an em.getconfig, which both seem to be replied. However, after the second rpc, there is no more activity and the App shows (after timeout) an error message ("Gerät nicht gefunden, Überprüfen Sie.....". Autodetection does not show any devices.

I am using latest version of Hoymiles FW and S-Miles Home App (on IOS). 

Below attached is the log of the system start and the two rpc calls -- I already added some trace information to the log, so it is not the 100% original 1.1.2 tracing.
 
Do you have any idea / hint where the problem might be?

Kind regards
Klaus Mühle

pi@pi4:~/Zaehler/uni-meter-main $ sudo /opt/uni-meter/bin/uni-meter.sh 
25-03-21 18:36:31.018 INFO  uni-meter                - ######################################################################
25-03-21 18:36:31.030 INFO  uni-meter                - # Universal electric meter converter 1.1.2-UDP (2025-03-21 18:35:33) #
25-03-21 18:36:31.031 INFO  uni-meter                - ######################################################################
25-03-21 18:36:31.031 INFO  uni-meter                - initializing actor system
25-03-21 18:36:32.714 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-03-21 18:36:33.455 TRACE uni-meter.controller     - UniMeter.createHttpServerController()
25-03-21 18:36:33.465 TRACE uni-meter.controller     - UniMeter.createOutput()
25-03-21 18:36:33.468 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-03-21 18:36:33.504 TRACE uni-meter.controller     - UniMeter.createInput()
25-03-21 18:36:33.505 INFO  uni-meter.controller     - creating GenericHttp input device
25-03-21 18:36:34.307 TRACE uni-meter.input          - GenericHttp.initChannelReaders()
25-03-21 18:36:34.313 TRACE uni-meter.input          - GenericHttp.executeHttpPolling()
25-03-21 18:36:34.547 TRACE uni-meter.output         - Shelly.startUdpServer()
25-03-21 18:36:34.547 TRACE uni-meter.output         - ShellyPro3EM.startOutboundWebsocketConnection()
25-03-21 18:36:34.546 TRACE uni-meter.controller     - UniMeter.onRegisterHttpRoute()
25-03-21 18:36:34.572 TRACE uni-meter.http.port-80   - HttpServer.onRegisterRoute()
25-03-21 18:36:34.573 TRACE uni-meter.http.port-80   - HttpServer.start()
25-03-21 18:36:36.353 TRACE uni-meter.http.port-80   - HttpServer.onNotifyBindSuccess()
25-03-21 18:36:36.354 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-03-21 18:36:38.807 TRACE uni-meter.output.http    - POST-1 RpcRequest: {"id":25,"method":"Shelly.GetDeviceInfo"}

25-03-21 18:36:39.062 TRACE uni-meter.output         - ShellyPro3EM.onRpcRequest()
25-03-21 18:36:39.064 TRACE uni-meter.output         - createRpcResult shelly.getdeviceinfo
25-03-21 18:36:39.065 TRACE uni-meter.output         - Shelly.rpcGetDeviceInfo()
25-03-21 18:36:39.066 TRACE uni-meter.output         - ShellyPro3EM.rpcGetDeviceInfo(): {"id":"shellypro3em-041812801251","mac":"041812801251","slot":1,"model":"SPEM-003CEBEU","gen":2,"fw_id":"20241011-114455/1.4.4-g6d2a586","ver":"1.4.4","app":"Pro3EM","auth_en":false,"auth_domain":null,"profile":"triphase"}
25-03-21 18:36:39.130 TRACE uni-meter.output         - createRpcResult ret {"id":"shellypro3em-041812801251","mac":"041812801251","slot":1,"model":"SPEM-003CEBEU","gen":2,"fw_id":"20241011-114455/1.4.4-g6d2a586","ver":"1.4.4","app":"Pro3EM","auth_en":false,"auth_domain":null,"profile":"triphase"}
25-03-21 18:36:39.297 TRACE uni-meter.output.http    - POST-1 RpcRequest: {"id":26,"method":"EM.GetConfig","params":{"id":0}}

25-03-21 18:36:39.314 TRACE uni-meter.output         - ShellyPro3EM.onRpcRequest()
25-03-21 18:36:39.314 TRACE uni-meter.output         - createRpcResult em.getconfig
25-03-21 18:36:39.314 TRACE uni-meter.output         - ShellyPro3EM.rpcEmGetConfig()
25-03-21 18:36:39.316 TRACE uni-meter.output         - createRpcResult ret {"id":0,"name":"Klaus3EM","blink_mode_selector":"active_energy","phase_selector":"a","monitor_phase_sequence":true,"reverse":{},"ct_type":"120A"}

sdeigm, 2025-03-23:
Hi Klaus,

I haven't tested to add the shelly to a hoymiles storage with the latest versions of uni-meter. I currently have no access to my system, but when that's the case again, I will try to build up a shelly connection to one of my storages to see if it is still working.

Some potential problems that come to my mind:

* you are using a test version: 1.1.2-UDP. Can you test with the release version 1.1.3 or 1.1.2? (or did I make a mistake and the release version prints out 1.1.2-UDP???)

* you have set up your own shelly id (mac address). Does the mac address in the uni-meter.conf matches the address specified in the avahi service files?

Otherwise I have no idea currently :-(

klausmuehle, 2025-03-23:
Stefan, thanks for your comment!
FYI: I used your latest source release to create the -udp version. 
The mac adress setting is correct, but i seem to have a problem with mDNS on different Lan/Wifi segments. I am checking this.

Do you know if automatic device detection from Hoymiles should work? Or if manual (ip adress based) detection works?

Kind regards
Klaus

sdeigm, 2025-03-23:
Ok, that clarifies the -UDP tag.

In general the Hoymiles storage or the App (I don't know which one does that part) looks up the Shelly's IP address using mDNS. Afterwards, the storage calls a number (I think two) HTTP requests to the Shelly to check its kind and configuration. When theses answers contain the data the storage expects, that storage builds up a websocket connection and polls the storage using the JSON RPC requests EM.GetStatus and EMData.GetStatus.

In the begining when I did the reverse engineering which methods are called by the storage and which answer is expected, at one point in time I had the feeling that the storage expects the properties of the JSON responses in a certain order. Therefore there are some @JsonPropertyOrder annonations at some Java classes. Hopefully I haven't mixed something up there in recent releases.

The last successful connection of a Hoymiles storage to the uni-meter I am aware of was version 1.0.9 in which I fixed the following issue: https://github.com/sdeigm/uni-meter/issues/52

klausmuehle, 2025-03-23:
Hi Stefan,
system is working here right now! I succeeded using version 1.1.3 running on a different machine, actually a VMware Linux on my m1 MacMini. This solves the mDNS topology problem: Now the Shelly Pro is automatically detected on integrated. 

Great SW! I expect this vonfiguration to be wifely used (digital powermeter with IR readout and Shelly emulation). 

Thanks!
Klaus

sdeigm, 2025-03-25:
Hi Klaus, good to hear that it is finally solved and only related to the mDNS topology and not a bug in the recent uni-meter versions. So I am closing this issue.

closed 2025-03-25 by sdeigm