sdeigm/uni-meter #80: [Workaround] APSystems EZHI - bind 3rd Party device after latest app update

kamilkosek, 2025-03-24:
Hi , 
in case anybody else ran into this problem after the latest app & firmware update for the inverter that you are unable to bind your emulated Shelly-Device, I have a solution for you: 
Just run this little python script 

```python
import requests
import json
import sys

# --- Helper functions for clean console output ---
def print_info(msg):
    print(f"[INFO] {msg}")

def print_success(msg):
    print(f"[SUCCESS] {msg}")

def print_warning(msg):
    print(f"[WARNING] {msg}")

def print_error(msg):
    print(f"[ERROR] {msg}")

def ask_input(msg):
    return input(f"[INPUT] {msg}: ")

# --- Step 1: User Login ---
print_info("Please enter your login credentials.")
username = ask_input("Username")
password = ask_input("Password")

headers_login = {
    "accept-language": "de_DE",
    "content-type": "application/x-www-form-urlencoded",
    "accept-encoding": "gzip",
    "user-agent": "okhttp/4.9.0",
}

payload_login = (
    f"language=de_DE"
    f"&password={password}"
    f"&app_id=4029817264d4821d0164d4821dd80015"
    f"&app_secret=EZAd2023"
    f"&username={username}"
)

print_info("Sending login request...")

login_response = requests.post(
    "https://app.api.apsystemsema.com:9223/api/token/generateToken/user/login?language=de_DE",
    headers=headers_login,
    data=payload_login,
)

if login_response.status_code != 200:
    print_error("Failed to contact API.")
    sys.exit(1)

login_data = login_response.json()
if login_data.get("code") != 0:
    print_error("Login failed. Please check your username/password.")
    sys.exit(1)

print_success("Login successful!")

access_token = login_data["data"]["access_token"]
user_id = login_data["data"]["user_id"]

# --- Step 2: Get device list ---
headers_auth = {
    "accept-language": "de_DE",
    "authorization": f"Bearer {access_token}",
    "accept-encoding": "gzip",
    "user-agent": "okhttp/4.9.0",
}

print_info("Fetching device list...")

device_response = requests.get(
    f"https://app.api.apsystemsema.com:9223/aps-api-web/api/v2/data/system/device/list/{user_id}?language=de_DE&systemId={user_id}",
    headers=headers_auth,
)

if device_response.status_code != 200:
    print_error("Failed to fetch device list.")
    sys.exit(1)

device_data = device_response.json()
if device_data.get("code") != 0:
    print_error("API returned an error while fetching devices.")
    sys.exit(1)

devices = device_data["data"]["list"]
if not devices:
    print_warning("No devices found in your account.")
    sys.exit(0)

# --- Step 3: Let user choose a device ---
print_info("Available devices:")
for idx, device in enumerate(devices):
    print(f"  [{idx}] {device['model']} (ID: {device['devId']})")

selected_index = int(ask_input("Select a device by index"))
selected_device_id = devices[selected_index]["devId"]

# --- Step 4: Ask for IP address ---
device_ip = ask_input("Enter the IP address of the Shelly Emulator")

# --- Step 5: Bind new device ---
headers_bind = {
    "accept-language": "de_DE",
    "authorization": f"Bearer {access_token}",
    "content-type": "application/x-www-form-urlencoded",
    "accept-encoding": "gzip",
    "user-agent": "okhttp/4.9.0",
}

device_info = {
    "deviceDevId": "shellypro3em-b827eb364242",
    "deviceName": "Emulator",
    "type": "0",
    "commType": "1",
    "inverterDevId": selected_device_id,
    "ip": device_ip,
    "model": "SPEM-003CEBEU",
}

payload_bind = f"language=de_DE&deviceInfo={json.dumps(device_info)}"

print_info("Sending device binding request...")

bind_response = requests.post(
    "https://app.api.apsystemsema.com:9223/aps-api-web/api/v2/device/ezhi/third/bind?language=de_DE",
    headers=headers_bind,
    data=payload_bind,
)

if bind_response.status_code != 200:
    print_error("Failed to contact binding API.")
    sys.exit(1)

bind_data = bind_response.json()
if bind_data.get("code") != 0:
    print_error(f"Binding failed: {bind_data.get('msg', 'Unknown error')}")
    sys.exit(1)

print_success("Device successfully bound to your account!")
```

The script assumes you are  using the default `shellypro3em.http.tcp.service`  config. 
After you run it and everythings successful you will be able to do the phase detection and turn on the "intelligent" mode.

sdeigm, 2025-03-24:
Hi, good to hear, that you found a workaround and published it here, thanks!

Just as an information for me since I don't own a EZHI device? You script tells the EZHI device via cloud the IP address of the local shelly/shelly-emulator? And afterwards the EZHI device connects to the shelly using a websocket connection or is it still the other way around and the shelly has to build up an outbound websocket connection to the EZHI device?

MuellBin2, 2025-03-24:
Wow this sound too easy. thank you very much! 
"just run that phyton script" ... needs some more than that to run, let me fiddl around with that!

MuellBin2, 2025-03-24:
Ok, you may need to run "pip install requests" first. 
Really magic, after injecting the data to the APSystems account, the shelly magically appears back in your app.

just run up Uni Meter again, updatet to 1.1.3 and used the pre configured websocket connection.

```
25-03-24 17:47:03.700 DEBUG uni-meter.output         - incoming websocket connection 236de024-c358-4257-80bf-50aedfc9b19d opened
25-03-24 17:47:03.894 DEBUG uni-meter.output         - incoming websocket connection 236de024-c358-4257-80bf-50aedfc9b19d failed: WebSocket upgrade did not finish because of 'unexpected status code: 404 Not Found'
25-03-24 17:47:03.895 DEBUG uni-meter.output         - outbound websocket connection to ws://apsystems/meter failed: unexpected response status: 404 Not Found
```

I walked the process all over again.
checked the config, it cointaing the websocket with the correct IP adress (192.168.1.96". 
inserted config, restarted the uni-meter, it startes receiving the values.
with the running uni meter, i removed the emulator from the APSystems App, I rund your script again, selected the EZHI, inserted the IP Adress of the UNI Meter "192.168.1.105"

... hmm still

every 5 seconds I find this inside the script

`25-03-24 17:55:25.552 DEBUG uni-meter.output         - outgoing websocket connection 6752f59e-5c90-4b20-9ab1-15f2ecef2105 to /192.168.1.96 created
25-03-24 17:55:25.554 DEBUG uni-meter.output         - incoming websocket connection 6752f59e-5c90-4b20-9ab1-15f2ecef2105 opened
25-03-24 17:55:25.674 DEBUG uni-meter.output         - incoming websocket connection 6752f59e-5c90-4b20-9ab1-15f2ecef2105 failed: WebSocket upgrade did not finish because of 'unexpected status code: 404 Not Found'`

MuellBin2, 2025-03-24:
```
25-03-24 18:14:42.114 INFO  uni-meter                - ##################################################################
25-03-24 18:14:42.121 INFO  uni-meter                - # Universal electric meter converter 1.1.3 (2025-03-21 18:55:40) #
25-03-24 18:14:42.122 INFO  uni-meter                - ##################################################################
25-03-24 18:14:42.122 INFO  uni-meter                - initializing actor system
25-03-24 18:14:42.869 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-03-24 18:14:43.231 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-03-24 18:14:43.249 INFO  uni-meter.controller     - creating TibberPulse input device
25-03-24 18:14:45.230 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-03-24 18:14:45.277 DEBUG uni-meter.output         - outgoing websocket connection 62c95ebf-ee4a-41a1-bae6-1d6c321cf936 to /192.168.1.96 created
25-03-24 18:14:45.319 DEBUG uni-meter.output         - incoming websocket connection 62c95ebf-ee4a-41a1-bae6-1d6c321cf936 opened
25-03-24 18:14:45.498 ERROR uni-meter.output         - outbound websocket connection to ws://192.168.1.96/meter failed: unexpected response status: 404 Not Found
25-03-24 18:14:45.512 DEBUG uni-meter.output         - incoming websocket connection 62c95ebf-ee4a-41a1-bae6-1d6c321cf936 failed: WebSocket upgrade did not finish because of 'unexpected status code: 404 Not Found'
```

kamilkosek, 2025-03-24:
We´ll figure it out. The same is at my site, obviously I was to excited to post it and didnt tested it 100%. 
Ive got some Shelly devices laying around, so ill check this as soon as the kids are going to sleep.

kamilkosek, 2025-03-24:
> Hi, good to hear, that you found a workaround and published it here, thanks!
>
> Just as an information for me since I don't own a EZHI device? You script tells the EZHI device via cloud the IP address [...]

Hey @sdeigm , I analyzed what the App was doing in the version before and just tried it out. As my I was seeing the 3rd party device in the AP Systems app, I thought it will just work  - but for now the device is just listed there but it´s not gathering any data.

kamilkosek, 2025-03-24:
Hi all, 
 so i´ve been tinkering with it for a bit now and this are my findings: 

1. Once you updated  the firmware of the EZHI Inverter, adding devices the "classic" way (like in the script) will just show them in the App, but there isn´t any functionality. 
2. Even if you add a Shelly device through the shelly cloud, the inverter is only communicating with `https://3.67.1.32:9005` which resolves to `ecu.apsystemsema.com` , so I don´t see any chance figure out what is happening there. 
3. What "could" work though, is to fiddle out how to add the emulated device to the shelly cloud. 

But I think it maybe not worth the effort , as AP Systems stated they will release the firmware with the local API by May.

sdeigm, 2025-03-25:
With the emulated device I have the same problem as you have with the HTTPS connection to ```ecu.apsystemsema.com```. The Shelly devices are also using a HTTPS connection and an undocumented interface when they are talking to their cloud. So there is also no chance to figure out what would have to be done. So when the vendors of a storage/inverter do not allow to specify/lookup the IP address of the shelly locally, there is no chance to use an emulation :-(

MuellBin2, 2025-03-25:
> Hi all, so i´ve been tinkering with it for a bit now and this are my findings:
> 
>     1. Once you updated  the firmware of the EZHI Inverter, adding devices the "classic" way (like in the script) will just show them in the App, but there isn´t any functionality.
> 

This was also possible by using the old app with the new firmware. On Android you can downgrade to still have the functionality.
Interestingly, I restarted the EZHI twice now (since there is no use at night ...) and the previous added shelly devices now get lost. Shouldn't be that way, either. 

>But I think it maybe not worth the effort , as AP Systems stated they will release the firmware with the local API by May.

Where do you get news about this?

kamilkosek, 2025-03-25:
> Where do you get news about this?

I´ve asked specifically about it: 

[image]

kamilkosek, 2025-05-06:
So some sort of good news here: 
APSystems released today a new version in which you can enable the local mode. 
I´ve created already a Homeassistant Add-On for it: [https://github.com/kamilkosek/EZHI](https://github.com/kamilkosek/EZHI) Feel free to try it out.

MuellBin2, 2025-05-15:
Lokal Mode is now available and with Firmware Version 1.5.0.9 the local API value is taken over, so basically the deivce can now be steered calling its link  (http:// <IP> /setPower?p=...) and a Power Value. I will try to use unimeter in order to have a local solution that is running with the Tibber Pulse readout of my meter.