sdeigm/uni-meter #86: Feature Request: Add configuration option for HA entity as source

thomluther, 2025-03-28:
Hello @sdeigm 

I take the chance to open a new feature request to add another configuration option for the power source of the uni-meter.

Along with the new HA Add-on capability, it would be just another simplification and great enhancement to utilize any of the power entities in HA as input source. This would even enable much more capabilities and power regulation modifications, since the HA power entity could already be a computed entity that tells the Storage what it has to consider as 'grid' power.

HA offers an Api while can be used to poll any entity state.
Similar capability exists in Tomquist's B2500 Add on. Here is the module that does the HA Api state parsing for the configured entity.
https://github.com/tomquist/b2500-meter/blob/main/powermeter/homeassistant.py

The HA Add on would need to have another configuration option for the access token, that must be used for the Api poll.

Gitsarry, 2025-03-29:
Second this, e.g. I have the Hichi IR optical read head with Tasmota in service, and that sends the power values via MQTT to Home Assistant MQTT broker, which is best way for reliability. So, I have already the values as entities in HA via the HA Tasmota integration (called e.g. `sensor.tasmota_lk13be_power`). No need to poll the Tasmota HTTP interface additionally (and the HTTP interface is not intended for this use, too)

meier-rene, 2025-03-30:
@Gitsarry you can easilly configure uni-meter to poll the data out of the mqtt broker in ha. Here is my config. You might need to adjust the topics and the json path
```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"
  input = "uni-meter.input-devices.mqtt"
  output-devices {
    shelly-pro3em {
      device {
        mac = "B827EB364242"
        hostname = "shellypro3em-b827eb364242"
      }
    }
  }
  input-devices {
    mqtt {
      url = "tcp://192.168.178.100:1883"
      username = "****"
      password = "****"

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"

      channels = [{
        type = "json"
        topic = "tele/tasmota_F7F3B8/SENSOR"
        channel = "power-total"
        json-path = "$..Power_curr"
      },{
        type = "json"
        topic = "tele/tasmota_F7F3B8/SENSOR"
        channel = "energy-consumption-total"
        json-path = "$..Total_in"
      },{
        type = "json"
        topic = "tele/tasmota_F7F3B8/SENSOR"
        channel = "energy-production-total"
        json-path = "$..Total_out"
      }]
    }
  }
}
```

Gitsarry, 2025-03-30:
Thanks, I got it running so far. Here my steps on Home Assistant Operating System qemu KVM:

- go to addon-store -> upper right menu -> repositories -> add "https://github.com/sdeigm/uni-meter"  (the add URL link from the readme did work yesterday, but not today, today there was no text in the popup window, so I canceled that)
- ssh as root to HA VM, `cd /addon_configs/`, then create directory: `mkdir 663b81ce_uni_meter`, and in that folder create the config file `uni-meter.conf`

Generate a random mac address. Question: Does it work when the mac address begins wit `02:` which indicates local generated mac address? This is correct approach. But I took your provided mac address and only changed that only at some places in the middle, to be sure.

To see the Tasmota mqtt topics, I went in HA web interface to config -> devices+services -> mqtt -> configure -> configure -> listen to topic (at the bottom), enter `tele/#` as topic and start listening.

Since my counter only provides new value every 1s, I thought it might be good idea to add `min-sample-period = 1000ms`, does that make sense?

Here is my `uni-meter.conf` for counter Logarex LK13BE:

```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"
  input = "uni-meter.input-devices.mqtt"
  output-devices {
    shelly-pro3em {
      device {
        mac = "B821EC367242"
        hostname = "shellypro3em-b821ec367242"
        min-sample-period = 1000ms
      }
    }
  }
  input-devices {
    mqtt {
      url = "tcp://127.0.0.1:1883"
      username = "some_created_ha_user"
      password = "***replace***"

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"

      channels = [{
        type = "json"
        topic = "tele/tasmota_FA33FC/SENSOR"
        channel = "power-total"
        json-path = "$..power"
      },{
        type = "json"
        topic = "tele/tasmota_FA33FC/SENSOR"
        channel = "energy-consumption-total"
        json-path = "$..energy_sum"
      },{
        type = "json"
        topic = "tele/tasmota_FA33FC/SENSOR"
        channel = "energy-production-total"
        json-path = "$..energy_supply"
      }]
    }
  }
}
```

This is the add-on protocol:
```
-----------------------------------------------------------
 Add-on: uni-meter
 A universal electrical meter converter
-----------------------------------------------------------
 Add-on version: 1.1.3
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.0  (amd64 / qemux86-64)
 Home Assistant Core: 2025.3.3
 Home Assistant Supervisor: 2025.03.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[14:23:24] INFO: using custom configuration file: uni-meter.conf
25-03-30 14:23:25.439 INFO  uni-meter                - ##################################################################
25-03-30 14:23:25.451 INFO  uni-meter                - # Universal electric meter converter 1.1.3 (2025-03-21 18:55:40) #
25-03-30 14:23:25.451 INFO  uni-meter                - ##################################################################
25-03-30 14:23:25.452 INFO  uni-meter                - initializing actor system
25-03-30 14:23:26.555 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-03-30 14:23:27.222 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-03-30 14:23:27.256 INFO  uni-meter.controller     - creating MQTT input device
25-03-30 14:23:28.006 INFO  uni-meter.input          - subscribing to topic: tele/tasmota_FA33FC/SENSOR
25-03-30 14:23:29.083 INFO  uni-meter.input          - MQTT stream connected
25-03-30 14:23:30.337 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
```
Remaining question for me now:
- The addon-service does not provide the mqtt-fetched values as entities, is this correct?
- How can I check if the addon-service has the mqtt-fetched values? E.g. a http query?

meier-rene, 2025-03-30:
Question 2: https://github.com/sdeigm/uni-meter/issues/75#issuecomment-2763085672

Question 3: I think I managed that today. Here are the steps:
-Install HACS
-install pyscript
-use script from this comment https://github.com/sdeigm/uni-meter/issues/75#issuecomment-2763996165 or use a hardcoded version like mine
```
import socket
from homeassistant.components import zeroconf
from zeroconf.asyncio import AsyncServiceInfo

@service
def register_service1():
    type="_shelly"
    name="shellypro3em-<yourmac>"
    ip="192.168.178.100"
    port=80
    aiozc = zeroconf.async_get_async_instance(hass)
    properties = {
        "fw_id": "20241011-114455/1.4.4-g6d2a586",
        "arch": "esp8266",
        "id": "shellypro3em-<yourmac>",
        "gen": "2"
    }
    info = AsyncServiceInfo(
        type_=type + "._tcp.local.",
        name=name + "." + type + "._tcp.local.",
        addresses=[socket.inet_aton(ip)],
        port=port,
        properties=properties
    )
    aiozc.async_register_service(info)

@service
def register_service2():
    type="_http"
    name="shellypro3em-<yourmac>"
    ip="192.168.178.100.20"
    port=80
    aiozc = zeroconf.async_get_async_instance(hass)
    properties = {
        "fw_id": "20241011-114455/1.4.4-g6d2a586",
        "arch": "esp8266",
        "id": "shellypro3em-<yourmac>",
        "gen": "2"
    }
    info = AsyncServiceInfo(
        type_=type + "._tcp.local.",
        name=name + "." + type + "._tcp.local.",
        addresses=[socket.inet_aton(ip)],
        port=port,
        properties=properties
    )
    aiozc.async_register_service(info)
```
-place the script in `/config/pyscript`
-restart, go to developer tools > actions
-try to find `Pyscript Python scripting: register_service1` and  `Pyscript Python scripting: register_service2` and execute both
Make sure that you use a uniq mac and use the same mac all over the place. 
After that I could include the uni-meter as shelly replacement in the hoymiles app.

Gitsarry, 2025-03-30:
> Question 2: [#75 (comment)](https://github.com/sdeigm/uni-meter/issues/75#issuecomment-2763085672)

Thanks, great it seems to work :)    (I'd suggest to put those commands in the docu)
```
$ curl --no-progress-meter http://192.168.1.4/rpc/EMData.GetStatus?id=0 | jq
{
  "id": 0,
  "a_total_act_energy": 1481.39,
  "a_total_act_ret_energy": 232.34,
  "b_total_act_energy": 1481.39,
  "b_total_act_ret_energy": 232.34,
  "c_total_act_energy": 1481.39,
  "c_total_act_ret_energy": 232.34,
  "total_act": 4444.17,
  "total_act_ret": 697.02
}

$ curl --no-progress-meter http://192.168.1.4/rpc/EM.GetStatus?id=0 | jq
{
  "id": 0,
  "a_current": 0.14,
  "a_voltage": 230,
  "a_act_power": 31.33,
  "a_aprt_power": 31.33,
  "a_pf": 1,
  "a_freq": 50,
  "b_current": 0.14,
  "b_voltage": 230,
  "b_act_power": 31.33,
  "b_aprt_power": 31.33,
  "b_pf": 1,
  "b_freq": 50,
  "c_current": 0.14,
  "c_voltage": 230,
  "c_act_power": 31.33,
  "c_aprt_power": 31.33,
  "c_pf": 1,
  "c_freq": 50,
  "total_current": 0.42,
  "total_act_power": 93.99,
  "total_aprt_power": 93.99
}
```

Gitsarry, 2025-03-30:
> ip="192.168.178.100.20"

Is this a typo in your script?

Gitsarry, 2025-03-30:
Using fifth octett returns error: "OSError: illegal IP address string passed to inet_aton"

meier-rene, 2025-03-30:
You are right. Did testing on different system and made a mistake copying over the script. So second function is unnecessary for sure.

Gitsarry, 2025-03-30:
> Question 3: I think I managed that today. Here are the steps:

Many thanks! I think I did now all the steps (additionally, fixed the IP in the http-service)

After starting the two services (they do not survive reboot), they both show up in the avahi scan on my host computer:
(mac address redacted)

```
$ avahi-browse -a -r

= bridge0 IPv6 shellypro3em-b82712345678                     _shelly._tcp         local
   hostname = [shellypro3em-b82712345678._shelly._tcp.local]
   address = [192.168.1.4]
   port = [80]
   txt = ["gen=2" "id=shellypro3em-b82712345678" "arch=esp8266" "fw_id=20241011-114455/1.4.4-g6d2a586"]

= bridge0 IPv4 shellypro3em-b82712345678                     Web-Angebot          local
   hostname = [shellypro3em-b82712345678._http._tcp.local]
   address = [192.168.1.4]
   port = [80]
   txt = ["gen=2" "id=shellypro3em-b82712345678" "arch=esp8266" "fw_id=20241011-114455/1.4.4-g6d2a586"]
```

If I should have the Hoymile device by tomorrow, I can check the install :blush:

sdeigm, 2025-03-30:
Thanks for all your findings and the testing effort!

I also found a REST API call to execute the service from within the uni-meter addon:

[``POST api/services/<domain>/<service>``](https://developers.home-assistant.io/docs/api/rest/)

So it should be possible to add a mDNS "auto-setup" for either Home Assistant or the plain avahi daemon. I will create an issue for this feature.

One thing that remains open: how do we get the Python function into pyscript? Is this just a documentation issue or is there a chance to install that service function togehter with the addon?

Gitsarry, 2025-03-30:
> how do we get the Python function into pyscript? Is this just a documentation issue

pyscript is a custom experimental integration that depends of installation of HACS (Home Assistant Community Store), reading: this is for custom, special, expert use.
See the warning on https://www.hacs.xyz/docs/use/download/download/
_"If you don't know what type of Home Assistant installation you are running, you should not use HACS (or any other custom integration)."_

I personally want my Home Assistant to be complete standard with first priority: stability. So, I only use the stable add-ons/integrations.

I only made this exception here to install HACS and the Hacs-Pyscript-integration, to get this feature working, as a one time exception.
I hope that in the near future, I can remove the Pyscript and Hacs alltogether again.

closed 2025-04-01 by sdeigm