sdeigm/uni-meter #87: Doc: Unclear, if step "Announcing the tool via mDNS" needed with HA-Addon

Gitsarry, 2025-03-29:
Thanks for this great work. Since I have already a Hichi IR read head with Tasmota Home Assistant Addon running, I do not need Shelly Pro E3M, but battery systems need that, so this emulator is perfectly what I need.

Reading the Readme, it remains unclear to me, if the step "Announcing the tool via mDNS" is needed when installing the Home Assistant Addon.

I would suggest to make the steps in a hierarchical structure, not like it is now, everything flat underneath below.

sdeigm, 2025-03-30:
For the Hoymiles storage it is also necessary to announce the virtual Shelly via mDNS. But it has to be done differently to a bare metal raspberry pi or docker container: the avahi daemon can not be used in Home Assistant. The my comments in the other issue.

Regarding the documentation you are absolutely right. There was quit a bit dynamic in the project in the last time. It started when I wrote the emulator for my own Hoymiles storage and it was just running on a raspberry pi. Then the Marstek storage was added, more and more input devices were added, a docker container and now Home Assistant. So while the documentation was initially a perfect fit, it now needs a major redesign :-)

Gitsarry, 2025-03-30:
Thank you so much for your work. I have Home Assistant running in a qemu KVM.