sdeigm/uni-meter #89: uni-meter does not work when at startup the mqtt input device was not available

Gitsarry, 2025-03-31:
uni-meter-1.1.3

Having in file `/etc/uni-meter.conf` the following MQTT input source, where 192.168.1.4:1883 is the Home Assistant MQTT broker:

```
  input-devices {
    mqtt {
      url = "tcp://192.168.1.4:1883"
      username = "mqttclient"
      password = "****"

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"

      channels = [{
        type = "json"
        topic = "tele/tasmota_FA33FC/SENSOR"
        channel = "power-total"
        json-path = "$..power"
      },{
        type = "json"
        topic = "tele/tasmota_FA33FC/SENSOR"
        channel = "energy-consumption-total"
        json-path = "$..energy_sum"
      },{
        type = "json"
        topic = "tele/tasmota_FA33FC/SENSOR"
        channel = "energy-production-total"
        json-path = "$..energy_supply"
      }]
    }
  }
```

When uni-meter systemd unit is started before the Home Assistant MQTT broker is up and running, then uni-meter will not start working at all, even not when the MQTT broker becomes available.

It would be better if uni-meter would try every e.g. 1 min again, and start working as soon as the input source becomes available.

Currently, the solution is to restart uni-meter after mqtt broker is running.

sdeigm, 2025-04-01:
Can confirm the problem. In theory uni-meter is already designed to always retry failed operations. Here I didn't reinitialized the underlyling MQTT library correctly.

closed 2025-04-01 by sdeigm

Gitsarry, 2025-04-01:
Thank you very much for your work

Gitsarry, 2025-04-03:
Have just tested with 1.1.4 and can confirm issue is fixed:

- stop uni-meter
- stop Home Assistant and with it the HA MQTT broker, which is input-source of uni-meter
- start uni-meter
- wait until log entry `MQTT stream failed: MqttException` occurs three times in a row
- start Home Assistant
- log entry `MQTT stream connected` shows up and uni-meter is working

```
25-04-03 06:10:58.007 INFO  uni-meter                - ##################################################################
25-04-03 06:10:58.012 INFO  uni-meter                - # Universal electric meter converter 1.1.4 (2025-04-01 05:12:23) #
25-04-03 06:10:58.012 INFO  uni-meter                - ##################################################################
25-04-03 06:10:58.012 INFO  uni-meter                - initializing actor system
25-04-03 06:10:58.781 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-04-03 06:10:59.252 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-04-03 06:10:59.277 INFO  uni-meter.controller     - creating MQTT input device
25-04-03 06:10:59.598 INFO  uni-meter.input          - subscribing to topic: tele/tasmota_FA33FC/SENSOR
25-04-03 06:10:59.885 ERROR uni-meter.input          - MQTT stream failed: MqttException
25-04-03 06:11:00.474 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-04-03 06:11:17.980 ERROR uni-meter.input          - MQTT stream failed: MqttException
25-04-03 06:11:50.041 ERROR uni-meter.input          - MQTT stream failed: MqttException
25-04-03 06:12:52.478 ERROR uni-meter.input          - MQTT stream failed: MqttException
25-04-03 06:13:52.871 INFO  uni-meter.input          - MQTT stream connected

```