sdeigm/uni-meter #94: ERROR uni-meter.http.port-80   - exception in HTTP server: java.lang.IllegalArgumentException: unhandled RPC method 'EM.GetConfig'

CountZ74, 2025-04-03:
Hi, ich sehe im Log obigen Fehler, wenn der MS-A2 versucht sich mit dem uni-meter Shelly zu verbinden. Es kommt dann eine Meldung in der S-Miles Home App "Daten konnten nicht abgerufen werden".

CountZ74, 2025-04-03:
Ich hatte ursprünglich 1.0.8 verwendet. Heute habe ich dann versucht, die Homeassistant Integration zu verwenden. Nachdem das nicht funktioniert hat, habe ich den Container mit mit uni-meter wieder gestartet und der MS-A2 bekam keine Daten mehr. Danach hab ich alles mögliche ausprobiert und irgendwann viel mir dieser Fehler auf. Jetzt habe ich alles mit 1.1.4 neu installiert und der Fehler ist weg und die Daten kommen wieder beim Speicher an.

sdeigm, 2025-04-03:
Ok, gut das funktioniert!

Wenn du das uni-meter in Home Assitant mit dem Hoymiles Storage verwenden willst, musst du es dort auch per mDNS bekannt machen. Wie das geht haben wir erst vor ein paar Tagen herausgefunden. Du musst dazu HACS und Pyscript installieren und dann eine Python Service Funktion aufrufen, die die entsprechenden mDNS Einstellungen vornimmt. Das muss ich noch mal genau dokumentieren, wenn ich dazu komme.

Bis dahin könntest du es hier nachlesen: https://github.com/sdeigm/uni-meter/issues/86#issuecomment-2764586640

Diesen Issue schließe ich damit dann gleich wieder.

closed 2025-04-03 by sdeigm