sdeigm/uni-meter #98: SHRDZM with uni-meter and Hoymiles MS-A2 mising values

wuk986, 2025-04-08:
First of all. Awesome project.
Thank you for developing it.

I added the uni-meter in a small Linux vm and after some attempts I'm able to connect my Hoymiles MS-A2 to the uni-meter.

I'm using a SHRDZM smart meter sending the data by UDP to the uni-meter, but what makes a bit wondering is, that the Hoymiles MS-A2 is not delivering the energy that I expect.
Maybe of the missing data from the grid and the data for phase A to C.

My uni-meter config:
`uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.shrdzm"
  
  output-devices {
    shelly-pro3em {
      type = "ShellyPro3EM"
      #port = 80
      #bindAddress = "0.0.0.0"
      device {
        mac = "A4EECDBC799F"
        hostname = "shellypro3em-a4eecdbc799f"
      }
    }
  }

  input-devices {
    shrdzm {
      port = 9522
      interface = "0.0.0.0"
    }
  }
}
`

I checked Wireshark and status web pages of the uni-meter and the S-Miles Home app.

**Wireshark:**
[image]

**uni-meter:**
[image]

**S-Miles Home:**
[image]
[image]

sdeigm, 2025-04-08:
At a first glance your configuration looks ok to me.

May it be the case that the data from UDP port 9522 is not routed into your virtual machine? Depends on the type of network you are using for the virtual machine (host, bridged ...). Maybe a port forwarding entry is missing in that configuration?

Otherwise you can enable the debug logging in the input device to see if the correct data arrives at the uni-meter (the data from the wireshark trace looks ok) by using the following ```logback.xml``` file:

```
<?xml version="1.0" encoding="UTF-8"?>
<Configuration status="WARN">
    <appender name="STDOUT" class="ch.qos.logback.core.ConsoleAppender">
        <encoder>
            <pattern>%d{yy-MM-dd HH:mm:ss.SSS} %-5level %-24logger - %msg%n</pattern>
        </encoder>
    </appender>

    <root level="INFO">
        <appender-ref ref="STDOUT" />
    </root>

    <logger name="uni-meter.input"  level="DEBUG" additivity="false">
        <appender-ref ref="STDOUT" />
    </logger>
</Configuration>
```

wuk986, 2025-04-08:
Thank you for the hint. The issue was a port mismatch with the ports of the uni-meter and SHRDZM.

closed 2025-04-08 by wuk986