sdeigm/uni-meter #99: Unable to pairing

badsmoke, 2025-04-08:
hey i seem to have similar/same problem as the issue, but wanted to open a new issue to be on the safe side
i also get the message that the data is not being fetched

https://github.com/sdeigm/uni-meter/issues/97

docker-compose
```
.
.
.
  uni-meter:
    image: sdeigm/uni-meter
    network_mode: host
    restart: unless-stopped
    ports:
      - 80:80/tcp
      - 1010:1010/udp
      - 5353:5353/udp
    expose:
      - 80
      - 1010
      - 5353
    volumes:
      - ./unit-meter/uni-meter.conf:/etc/uni-meter.conf
      - ./unit-meter/logback.xml:/opt/uni-meter/config/logback.xml
      - ./unit-meter/shellypro3em.http.tcp.service:/etc/avahi/services/shellypro3em.http.tcp.service
      - ./unit-meter/shellypro3em.shelly.tcp.service:/etc/avahi/services/shellypro3em.shelly.tcp.service

```

uni-meter.conf
```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.generic-http"

  output-devices {
    shelly-pro3em {
      udp-port = 1010
      device {
         mac = "B327EB8C72F1"
         hostname = "shellypro3em-b327eb8c72f1"
      }

      min-sample-period = 5000ms
      forget-interval = 1m

      default-power-total = 0

      default-power-l1 = 0
      default-power-l2 = 0
      default-power-l3 = 0

    }
  }

  input-devices {
    generic-http {
      url = "http://192.168.0.106:7004/api/v3/uni-meter"

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"

      channels = [{
        type = "json"
        channel = "energy-consumption-total"
        json-path = "$.data[0].tuples[0][1]"
        scale = 0.001
      },{
        type = "json"
        channel = "energy-production-total"
        json-path = "$.data[1].tuples[0][1]"
        scale = 0.001
      },{
        type = "json"
        channel = "power-total"
        json-path = "$.data[2].tuples[0][1]"
      }]
    }
  }
}

```
```
root@hydra:/srv/home-automation/unit-meter# ll
total 24
drwxr-xr-x  2 root     root     4096 Apr  8 07:42 ./
drwxr-xr-x 10 badsmoke badsmoke 4096 Apr  7 18:24 ../
-rw-r--r--  1 root     root      490 Apr  7 16:39 logback.xml
-rw-r--r--  1 root     root      478 Apr  7 11:14 shellypro3em.http.tcp.service
-rw-r--r--  1 root     root      480 Apr  7 11:13 shellypro3em.shelly.tcp.service
-rw-r--r--  1 root     root     1122 Apr  8 07:42 uni-meter.conf

```
i have also adjusted the name of the shelly in the two service files
even if not, it should work by adding it manually

the shelly device also seems to work, a curl on the status and shelly page works and also brings the correct valuesthe shelly device also seems to work, a curl on the status and shelly page works and also brings the correct values

`curl 192.168.0.106/status`
```json
{"wifi_sta":{"connected":true,"ssid":"kabellos","ip":"192.168.178.99","rssi":-83},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"07:50","unixtime":1744098622,"serial":1,"has_update":false,"mac":"B327EB8C72F1","ram_total":50648,"ram_free":38376,"ram_lwm":32968,"fs_size":233681,"fs_free":174194,"uptime":425,"temperature":28.08,"overtemperature":false,"tmp":{"tC":28.08,"tF":82.54,"is_valid":true},"emeters":[{"power":-19.0,"pf":1.0,"current":-0.08,"voltage":230.0,"is_valid":true,"total":0.04,"total_returned":0.12},{"power":-19.0,"pf":1.0,"current":-0.08,"voltage":230.0,"is_valid":true,"total":0.04,"total_returned":0.12},{"power":-19.0,"pf":1.0,"current":-0.08,"voltage":230.0,"is_valid":true,"total":0.04,"total_returned":0.12}],"total_power":-57.0,"fs_mounted":true}
```
only the fake wifi ip is a bit irritating, can it be adjusted?

`curl 192.168.0.106/shelly`

```json
{"id":"shellypro3em-b327eb8c72f1","mac":"B327EB8C72F1","slot":1,"model":"SPEM-003CEBEU","gen":2,"fw_id":"20241011-114455/1.4.4-g6d2a586","ver":"1.4.4","app":"Pro3EM","auth_en":false,"auth_domain":null,"profile":"triphase"}
```

uni-meter logs:
```
##################################################################
25-04-08 07:43:17.007 INFO  uni-meter                - # Universal electric meter converter 1.1.4 (2025-04-01 05:12:23) #
25-04-08 07:43:17.007 INFO  uni-meter                - ##################################################################
25-04-08 07:43:17.007 INFO  uni-meter                - initializing actor system
25-04-08 07:43:17.274 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-04-08 07:43:17.399 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-04-08 07:43:17.411 INFO  uni-meter.controller     - creating GenericHttp input device
25-04-08 07:43:17.778 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-04-08 07:43:17.925 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80

```

thanks for your help

sdeigm, 2025-04-08:
I think it is just a mDNS issue. The question is, if the mDNS traffic on port 5353 is routed into the container. Is there a avahi daemon running on your host machine, that runs the container? If that is the case, you have to copy the two avahi services into the /etc/avahi/services directory of your host computer and you don't have to mount them in the container. Since then the port 5353 is not routed into the container.

badsmoke, 2025-04-08:
hey, avahi, wasn't installed, so shouldn't have interfered.

after i installed avahi and the two services in the right folder, the shelly was found, it works now

thanks

sdeigm, 2025-04-08:
Ok, good to hear that it is working now!

But just to fix future problems with my docker-compose file, can you do me a favor and test the container again?

For this please stop the newly installed avahi-daemon once again be using

```sudo systemctl stop avahi-daemon```

and then start the docker container with the following docker compose setup:

```
.
.
.
 uni-meter:
    image: sdeigm/uni-meter
    network_mode: host
    restart: unless-stopped
    expose:
      - 80
      - 5353/udp
    volumes:
      - ./unit-meter/uni-meter.conf:/etc/uni-meter.conf
      - ./unit-meter/logback.xml:/opt/uni-meter/config/logback.xml
      - ./unit-meter/shellypro3em.http.tcp.service:/etc/avahi/services/shellypro3em.http.tcp.service
      - ./unit-meter/shellypro3em.shelly.tcp.service:/etc/avahi/services/shellypro3em.shelly.tcp.service
```

I think the missing 5353/udp prefix was the problem and I should correct that in my sample file ...

badsmoke, 2025-04-08:
no that does not seem to be the trick, it is then not found again

closed 2025-05-05 by sdeigm