sdeigm/uni-meter #102: Failed to parse status response: wrong crc / Timeout

kopfplatzwunde, 2025-04-12:
Hallo,

im Log wird immer im Abstand von wenigen Minuten folgender ERROR ausgegeben:
     
##################################################################
25-04-12 10:30:55.429 INFO  uni-meter                - # Universal electric meter converter 1.1.5 (2025-04-11 08:31:17) #
25-04-12 10:30:55.430 INFO  uni-meter                - ##################################################################
25-04-12 10:30:55.430 INFO  uni-meter                - initializing actor system
25-04-12 10:30:55.885 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-04-12 10:30:56.167 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-04-12 10:30:56.175 INFO  uni-meter.controller     - creating TibberPulse input device
25-04-12 10:30:56.901 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-04-12 10:30:57.225 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80

25-04-12 10:34:19.831 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-12 10:35:45.677 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-12 10:36:57.256 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-12 10:45:19.130 ERROR uni-meter.input          - Failed to parse status response: wrong crc
25-04-12 10:47:51.218 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-12 10:50:23.998 ERROR uni-meter.input          - Failed to parse status response: Timeout

Hat einer ein ähnliches Problem oder woran könnte das liegen?

sdeigm, 2025-04-12:
Hallo, im Photovoltaikforum hatte neulich jemand das selbe Problem. Da lag es an einer fehlenden Portfreigabe: https://www.photovoltaikforum.com/thread/241171-marstek-venus-e-c-reiner-ac-speicher/?postID=4247779#post4247779

Es sieht einfach so aus, als wenn das uni-meter deine Tibber Bridge nicht erreichen kann. Kommt bei dir denn etwas, wenn du die Tibber URL einfach in den Browser eingibst?

Ansonsten am Besten die uni-meter.conf hier posten ...

kopfplatzwunde, 2025-04-12:
Ok Danke für den Hinweis.

Habe den Uni-Meter in einen LXC Container auf meinen Proxmox System laufen. Hoffe das ich die Portfreigabe hin bekomme, wenn jemand ne schnelle Lösung hat gern kommunizieren?

Port 80
Port 1010 (für Marstek)

Was etwas komisch ist, das über den Log Trace Daten vom Tibber kommen. Die bridge wird also scheinbar erreicht.

kopfplatzwunde, 2025-04-13:
Also komme irgendwie nicht weiter ... meiner Meinung sind die Ports alle verfügbar und offen ... 

Hier mal die conf:

uni-meter {
    output = "uni-meter.output-devices.shelly-pro3em"
    
    input = "uni-meter.input-devices.tibber-pulse"
    
    output-devices {
        shelly-pro3em {
        udp-port = 1010
                        }
                }
    
    input-devices {
        tibber-pulse {
            url = "http://192.168.188.101"
            node-id = 1
            user-id = "admin"
            password = "XXXX-XXXX"
    }
  }
}

Die Tibber Bridge ist auch verfügbar über die URL.

[image]

Hier mal ein Auszug aus den Log:

25-04-13 11:20:47.285 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-04-13 11:20:48.304 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-04-13 11:20:48.304 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-04-13 11:20:48.414 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-04-13 11:20:48.414 TRACE uni-meter.input          - Pulse.handlePulseStatusEntity()
25-04-13 11:20:48.414 DEBUG uni-meter.input          - Found tag OPEN_RESPONSE
25-04-13 11:20:48.414 DEBUG uni-meter.input          - Found tag GET_LIST_RESPONSE
25-04-13 11:20:48.414 DEBUG uni-meter.input          - Found OBIS code 81 81 C7 82 03 FF
25-04-13 11:20:48.414 DEBUG uni-meter.input          - Found OBIS code 01 00 00 00 09 FF
25-04-13 11:20:48.414 DEBUG uni-meter.input          - Found OBIS code 01 00 01 08 00 FF
25-04-13 11:20:48.414 DEBUG uni-meter.input          - Energy Import (Wh): 7414747.300000001
25-04-13 11:20:48.414 DEBUG uni-meter.input          - Energy Export (Wh): 0.0
25-04-13 11:20:48.414 DEBUG uni-meter.input          - Power (W): 69.0
25-04-13 11:20:48.414 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-04-13 11:20:48.415 TRACE uni-meter.output         - OutputDevice.onNotifyTotalPowerData()
25-04-13 11:20:48.415 DEBUG uni-meter.output         - power phase 0: PowerData[power=23.0, apparentPower=23.0, powerFactor=1.0, current=0.1, voltage=230.0, frequency=50.0]
25-04-13 11:20:48.415 DEBUG uni-meter.output         - power phase 1: PowerData[power=23.0, apparentPower=23.0, powerFactor=1.0, current=0.1, voltage=230.0, frequency=50.0]
25-04-13 11:20:48.415 DEBUG uni-meter.output         - power phase 2: PowerData[power=23.0, apparentPower=23.0, powerFactor=1.0, current=0.1, voltage=230.0, frequency=50.0]
25-04-13 11:20:48.415 TRACE uni-meter.input          - InputDevice.onWrappedOutputDeviceAck()
25-04-13 11:20:48.415 TRACE uni-meter.output         - OutputDevice.onNotifyTotalEnergyData()
25-04-13 11:20:48.415 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=2471582.43, totalProduction=0.0]
25-04-13 11:20:48.415 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=2471582.43, totalProduction=0.0]
25-04-13 11:20:48.415 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=2471582.43, totalProduction=0.0]
25-04-13 11:20:49.017 DEBUG uni-meter.output.udp-server - recv => {"id":0,"method":"EM.GetStatus","params":{"id":0}}
25-04-13 11:20:49.018 TRACE uni-meter.output         - ShellyPro3EM.onWrappedUdpServerNotification()
25-04-13 11:20:49.018 TRACE uni-meter.output         - ShellyPro3EM.onUdpServerDatagramReceived()
25-04-13 11:20:49.018 TRACE uni-meter.output         - Shelly.getUdpClientContext()
25-04-13 11:20:49.018 TRACE uni-meter.output         - received udp datagram from /192.168.188.166:22222: {"id":0,"method":"EM.GetStatus","params":{"id":0}}
25-04-13 11:20:49.018 TRACE uni-meter.output         - ShellyPro3EM.onUdpClientProcessPendingEmGetStatusRequest()
25-04-13 11:20:49.018 TRACE uni-meter.output         - ShellyPro3EM.processUdpRpcRequest()
25-04-13 11:20:49.018 TRACE uni-meter.output         - using default power factor 1.0 for /192.168.188.166
25-04-13 11:20:49.018 TRACE uni-meter.output         - ShellyPro3EM.rpcEmGetStatus()
25-04-13 11:20:49.018 DEBUG uni-meter.output.udp-server - send => {"id":0,"src":"shellypro3em-bc24114eea54","result":{"id":0,"a_current":0.1,"a_voltage":230,"a_act_power":23,"a_aprt_power":23,"a_pf":1,"a_freq":50,"b_current":0.1,"b_voltage":230,"b_act_power":23,"b_aprt_power":23,"b_pf":1,"b_freq":50,"c_current":0.1,"c_voltage":230,"c_act_power":23,"c_aprt_power":23,"c_pf":1,"c_freq":50,"total_current":0.3,"total_act_power":69,"total_aprt_power":69}}
25-04-13 11:20:49.424 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-04-13 11:20:49.424 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-04-13 11:20:49.537 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-04-13 11:20:49.539 TRACE uni-meter.input          - Pulse.handlePulseStatusEntity()
25-04-13 11:20:49.539 DEBUG uni-meter.input          - Found tag OPEN_RESPONSE
25-04-13 11:20:49.539 DEBUG uni-meter.input          - Found tag GET_LIST_RESPONSE
25-04-13 11:20:49.539 DEBUG uni-meter.input          - Found OBIS code 81 81 C7 82 03 FF
25-04-13 11:20:49.539 DEBUG uni-meter.input          - Found OBIS code 01 00 00 00 09 FF
25-04-13 11:20:49.539 DEBUG uni-meter.input          - Found OBIS code 01 00 01 08 00 FF

Woher kommt der Fehler?

Mir ist auch aufgefallen, dass der ERROR nur manchmal auftritt teilweise mit zeitlichen Abständen...

25-04-13 00:14:01.461 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-13 00:16:01.013 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-13 00:17:44.640 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-13 00:19:28.763 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-13 00:22:32.037 ERROR uni-meter.input          - Failed to parse status response: Timeout

sdeigm, 2025-04-13:
Also prinzipiell sieht das bei dir oben gut aus. Du hast zumindest zeitweilig eine Verbindung zur TibberBridge und bekommst auch Werte. Um 11:20:49.018 sendet das uni-meter gerade 69 Watt an den (Marstek?) Speicher. Um 11:20:49.539 wird auch eine Antwort von der TibberBridge verarbeitet (Pulse.onPulseStatusRequestSuccess()).

Kann es sein, daß bei dir einfach die WLAN Verbindung zur TibberBridge im Zählerkasten sehr schwach ist? Kannst du evtl. einmal die Tür vom Zählerschrank offen lassen, so dass die Tibber Bridge "gute Sicht" auf den WLAN AccessPoint/Repeater hat. Oder einmal einen weiteren WLAN Repeater ganz dicht am Zählerkasten testweise installieren?

kopfplatzwunde, 2025-04-14:
Danke, werde heute mal per Verlängerungskabel die Bridge näher zum eigentlichen Pulse platzieren. Die Sendeleistung vom Pulse schwankt sehr stark, kann sein das hier, dass Problem liegt und einige Daten auf der Strecke bleiben. Auf der Seite von @marq24 Tibber Pulse IR LOCAL schreibt er auch das man wohl die update frequency vom Pulse noch beeinflussen kann, in den man den IR Kopf etwas dreht.

Ich habe aber noch ein anderes Problem, im photovoltaikforum habe ich einige Antworten von dir gelesen, dass die Möglichkeit besteht, die Summe der Phasen (A+B+C) auf eine Phase zu legen. Der Marstek kann scheinbar die Werte bei gleichen Phasen-Werten (A, B, C) nicht interpretieren und reagiert im Eigenverbrauchsmodus auch nicht. Bisher kann ich den Speicher nur manuell steuern.

Wie kann ich in der Conf beim Tibber Pulse die Gesamtsumme auf eine Phase fix zuweisen? Ich benötige nicht die Drittelung auf alle 3-phasen sondern die Gesamtsumme auf eine Phase z.B. power-l1

uni-meter {
output = "uni-meter.output-devices.shelly-pro3em"
input = "uni-meter.input-devices.tibber-pulse"
output-devices {
    shelly-pro3em {
    udp-port = 1010
                    }
            }    
input-devices {
    tibber-pulse {
        url = "http://192.168.188.101"
        node-id = 1
        user-id = "admin"
        password = "XXXX-XXXX"
     }
  }
}

Oder muss ich den Umweg über MQTT/IO-Broker gehen, was ich tendenziell vermeiden will.

Kannst du mir hier helfen?

sdeigm, 2025-04-14:
Eine Phase ist für Tibber leider im Augenblick nicht möglich. Das werde ich in der nächsten Version konfigurierbar machen. Im Augenblick wir das immer gleich auf die 3 Phasen aufgeteilt.

sdeigm, 2025-04-14:
Wenn du mir sagst, wie du das uni-meter betreibst (Docker, echter Rechner, HomeAssistant) kann ich dir heute Abend eine Entwicklerversion bauen, die das behebt ...

kopfplatzwunde, 2025-04-14:
Wow das wäre Mega, würde mich freuen.... ich betreibe den uni-meter in einen LXC Container oder einer VM, beide Varianten laufen stabil, die Basis ist ein Debian Lunix ... quasi als echter Rechner zu interpretieren ... 

Wie komme ich dann an die Entwicklerversion?

Nochmal vielen Dank für deine schnelle Hilfe...

sdeigm, 2025-04-14:
Es gibt jetzt eine Docker-Container uni-meter:1.1.6-SNAPSHOT, den müsstest du ausführen. In der uni-meter.conf kannst du Tibber jetzt auf mono-phase Mode schalten und sagen dass er Phase 1 benutzen soll:

```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"
  
  input = "uni-meter.input-devices.tibber-pulse"

  input-devices {
    tibber-pulse {
      url = "<tibber-device-url>"
      node-id = 1
      user-id = "admin"
      password = "<tibber-device-password>"
      power-phase-mode = "mono-phase"
      power-phase = "l1"
    }
  }
}
```

Alles noch völlig ungetestet, da ich keine Tibber Bridge habe :-)

kopfplatzwunde, 2025-04-14:
Hi, kannst du mir das als "uni-meter 1.1.6-SNAPSHOT.tgz" bereitstellen, ich arbeite nicht mit den Docker Containern ich müsste mir sonst erst Docker einrichten und eine Umgebung dafür aufbauen ... sorry

sdeigm, 2025-04-14:
Hier jetzt als tgz: [uni-meter-1.1.6-SNAPSHOT.tgz](https://github-registry-files.githubusercontent.com/898342116/b8264500-1971-11f0-9eb7-8d63520620a1?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20250414%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250414T184822Z&X-Amz-Expires=300&X-Amz-Signature=fdf3042bd1c7e84fd11970b8b0980184fa23097b9ed41523f13315080f3209c5&X-Amz-SignedHeaders=host&response-content-disposition=filename%3Duni-meter-1.1.6-20250414.184637-1.tgz&response-content-type=application%2Foctet-stream)

kopfplatzwunde, 2025-04-14:
Leider funktioniert der Link nicht ... "AccessDenied" ...

sdeigm, 2025-04-14:
Oh, das passiert wohl wenn man den Link kopiert wenn man eingeloggt ist. Das *.tgz ist in den Packages zu diesem Projekt:

https://github.com/sdeigm/uni-meter/packages/2336213

kopfplatzwunde, 2025-04-14:
Besten Dank probiere das morgen mal aus und gebe dir Feedback in deinen neuen issue "Support mono-phase/tri-phase mode for tibbern pulse"

Habe heute auch mal die WLan Verbindung der Bridge getestet, also Bridge direkt zum Pulse .... Leider treten die ERROR weiterhin auf .. 

der ERROR: Failed to parse status response: wrong crc ... der wrong crc kommt wohl wenn der IR Kopf nicht ordentlich von Zähler liest (also Lesefehler), sobald ich am Kopf gedreht habe z.B. ....

25-04-14 22:15:56.111 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-14 22:20:21.364 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-14 22:22:20.782 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-14 22:26:31.988 ERROR uni-meter.input          - Failed to parse status response: wrong crc
25-04-14 22:27:47.742 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-04-14 22:29:19.230 ERROR uni-meter.input          - Failed to parse status response: Timeout

Den Timeout kann ich mir aktuell noch nicht erklären, vor allem in den komischen zeitlichen Abständen. Werde mal die Signalstärken vom Pulse und der Bridge überwachen. Durch leichte Anpassungen konnte ich zumindest mal die Verbindung zum Pulse optimieren.

closed 2025-04-15 by sdeigm

BenjiBizz, 2025-05-22:
> Den Timeout kann ich mir aktuell noch nicht erklären, vor allem in den komischen zeitlichen Abständen. Werde mal die Signalstärken vom Pulse und der Bridge überwachen. Durch leichte Anpassungen konnte ich zumindest mal die Verbindung zum Pulse optimieren.

Servus ! 

Hab die selben Probleme. Hast du das optimieren können mit dem timeout?

Den CRC Fehler: den bekomme ich ab und an mal. 
Ich vernachlässige das mal weils zu 95% klappt.

kopfplatzwunde, 2025-05-23:
Hi, von welchen Problem sprichst du genau ... vom TIMEOUT oder vom CRC Fehler? ... der CRC Fehler kommt bei mir eher selten vor, damit kann ich leben ... bei mir sind die TIMEOUT Fehler eher die Häufigkeit ... das liegt aber bei mir wahrscheinlich an der Verbindung zw. Bridge und Pulse ... die scheint die Daten immer mal zu verlieren ... hängt halt daran das der Pulse im Zählerschrank im Hausflur steht und die Bridge, getrennt durch 2 Türen, im Keller ... und dann noch im 2,4ghz WLAN was sowieso anfällig ist ...

Ich habe Leider keine andere Möglichkeit, ohne aufwändig in die Infrastruktur einzugreifen ... somit muss ich mit den TIMEOUT's leben ... was aber auch nicht so gravierend ist, dann werden halt mal paar Daten nicht übertragen .. insgesamt läuft es und der Marstek bekommt seine Informationen ...

SolarTester, 2025-05-26:
Ich habe ebenfalls diese Timeouts. 
An der Verbindung der Bridge zum Netzwerk kann es aber nicht liegen, da ich mir über das Webinterface der Bridge ja die Infos "Last seen" und "Last data" anzeigen lassen kann und die immer unter 3 Sekunden bleiben - auch wenn im "uni-meter"-Webinterface sich der Wert bei "power" über mehr als 30 Sekunden nicht ändert. 
Und diese Aussetzer sind dann zeitgleich in der Tibber-App zu sehen. Der Unterschied ist hier, dass die Tibber-App wieder früher mit den nächsten Messpunkten/Aktualisierungen weitermacht. Da sind es vielleicht 10 Sekunden. Beim uni-meter sind das über 30 Sekunden.
Es scheint mir so, als ob beim uni-meter entweder ein höherer Timeout bei den Request eingestellt ist oder möglicherweise wird mehrmals versucht das fehlende Datenpaket abzuholen (Anzahl der Retries).

@sdeigm Könnte man die Poll-Frequenz und den Timeout/Retries in der Tibber-Config einstellbar machen? Bei mir sind die Aktualisierungszeiten in der Bridge immer zwischen 2 und 3 Sekunden. Da würde mir eine Poll-Rate von 3 Sekunden reichen.
Und bei einem Timeout von 5 Sekunden würden dann ca. 2 Pakete verloren gehen, was absolut kein Problem ist. Bei 40 Sekunden ohne Aktualisierung kann das bei der Häufigkeit der Timeouts schon etwas zwischen Laden und Entladen des Speichers ausmachen. Wobei ich aber sagen muss - das ist Jammern auf hohem Niveau. Dein Programm hilft vielen ein weiteres Gerät im Schaltschrank und den Einsatz eines Elektrikers zu vermeiden und dafür möchte ich dir herzlich danken.

SolarTester, 2025-05-26:
Bei mir sind die Timeouts fast jede Minute.
Das Polling-Intervall habe ich von 1s bis 10s ausprobiert - macht bei mir keinen Unterschied.
Auf alle Fälle scheinen diese Unterbrechungen nicht ein Problem vom uni-meter zu sein, da ich die gleichen Verzögerungen in der Tibber-App am Smartphone sehe. Auch da wird zum gleichen Zeitpunkt nichts mehr aktualisiert. Die Wiederaufnahme geschieht halt bei der Tibber-App nach ca. 10 Sekunden und beim uni-meter nach genau 30s (sieht man im zweiten Log).

```
25-05-26 23:50:44.696 INFO  uni-meter                - ###################################################################
25-05-26 23:50:44.728 INFO  uni-meter                - # Universal electric meter converter 1.1.10 (2025-05-19 06:27:22) #
25-05-26 23:50:44.729 INFO  uni-meter                - ###################################################################
25-05-26 23:50:44.731 INFO  uni-meter                - initializing actor system
25-05-26 23:50:49.188 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-05-26 23:50:50.861 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-05-26 23:50:50.945 INFO  uni-meter.controller     - creating TibberPulse input device
25-05-26 23:50:53.580 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-05-26 23:50:55.153 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-05-26 23:51:51.012 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-05-26 23:52:49.474 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-05-26 23:53:46.498 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-05-26 23:54:44.869 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-05-26 23:55:27.331 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-05-26 23:56:41.270 ERROR uni-meter.input          - Failed to parse status response: Timeout
```

Hier mal mit einem anderen Debug-Level

```
25-05-26 23:46:29.073 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-05-26 23:46:29.073 DEBUG uni-meter.output         - power phase 0: PowerData[power=331.0, apparentPower=331.0, powerFactor=1.0, current=1.4391304347826086, voltage=230.0, frequency=50.0]
25-05-26 23:46:29.074 DEBUG uni-meter.output         - power phase 1: PowerData[power=0.0, apparentPower=0.0, powerFactor=1.0, current=0.0, voltage=230.0, frequency=50.0]
25-05-26 23:46:29.075 DEBUG uni-meter.output         - power phase 2: PowerData[power=0.0, apparentPower=0.0, powerFactor=1.0, current=0.0, voltage=230.0, frequency=50.0]
25-05-26 23:46:29.076 TRACE uni-meter.output         - OutputDevice.onNotifyTotalEnergyData()
25-05-26 23:46:29.076 TRACE uni-meter.input          - InputDevice.onWrappedOutputDeviceAck()
25-05-26 23:46:29.077 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=579301.5, totalProduction=1265608.2]
25-05-26 23:46:29.078 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=579301.5, totalProduction=1265608.2]
25-05-26 23:46:29.079 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=579301.5, totalProduction=1265608.2]
25-05-26 23:46:30.091 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-05-26 23:46:30.093 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-05-26 23:46:30.254 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-05-26 23:46:30.256 TRACE uni-meter.input          - Pulse.handlePulseStatusEntity()
25-05-26 23:46:30.258 DEBUG uni-meter.input          - Found tag OPEN_RESPONSE
25-05-26 23:46:30.259 DEBUG uni-meter.input          - Found tag GET_LIST_RESPONSE
25-05-26 23:46:30.261 DEBUG uni-meter.input          - Found OBIS code 01 00 60 32 01 01
25-05-26 23:46:30.262 DEBUG uni-meter.input          - Found OBIS code 01 00 60 01 00 FF
25-05-26 23:46:30.264 DEBUG uni-meter.input          - Found OBIS code 01 00 01 08 00 FF
25-05-26 23:46:30.266 DEBUG uni-meter.input          - Energy Import (Wh): 1737904.5
25-05-26 23:46:30.267 DEBUG uni-meter.input          - Energy Export (Wh): 3796824.6
25-05-26 23:46:30.268 DEBUG uni-meter.input          - Power (W): 331.0
25-05-26 23:46:30.269 TRACE uni-meter.input          - Pulse.notifyPowerData()
25-05-26 23:46:30.270 TRACE uni-meter.input          - Pulse.notifyEnergyData()
25-05-26 23:46:30.270 TRACE uni-meter.output         - OutputDevice.onNotifyPhasesPowerData()
25-05-26 23:46:30.270 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-05-26 23:46:30.270 DEBUG uni-meter.output         - power phase 0: PowerData[power=331.0, apparentPower=331.0, powerFactor=1.0, current=1.4391304347826086, voltage=230.0, frequency=50.0]
25-05-26 23:46:30.272 DEBUG uni-meter.output         - power phase 1: PowerData[power=0.0, apparentPower=0.0, powerFactor=1.0, current=0.0, voltage=230.0, frequency=50.0]
25-05-26 23:46:30.273 DEBUG uni-meter.output         - power phase 2: PowerData[power=0.0, apparentPower=0.0, powerFactor=1.0, current=0.0, voltage=230.0, frequency=50.0]
25-05-26 23:46:30.275 TRACE uni-meter.input          - InputDevice.onWrappedOutputDeviceAck()
25-05-26 23:46:30.275 TRACE uni-meter.output         - OutputDevice.onNotifyTotalEnergyData()
[... cut ...]
25-05-26 23:47:02.638 DEBUG uni-meter.input          - Energy Import (Wh): 1737907.6
25-05-26 23:47:02.639 DEBUG uni-meter.input          - Energy Export (Wh): 3796824.6
25-05-26 23:47:02.640 DEBUG uni-meter.input          - Power (W): 329.0
25-05-26 23:47:02.641 TRACE uni-meter.input          - Pulse.notifyPowerData()
25-05-26 23:47:02.641 TRACE uni-meter.input          - Pulse.notifyEnergyData()
25-05-26 23:47:02.642 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
```

sdeigm, 2025-05-27:
Moin ihr beiden, ich bin der Sache mal auf den Grund gegangen, woher die Verzögerung der Wiederaufnahme des Pollings beim uni-meter kommt. Das sind Parameter der zugrunde liegenden HTTP-Bibliothek, die man zum Glück konfigurieren kann. Ihr könnt die folgende Sektion ausserhalb der uni-meter Sektion in eure Config-Datei einfügen:

```
pekko.http {
  client {
    connection-timeout = 5s
  }
   
  host-connection-pool {
    max-connection-backoff = 5s
  }
}

uni-meter {
  # ...
}  
```

Der erste Parameter ist der Timeout beim Aufbau der Verbindung. Der zweite Parameter begrenzt die Wartezeit, die die Bibliothek automatisch nach fehlgeschlagenen Request vor erneuten Versuchen einbaut. Standardmässig verdoppelt sich das nach jedem fehlgeschlagenen Versuch. Oben wird das dann auf 5 Sekunden begrenzt. Mit den Parametern solltet ihr euch das wie gewünscht einstellen können.

BenjiBizz, 2025-05-27:
Danke für die Ergänzung ! 
Fehler kommt zwar Immernoch aber nicht mehr so arg häufig. 
Aber dennoch schon noch oft. 

Naja funktioniert ja :) danke dir auf jedenfall für dieses AddOn!! Hast mir den Elektriker plus den Shelly gespart :)

SolarTester, 2025-05-27:
Ich habe das mal in die config aufgenommen.
Leider macht das bei mir gar keinen Unterschied.
Ich habe immer noch einen Timeout-Wartezeit von 30 Sekunden.

Möglicherweise habe ich es ja auch falsch in die Config eingetragen oder in die falsche Config.
Ich habe es mit 5s/5s und 4s/1s und 10s/10s versucht und immer die 30s Wartezeit.

```
pekko.http {
  client {
    connection-timeout = 10s
  }

  host-connection-pool {
    max-connection-backoff = 10s
  }
}

uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.tibber-pulse"

  http-server {
    port = 80
  }

  output-devices {
    shelly-pro3em {
      udp-port = 1010
    }
  }

  input-devices {
    tibber-pulse {
      url = "http://192.168.xxx.xxx"
      node-id = 1
      user-id = "admin"
      password = "xxdxx-xxxx"
      polling-interval = 2s

      # The Marstek storage needs input data on a single phase.
      # This can be controlled by the configuration options below
      power-phase-mode = "mono-phase"
      power-phase = "l1"
    }
  }
}
```

```
25-05-27 22:14:44.633 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-05-27 22:14:44.633 TRACE uni-meter.output         - OutputDevice.onNotifyPhasesPowerData()
25-05-27 22:14:44.634 DEBUG uni-meter.output         - power phase 0: PowerData[power=476.0, apparentPower=476.0, powerFactor=1.0, current=2.0695652173913044, voltage=230.0, frequency=50.0]
25-05-27 22:14:44.635 DEBUG uni-meter.output         - power phase 1: PowerData[power=0.0, apparentPower=0.0, powerFactor=1.0, current=0.0, voltage=230.0, frequency=50.0]
25-05-27 22:14:44.636 DEBUG uni-meter.output         - power phase 2: PowerData[power=0.0, apparentPower=0.0, powerFactor=1.0, current=0.0, voltage=230.0, frequency=50.0]
25-05-27 22:14:44.638 TRACE uni-meter.output         - OutputDevice.onNotifyTotalEnergyData()
25-05-27 22:14:44.638 TRACE uni-meter.input          - InputDevice.onWrappedOutputDeviceAck()
25-05-27 22:14:44.639 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=580489.83, totalProduction=1277132.57]
25-05-27 22:14:44.640 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=580489.83, totalProduction=1277132.57]
25-05-27 22:14:44.641 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=580489.83, totalProduction=1277132.57]
25-05-27 22:14:46.648 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-05-27 22:14:46.652 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-05-27 22:14:46.806 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-05-27 22:14:46.807 TRACE uni-meter.input          - Pulse.handlePulseStatusEntity()

25-05-27 22:15:16.831 ERROR uni-meter.input          - Failed to parse status response: Timeout

25-05-27 22:15:16.833 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-05-27 22:15:18.857 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-05-27 22:15:18.861 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-05-27 22:15:19.065 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-05-27 22:15:19.080 TRACE uni-meter.input          - Pulse.handlePulseStatusEntity()
25-05-27 22:15:19.088 DEBUG uni-meter.input          - Found tag OPEN_RESPONSE
25-05-27 22:15:19.089 DEBUG uni-meter.input          - Found tag GET_LIST_RESPONSE
25-05-27 22:15:19.091 DEBUG uni-meter.input          - Found OBIS code 01 00 60 32 01 01
25-05-27 22:15:19.093 DEBUG uni-meter.input          - Found OBIS code 01 00 60 01 00 FF
25-05-27 22:15:19.094 DEBUG uni-meter.input          - Found OBIS code 01 00 01 08 00 FF
25-05-27 22:15:19.098 DEBUG uni-meter.input          - Energy Import (Wh): 1741474.0
25-05-27 22:15:19.099 DEBUG uni-meter.input          - Energy Export (Wh): 3831397.7
25-05-27 22:15:19.100 DEBUG uni-meter.input          - Power (W): 473.0
25-05-27 22:15:19.101 TRACE uni-meter.input          - Pulse.notifyPowerData()
25-05-27 22:15:19.102 TRACE uni-meter.input          - Pulse.notifyEnergyData()
25-05-27 22:15:19.103 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
```

BenjiBizz, 2025-05-28:
Wie bekomme ich dieses snapshot in Home Assistant ? 
Wäre schön wenn mir jemand eine kurze Anleitung schreiben könnte. Danke im Vorraus :)

kopfplatzwunde, 2025-05-28:
Also ich habe jetzt ne weile laufen lassen mit den Einstellungen (pooling und pekko.http) und wie auch bei @BenjiBizz  und @SolarTester es hat sich an den Timeouts nur wenig verändert ... 

@sdeigm werde mal den Snapshot testen und ebenfalls den Trace erzeugen bzw. bereitstellen ... bin gespannt

SolarTester, 2025-05-28:
@sdeigm Jetzt hat sich auf alle Fälle etwas verändert 👍 
Das polling-interval = 2s bei diesem Log.

Für mich scheint das jetzt exakt so zu funktionieren wie bei der Tibber-App.
Keine Ahnung warum die Bridge jede Minute eine Denkpause braucht.
Wenn wenigsten auch im Webinterface der Bridge immer wieder diese Pausen auftreten würden, dann würde ich die WLAN-Verbindung als Problem vermuten und mal die Bridge näher am Router einstecken. Aber das läuft problemlos und wird laufend aktualisiert.

Tolle Arbeit @sdeigm und ein super Projekt.

```
25-05-28 08:08:58.932 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-05-28 08:08:58.936 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-05-28 08:08:59.033 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-05-28 08:08:59.035 TRACE uni-meter.input          - Pulse.onStrictEntity()
25-05-28 08:08:59.037 DEBUG uni-meter.input          - Found tag OPEN_RESPONSE
25-05-28 08:08:59.038 DEBUG uni-meter.input          - Found tag GET_LIST_RESPONSE
25-05-28 08:08:59.040 DEBUG uni-meter.input          - Found OBIS code 01 00 60 32 01 01
25-05-28 08:08:59.042 DEBUG uni-meter.input          - Found OBIS code 01 00 60 01 00 FF
25-05-28 08:08:59.044 DEBUG uni-meter.input          - Found OBIS code 01 00 01 08 00 FF
25-05-28 08:08:59.047 DEBUG uni-meter.input          - Energy Import (Wh): 1744525.4000000001
25-05-28 08:08:59.048 DEBUG uni-meter.input          - Energy Export (Wh): 3831543.8000000003
25-05-28 08:08:59.049 DEBUG uni-meter.input          - Power (W): 54.0
25-05-28 08:08:59.050 TRACE uni-meter.input          - Pulse.notifyPowerData()
25-05-28 08:08:59.050 TRACE uni-meter.input          - Pulse.notifyEnergyData()
25-05-28 08:08:59.051 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-05-28 08:08:59.051 TRACE uni-meter.output         - OutputDevice.onNotifyPhasesPowerData()
25-05-28 08:08:59.052 DEBUG uni-meter.output         - power phase 0: PowerData[power=54.0, apparentPower=54.0, powerFactor=1.0, current=0.23478260869565218, voltage=230.0, frequency=50.0]
25-05-28 08:08:59.053 DEBUG uni-meter.output         - power phase 1: PowerData[power=0.0, apparentPower=0.0, powerFactor=1.0, current=0.0, voltage=230.0, frequency=50.0]
25-05-28 08:08:59.055 DEBUG uni-meter.output         - power phase 2: PowerData[power=0.0, apparentPower=0.0, powerFactor=1.0, current=0.0, voltage=230.0, frequency=50.0]
25-05-28 08:08:59.058 TRACE uni-meter.input          - InputDevice.onWrappedOutputDeviceAck()
25-05-28 08:08:59.058 TRACE uni-meter.output         - OutputDevice.onNotifyTotalEnergyData()
25-05-28 08:08:59.059 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=581508.47, totalProduction=1277181.27]
25-05-28 08:08:59.062 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=581508.47, totalProduction=1277181.27]
25-05-28 08:08:59.063 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=581508.47, totalProduction=1277181.27]

25-05-28 08:09:01.075 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-05-28 08:09:01.080 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-05-28 08:09:01.187 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-05-28 08:09:01.191 TRACE uni-meter.input          - Pulse.onStrictEntity()
25-05-28 08:09:01.246 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-05-28 08:09:01.249 DEBUG uni-meter.input          - Raw data:
25-05-28 08:09:01.251 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-05-28 08:09:03.274 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-05-28 08:09:03.278 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-05-28 08:09:03.441 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-05-28 08:09:03.443 TRACE uni-meter.input          - Pulse.onStrictEntity()
25-05-28 08:09:03.496 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-05-28 08:09:03.501 DEBUG uni-meter.input          - Raw data:
25-05-28 08:09:03.505 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-05-28 08:09:05.524 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-05-28 08:09:05.526 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-05-28 08:09:05.689 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-05-28 08:09:05.691 TRACE uni-meter.input          - Pulse.onStrictEntity()
25-05-28 08:09:05.794 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-05-28 08:09:05.800 DEBUG uni-meter.input          - Raw data:
25-05-28 08:09:05.803 TRACE uni-meter.input          - Pulse.startNextPulseStatusPollingTimer()
25-05-28 08:09:07.824 TRACE uni-meter.input          - Pulse.onExecuteNextPulseStatusPolling()
25-05-28 08:09:07.827 TRACE uni-meter.input          - Pulse.executePulseStatusPolling()
25-05-28 08:09:07.950 TRACE uni-meter.input          - Pulse.onPulseStatusRequestSuccess()
25-05-28 08:09:07.951 TRACE uni-meter.input          - Pulse.onStrictEntity()
25-05-28 08:09:08.004 ERROR uni-meter.input          - Failed to parse status response: Timeout
25-05-28 08:09:08.009 DEBUG uni-meter.input          - Raw data:
[... cut ...]
25-05-28 08:09:14.728 TRACE uni-meter.output         - OutputDevice.onNotifyTotalEnergyData()
25-05-28 08:09:14.729 DEBUG uni-meter.output         - energy phase 0: EnergyData[totalConsumption=581508.57, totalProduction=1277181.27]
25-05-28 08:09:14.730 DEBUG uni-meter.output         - energy phase 1: EnergyData[totalConsumption=581508.57, totalProduction=1277181.27]
25-05-28 08:09:14.731 DEBUG uni-meter.output         - energy phase 2: EnergyData[totalConsumption=581508.57, totalProduction=1277181.27]
```

sdeigm, 2025-05-28:
@Alle die hier geantwortet haben, 

> Für mich scheint das jetzt exakt so zu funktionieren wie bei der Tibber-App. Keine Ahnung warum die Bridge jede Minute eine Denkpause braucht. Wenn wenigsten auch im Webinterface der Bridge immer [...]

Das war die ganze Zeit mein Denkfehler. Aufgrund der Fehlermeldungen "Timeout" und "CRC-Fehler" haben ich an Netzwerkprobleme gedacht. Das ist aber überhaupt nicht der Fall. Die Tibber-Bridge scheint so zu arbeiten, daß sie über Infrarot die Zählerdaten ausliest und die gesamten empfangen Raw-Daten einfach per HTTP Request weitergibt. Wenn die Daten dann aufgrund von Übertragungsfehlern auf der Infrarotschnittstelle unvollständig oder fehlerhaft sind kommt es dann auf der Client-Seite - also hier uni-meter - zu CRC-Fehlern oder sogar zu diesem "Timeout" Fehler. Dieser Timeout ist natürlich irreführend, weil so wie es im uni-meter läuft der Bibliothek alle Daten zur Verfügung stehen und es eigentlich gar nicht zu einem Timeout kommen kann. Die Bibliothek ist denke ich aber eigentlich für die USB-Schnittstelle gedacht und dann könnten sehr wohl noch Daten kommen.

Wenn ihr also weniger Fehler haben wollt, müsstet ihr versuchen die Verbindung Stromzähler->Pulse verbessern. Evtl. ist der Infrarotsender schon altersschwach geworden (das soll bei einigen Zählern vorkommen) dann könnt ihr nichts machen oder aber eure Bridge sitzt vielleicht nicht ganz optimal auf den beiden optischen Ausgängen? Der zweite Fall passiert bei mir manchmal mit meinem USB-IR-Lesekopf. Dann bekomme ich auch oft Pausen zwischen den Werten die mir der VZ-Logger liefert. Bei mir hilft es dann immer den IR-Lesekopf wieder exact auszurichten. Keine Ahnung wie das bei euch aussieht, ich kenne diese Pulse-Bridge nicht.
 
> Ein Frage zum Log hätte ich noch. Wenn der uni-meter als Dienst läuft, wird dann das Log als Datei gespeichert? Bei mir werden ja viele Zeilen geschrieben auch wenn ich den Log-Level auf ERROR stelle. Nicht, dass mir da ein Verzeichnis vollläuft.

Ja, das Log wird als Datei gepeichert /var/log/uni-meter.log. Wenn du die Ausgabe nicht haben willst kannst du das in der Datei /etc/systemd/system/uni-meter.service einfach abschalten indem du sie mit einem # vor den Zeilen die mit StandardOutput und StandardError beginnen auskommentierst. Ich werde die Logmeldungen aber auch irgendwie versuchen abzufangen bzw. zu limitieren, da ich ja jetzt weiß, daß sie sehr oft auftreten.

@BenjiBizz eine HA-Addon Snapshot bereit zu stellen ist sehr aufwendig, erfordert auch eine komplett andere, lokale Installation. Da musst du dich bitte noch gedulden, bis ich dies hier als offizielle Version verfügbar mache.

kopfplatzwunde, 2025-05-28:
> > Ja das Log wird als Datei geschrieben ... aber keine Panik so schnell läuft die nicht voll .. bei mir läuft der uni-meter nativ bzw. als LXC Container ... du findest den Log hier /var/log/uni-meter.log hat [@sdeigm](https://github.com/sdeigm) eigentlich auch beschrieben in der Doku .... wenn du es im Container laufen hast dann musst halt da drin schauen ..
> 
> Ich habe da nur mal so die Zeilenzahl überschlagen und das wären 60 x 24 x 10 x 365 = 5.256.000 pro Jahr 😄
> 
> > Wenn ihr also weniger Fehler haben wollt, müsstet ihr versuchen die Verbindung Stromzähler->Pulse verbessern. Evtl. ist der Infrarotsender schon altersschwach geworden (das soll bei einigen Zählern vorkommen) dann könnt ihr nichts machen oder aber eure Bridge sitzt vielleicht nicht ganz optimal auf den beiden optischen Ausgängen? Der zweite Fall passiert bei mir manchmal mit meinem USB-IR-Lesekopf. Dann bekomme ich auch oft Pausen zwischen den Werten die mir der VZ-Logger liefert. Bei mir hilft es dann immer den IR-Lesekopf wieder exact auszurichten. Keine Ahnung wie das bei euch aussieht, ich kenne diese Pulse-Bridge nicht.
> 
> da werde ich mal etwas rumspielen - danke für den Hinweis in diese Richtung.

@SolarTester naja ich habe den Log auf "INFO" umgestellt, da werden natürlich weniger Sätze geschrieben .. den "TRACE" habe ich nur mal temporär laufen ... ansonsten kannst den LOG auch manuell mal cleanen und vorher als Backup runter ziehen, damit dir die Info nicht verloren geht ...

kopfplatzwunde, 2025-05-28:
> Das war die ganze Zeit mein Denkfehler. Aufgrund der Fehlermeldungen "Timeout" und "CRC-Fehler" haben ich an Netzwerkprobleme gedacht. Das ist aber überhaupt nicht der Fall. Die Tibber-Bridge [...]

@sdeigm ich habe eigentlich schon länger das gleich vermutet .... mein Pulse läuft jetzt schon seit ca. 1 1/2 Jahren 24H ... ich habe mir mal die Batteriespannung im Pulse angeschaut, die geht auch langsam die neige ... ich werde die mal zeitnahe tauschen und beobachten ob das eventuell auch eine Verbesserung bringt bei der Verbindung zw. Bridge und Pulse ... 

Das der Infrarotsender altersschwach wird war mir neu ... aber eigentlich nach 1 1/2 Jahren sollte das ja noch nicht sein ... da muss ich ja mal überlegen ob ich Tibber dazu mal anschreibe, wäre ja dann eigentlich ein Geräte defekt .. 

Das mit den Ausrichten habe ich schon probiert .. über das Webinterface der Bridge kann man eigentlich ganz gut die Signalstärke und die Last data Frequenzen überwachen ... aber das ist echt schwierig hier besser Ergebnisse hin zu bekommen, da kann jeder Millimeter schon schlechtere Ergebnisse bringen ... bei mir bekomme ich die besten Ergebnisse auf einer 5 - 4 Uhr Stellung so wie @marq24 das beschrieben hat ... aber ist absolutes gefummel .. vielleicht versuche ich noch mal mein Glück und probiere es .. aber bei mir treten die Timeouts nun seltener auf, teilweise alle 10 min nur ....

BenjiBizz, 2025-05-28:
> [@BenjiBizz](https://github.com/BenjiBizz) eine HA-Addon Snapshot bereit zu stellen ist sehr aufwendig, erfordert auch eine komplett andere, lokale Installation. Da musst du dich bitte noch gedulden, bis ich dies hier als offizielle Version verfügbar mache.

Okay :) dann warte ich mal. Danke dir !!

SolarTester, 2025-05-30:
@sdeigm 
Ich habe mal alles neu installiert und jetzt funktioniert alles ohne irgendwelche Änderungen an den Parametern.
Danke schön

kopfplatzwunde, 2025-05-30:
> [@sdeigm](https://github.com/sdeigm) Ich habe mal alles neu installiert und jetzt funktioniert alles ohne irgendwelche Änderungen an den Parametern. Danke schön

Was genau hast du den neu installiert, dass es jetzt funktionieren soll?

SolarTester, 2025-05-30:
Ich habe die neue Version 1.1.11 installiert und in der config alle Änderungen der Parameter rausgenommen (nur die Eintragung für Passwort und IP sind angepasst). Jetzt funktioniert alles so wie auch in der Tibber App.
Die Timeouts kommen natürlich weiterhin aber ohne Wartezeit und sobald wieder vernünftige Daten vom Tibber kommen werden sie auch im uni-meter angezeigt.
Die Änderung mit dem Log hätte ich nicht unbedingt gebraucht. Ich hätte das Log halt ausgeschaltet. Und möglicherweise will man ja auch mal alle Zeilen im Log haben um zu sehen in welchen Zeitabständen dieser Timeout passiert. Aber es ist ein super Feature damit einem nicht die Log-Datei zu groß wird.
Also eine bessere Umsetzung kann man von keinem Entwickler erwarten - Super Arbeit @sdeigm .