sdeigm/uni-meter #104: call of home assistant service pyscript.uni_meter_mdns_register failed with status: 400 Bad Request

romibe, 2025-04-12:
Trying to geti t work with Hoymiles HM2. Getting

[15:05:42] INFO: using custom configuration file: uni-meter.conf
25-04-12 15:05:45.918 INFO  uni-meter                - ##################################################################
25-04-12 15:05:45.955 INFO  uni-meter                - # Universal electric meter converter 1.1.5 (2025-04-11 08:31:17) #
25-04-12 15:05:45.963 INFO  uni-meter                - ##################################################################
25-04-12 15:05:45.963 INFO  uni-meter                - initializing actor system
25-04-12 15:05:48.575 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-04-12 15:05:50.250 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-04-12 15:05:50.345 INFO  uni-meter.controller     - creating HomeAssistant input device
25-04-12 15:05:56.561 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-04-12 15:05:57.397 INFO  uni-meter.mdns.ha        - call of home assistant service pyscript.uni_meter_mdns_register failed with status: 400 Bad Request

Running in docker on synology.

Script is in /homeassistant/config/pyscript/uni-meter-mdns.py

s-Home app does not see shelly when trying to add to.
Please assist. Thank u.

sdeigm, 2025-04-12:
Are you running the uni-meter as home assistant addon? If yes, do you have installed the pyscript addon? That is necessary. The pyscript addon picks up the uni-meter-mdns.py file from /config/pyscript/uni-meter-mdns.py on the machine where the home assistant is running and provides the service uni-meter is calling.

romibe, 2025-04-13:
yes. addon.
installed pyscript  from #https://github.com/custom-components/pyscript

now:

#s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[11:46:03] INFO: using custom configuration file: uni-meter.conf
25-04-13 11:46:07.502 INFO  uni-meter                - ##################################################################
25-04-13 11:46:07.531 INFO  uni-meter                - # Universal electric meter converter 1.1.5 (2025-04-11 08:31:17) #
25-04-13 11:46:07.532 INFO  uni-meter                - ##################################################################
25-04-13 11:46:07.532 INFO  uni-meter                - initializing actor system
25-04-13 11:46:10.106 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-04-13 11:46:12.751 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-04-13 11:46:13.110 INFO  uni-meter.controller     - creating HomeAssistant input device
25-04-13 11:46:30.658 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-04-13 11:46:33.096 INFO  uni-meter.mdns.ha        - call of home assistant service pyscript.uni_meter_mdns_register failed with status: 502 Bad Gateway
25-04-13 11:46:33.553 ERROR uni-meter.input          - failed to process http response: For input string: "unknown"

sdeigm, 2025-04-13:
I can confirm that I had exactly the same problem when I just rebootet my home assistant system. When the uni-meter came up, it throw exactly the same error (502 Bad Gateway). Then I just restarted the uni-meter addon and the error had gone. I guess it is related that the pyscript extension had not come up when the uni-meter started initially:

Log directly after boot: 

```
[17:38:34] INFO: using custom configuration file: uni-meter.conf
25-04-13 17:38:40.754 INFO  uni-meter                - ##################################################################
25-04-13 17:38:40.783 INFO  uni-meter                - # Universal electric meter converter 1.1.5 (2025-04-11 08:31:17) #
25-04-13 17:38:40.784 INFO  uni-meter                - ##################################################################
25-04-13 17:38:40.786 INFO  uni-meter                - initializing actor system
25-04-13 17:38:44.697 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-04-13 17:38:46.361 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-04-13 17:38:46.477 INFO  uni-meter.controller     - creating GenericHttp input device
25-04-13 17:38:50.389 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-04-13 17:38:52.191 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-04-13 17:38:53.101 INFO  uni-meter.mdns.ha        - call of home assistant service pyscript.uni_meter_mdns_register failed with status: 502 Bad Gateway
```

Then  after manual restart of the uni-meter addon:

```
[17:59:50] INFO: using custom configuration file: uni-meter.conf
25-04-13 17:59:56.274 INFO  uni-meter                - ##################################################################
25-04-13 17:59:56.304 INFO  uni-meter                - # Universal electric meter converter 1.1.5 (2025-04-11 08:31:17) #
25-04-13 17:59:56.306 INFO  uni-meter                - ##################################################################
25-04-13 17:59:56.307 INFO  uni-meter                - initializing actor system
25-04-13 17:59:59.907 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-04-13 18:00:01.687 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-04-13 18:00:01.800 INFO  uni-meter.controller     - creating GenericHttp input device
25-04-13 18:00:06.071 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-04-13 18:00:08.027 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-04-13 18:00:09.406 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em-d2303a300641
```

Can you please try to restart the uni-meter addon manually and check if the error is than also gone in your environment?

I will try to improve this in the next version ...

sdeigm, 2025-04-14:
You can try to add the uni-meter manually, but I think the Hoymiles storage also tries to look up the mDNS information and the action will not succeed.

But that 400 error from above is again different from the 502 error you were talking about after installing the pyscript addon. For me it seems as if the pyscript addon is not installed correctly.

Where is that /homeassistant/... path coming from? When I have a look to my environment, the configuration of the pyscript addon is directly stored in /config/pyscript:

[image]

If your pyscript addon is installed correctly, you can call the method directly from the developer tools using the yaml mode (please adjust the IP address to the IP address of your homeassistant system):

[image]

These are the parameters for copy & paste:

```
action: pyscript.uni_meter_mdns_register
data: {type: '_http', name: 'shellypro3em-fad951f44444', ip: '192.168.178.51', port: 80, properties: {'gen': 2, id: 'shellypro3em-fad951f44444'}}
```

When that call succeeds, the uni-meter should be announced via mDNS

romibe, 2025-04-14:
Hello, thank you for your time.
Asked KI, it menas:

❗ Ursache

Pyscript läuft in einer eingeschränkten Umgebung innerhalb von Home Assistant. Bestimmte Standard-Python-Module wie socket, os, subprocess, threading, etc. sind aus Sicherheitsgründen nicht erlaubt.

Darum schlägt der Import von socket fehl – und dein ganzes Skript wird nicht geladen. Deshalb siehst du auch den Dienst pyscript.uni_meter_mdns_register nicht in der Liste.
✅ Lösung: Ersatz für socket.inet_aton

Da socket.inet_aton(ip) lediglich die IP-Adresse in Byteform umwandelt, brauchen wir eine Alternative ohne socket. Glücklicherweise akzeptiert AsyncServiceInfo in Home Assistant auch normale IP-Adressen als bytes, wenn man es manuell umwandelt.

So i let rewrite your script. With new one in /homeassistant/config/pyscript/uni-meter-mdns.py it finally works.

[image]

I did not dig in the new script as im not doing python/sockets, but here it is.

`
from homeassistant.components import zeroconf
from zeroconf.asyncio import AsyncServiceInfo

def ip_to_bytes(ip):
    return bytes(int(x) for x in ip.split("."))

@service
def uni_meter_mdns_register(type=None, name=None, ip=None, port=None, properties=None):
    log.info(f"[uni-meter-mdns] Register called: type={type}, name={name}, ip={ip}, port={port}, properties={properties}")
    if not all([type, name, ip, port]):
        log.error("[uni-meter-mdns] Missing one or more required parameters.")
        return
    try:
        aiozc = zeroconf.async_get_async_instance(hass)
        info = AsyncServiceInfo(
            type_=f"{type}._tcp.local.",
            name=f"{name}.{type}._tcp.local.",
            addresses=[ip_to_bytes(ip)],
            port=int(port),
            properties=properties or {}
        )
        aiozc.async_register_service(info)
        log.info("[uni-meter-mdns] Service registered successfully.")
    except Exception as e:
        log.error(f"[uni-meter-mdns] Failed to register service: {e}")

@service
def uni_meter_mdns_unregister(type=None, name=None, ip=None, port=None, properties=None):
    log.info(f"[uni-meter-mdns] Unregister called: type={type}, name={name}")
    try:
        aiozc = zeroconf.async_get_async_instance(hass)
        info = AsyncServiceInfo(
            type_=f"{type}._tcp.local.",
            name=f"{name}.{type}._tcp.local.",
            addresses=[ip_to_bytes(ip)],
            port=int(port),
            properties=properties or {}
        )
        aiozc.async_unregister_service(info)
        log.info("[uni-meter-mdns] Service unregistered.")
    except Exception as e:
        log.error(f"[uni-meter-mdns] Failed to unregister service: {e}")

@service
def uni_meter_unregister_all():
    log.info("[uni-meter-mdns] Unregistering all services.")
    try:
        aiozc = zeroconf.async_get_async_instance(hass)
        aiozc.async_unregister_all_services()
        log.info("[uni-meter-mdns] All services unregistered.")
    except Exception as e:
        log.error(f"[uni-meter-mdns] Failed to unregister all services: {e}")

`

sdeigm, 2025-04-14:
Ok, great that you figured this out! I will adust the script accordingly.

So the reason why this did not happen in my environment is that I am running the pyscript addon with less security restrictions:

[image]

I was not aware that this has an impact on my script.

sdeigm, 2025-04-14:
Can this be a firewall issue? UDP Port 5353 has to be open

romibe, 2025-04-14:
Yes. Hass & HM Stotage are on same network.
You mean UDP open because mDNS goes oitside? Im on mobile connection over 5G in Gartenkolonie. Thats why Nabucasa. And the shelly plug i bought works on it. Have no firewall for intranet communication.

How can i find out the ip adress of emulated shelly to enter it in hoymiles app manually? Thank you.

sdeigm, 2025-04-14:
It is the IP address of the Home Assistant system if the uni-meter is running as addon within the Home Assistant system.

sdeigm, 2025-04-14:
To see data from the uni-meter try

http://192.168.1.175/rpc/EM.GetStatus?id=0

To get more info add a logback.xml file with the following content in the same directory as uni-meter.conf:

```
<?xml version="1.0" encoding="UTF-8"?>
<Configuration status="WARN">
    <appender name="STDOUT" class="ch.qos.logback.core.ConsoleAppender">
        <encoder>
            <pattern>%d{yy-MM-dd HH:mm:ss.SSS} %-5level %-24logger - %msg%n</pattern>
        </encoder>
    </appender>

	<appender name="FILE" class="ch.qos.logback.core.FileAppender">
		<file>/var/log/uni-meter.log</file>
		<append>true</append>
		<encoder>
            <pattern>%d{yy-MM-dd HH:mm:ss.SSS} %-5level %-24logger - %msg%n</pattern>
		</encoder>
	</appender>

    <root level="INFO">
        <appender-ref ref="STDOUT" />
        <appender-ref ref="FILE" />
    </root>

    <logger name="uni-meter.output"  level="DEBUG" additivity="false">
        <appender-ref ref="STDOUT" />
        <appender-ref ref="FILE" />
    </logger>
</Configuration>
```

The hoymiles storage does two things: it contacts the uni-meter via HTTP and compares that data to the data gathered via mDNS. In you system the mDNS data is missing.

romibe, 2025-04-14:
ok. thanks a lot. i think i'll give up. 

i can read values from http://192.168.1.175/rpc/EM.GetStatus?id=0

logs shows recently 

-----------------------------------------------------------
 Add-on: uni-meter
 A universal electrical meter converter
```
-----------------------------------------------------------
 Add-on version: 1.1.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.1  (amd64 / qemux86-64)
 Home Assistant Core: 2025.4.2
 Home Assistant Supervisor: 2025.04.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[17:40:03] INFO: using custom configuration file: uni-meter.conf
25-04-14 17:40:07.582 INFO  uni-meter                - ##################################################################
25-04-14 17:40:07.611 INFO  uni-meter                - # Universal electric meter converter 1.1.5 (2025-04-11 08:31:17) #
25-04-14 17:40:07.614 INFO  uni-meter                - ##################################################################
25-04-14 17:40:07.615 INFO  uni-meter                - initializing actor system
25-04-14 17:40:10.158 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-04-14 17:40:11.624 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-04-14 17:40:11.751 INFO  uni-meter.controller     - creating HomeAssistant input device
25-04-14 17:40:18.043 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
25-04-14 17:40:19.303 INFO  uni-meter.mdns.ha        - successfully registered mdns service shellypro3em-ee4eb13ef4c3
25-04-14 17:40:20.325 WARN  org.apache.pekko.http.impl.engine.client.PoolId - [0 (WaitingForResponseEntitySubscription)]Response entity was not subscribed after 1 second. Make sure to read the response `entity` body or call `entity.discardBytes()` on it -- in case you deal with `HttpResponse`, use the shortcut `response.discardEntityBytes()`. POST /core/api/services/pyscript/uni_meter_mdns_register Strict(198 bytes) -> 200 OK Default(2 bytes)
```

Is there any addon to emulate EcoTracker for MS-A2?

sdeigm, 2025-04-14:
Yesterday I added Eco-Tracker support and in my environment it is alreay running :-)

But also there hoymiles queries mDNS information so that will not help you.

Can you please tell me your exact network setup? Above you told 

"Im on mobile connection over 5G in Gartenkolonie. Thats why Nabucasa. And the shelly plug i bought works on it. Have no firewall for intranet communication."

So for the initial connection between your Hoymiles storage and the Home Assistant system, they once have to be in the exact same network since the mDNS traffic is not routed. Perhaps you can move your storage ones to the location of the Home Assistant system and do the inital pairing there. Then you can move your storage again if it still gets the same IP address in the other place.

romibe, 2025-04-14:
both ms-h2 and hass VM on the same and only network routed by Huawei LTE Router over Telekom LTE.

[image]

So Hass iss 192.168.1.175 and Hoymiles 192.168.1.222.
As the connections over mobile are NAT-es im using nabucasa to accces from out there
https://•••••••••••••••••.ui.nabu.casa

Setup works good for 3 years with diff services, rest, hooks usw. The shelly plug i bought today is on the same network.
Actions with pyscript runs green.

[image]

Here some nmap logs, cant see Hoymiles there but it deff. there.

[image]

sdeigm, 2025-04-14:
You have to explicitly enable UDP broadcast on the LTE router:

https://info.support.huawei.com/hedex/api/pages/EDOC1100331435/AEM10132/04/resources/dc/udp-helper_enable.html

sdeigm, 2025-05-05:
I hope you were able to fix your issuses and close this.

closed 2025-05-05 by sdeigm