sdeigm/uni-meter #137: Add uni-meter so shelly-cloud for connection to Anker Solix2?

martiko70, 2025-05-01:
Hi,
I really like uni-meter and thought it would solve my problem to act as a proxy to connect my Tibber-Puls to my Anker Solix2.

But unfortunately, Anke Solix does not locally search for Shelly Pro 3EM but connects to the Shelly Cloud and searches for compatible devices...

As far as I have read in the comments here, this is not (yet?) possible...

Is this a planned feature for a future release, or do I have to search for an different solution?

Thanks
Martin

sdeigm, 2025-05-02:
Hi Martin, unfortunately that is something I can not implement. The interface Shellies are using to update the cloud is not documented and the Shelly devices doing so are using encrypted SSL connections. So no reverse engineering is possible. I fear you have to search for a different solution :-(

closed 2025-05-02 by sdeigm

lnxbil, 2025-08-24:
> Hi Martin, unfortunately that is something I can not implement. The interface Shellies are using to update the cloud is not documented and the Shelly devices doing so are using encrypted SSL connections. So no reverse engineering is possible. I fear you have to search for a different solution :-(

Maybe one can add a man-in-the-middle proxy an upload its SSL certificate [via the API](https://shelly-api-docs.shelly.cloud/gen2/General/SSLsupportForOutboundConnections)?