sdeigm/uni-meter #139: Dyness Junior Box support

julianxv, 2025-05-02:
I'm using a Dyness Junior Box and it's only working with the Shelly Pro 3 em but im using the Shelly 3em so this should be perfect. I added it in Home Assistant and I get values on the status page of uni-meter. But unfortunatly there are no Values in the Dyness App. Whats the proper Way to Debug it? Listening with Wireshark how the Dyness Junior Box is trying to fetch data from the shelly?

sdeigm, 2025-05-02:
Yes, either use wireshark to debug the traffic and send me information or set up a logback.xml file where you set the level for the "uni-meter.output" logger to "TRACE" as described in the "Troubleshooting" documentation. Then the uni-meter logfile should provide information which HTTP/RPC methods are called by the Dyness Junior Box.

julianxv, 2025-06-04:
It recognizes it but it doesn't use it. I tried to configure it on the Box while I Was capturing packets but there are still no packets from the Box.

sdeigm, 2025-06-04:
I did a quick search for "Dyness" and "Shelly Pro3EM" and found this:

"Der Speicher arbeitete im Test mit den Smart Plugs und dem Energiemonitor 3EM Pro von Shelly zusammen. Dafür musste man unter den MQTT-Einstellungen in der Shelly-App manuell den Dyness-Account hinterlegen. Die Lastkurve eines Verbrauchers wurde nur mit Zeitverzug und in groben Schritten ab 150 Watt ausgeglichen. Für unseren Testaufbau war die Erkennung zu grob gestrickt."

Is it possible, that the Dyness Box expects a communication

Shelly Pro3EM -> MQTT (Dyness Cloud) -> Dyness Box

Is that somewhere documented? 

The MQTT output interface of the Shelly Pro3EM is not yet implemented. That would need an enhancement.