sdeigm/uni-meter #140: Shelly 3EM Pro Emulation via Home Assistant - Error Log and Configuration Issue

Crown3r, 2025-05-03:
Hey,
I’m new to this topic and a complete beginner. I believe I’ve successfully emulated the Shelly 3EM Pro via Home Assistant. I wanted to test this first before buying a Marstek battery storage. I’m getting the data via Tibber Pulse, but accessing the entities/values through Home Assistant. So far, the Shelly is detected by the Shelly app in my home network, but I can’t select it as it appears greyed out. This may not be a problem, though. However, I’m still getting the following error in the log:

`25-05-03 17:45:18.752 ERROR uni-meter.http.port-80 - exception in HTTP server: class org.apache.pekko.http.scaladsl.model.HttpEntity$Default cannot be cast to class org.apache.pekko.http.javadsl.model.HttpEntity$Strict (org.apache.pekko.http.scaladsl.model.HttpEntity$Default and org.apache.pekko.http.javadsl.model.HttpEntity$Strict are in unnamed module of loader 'app')`

Here’s my config:

```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"
    udp-port = 1010
    udp-interface = "0.0.0.0" # default, can be omitted

  input = "uni-meter.input-devices.home-assistant"

  input-devices {
    home-assistant {
      url = "http://192.168.178.31:8123"
      access-token = "eyJhbGciO....XXX"

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"

      power-sensor = "sensor.stromverbrauch_haus_2" # Power consumption + generation from balcony solar system in W
      energy-consumption-sensor = "sensor.tibber_pulse_das_haus_energie" # Current power consumption without solar system, i.e. grid usage
      energy-production-sensor = "sensor.tibber_pulse_das_haus_stromerzeugung" # Feed-in in W
    }
  }
}
```

Trinsic82, 2025-05-03:
Hello everyone,

I have exactly the same problem. I also use the HomeAssistant integration. My config looks like this:

```
uni-meter {
 output = "uni-meter.output-devices.shelly-pro3em"
 input = "uni-meter.input-devices.tasmota"
  
  # ...
 output-devices {
 shelly-pro3em {
 }
  }

  input-devices {
 tasmota {
 url = "http://192.168.7.237"
      # username=""
      # password=""
 power-json-path = "$.StatusSNS.eHZ.Power"
 power-scale = 1.0 # default, can be omitted
 energy-consumption-json-path = "$.StatusSNS.eHZ.E_in"
 energy-consumption-scale = 1.0 # default, can be omitted
 energy-production-json-path = "$.StatusSNS.eHZ.E_out"
 energy-production-scale = 1.0 # default, can be omitted
 }
  }  
}
```

When I call up the link:
http://<home-assistant-ip>/rpc/EM.GetStatus?id=0
I also get the correct values displayed.

In the Shelly app I can also see the Shelly 3EM (when I search via IP), but cannot add it, it is grey.

In the log I also get this message:

> 25-05-03 18:06:42.559 ERROR uni-meter.http.port-80 - exception in HTTP server: class org.apache.pekko.http.scaladsl.model.HttpEntity$Default cannot be cast to class org. apache.pekko.http.javadsl.model.HttpEntity$Strict (org.apache.pekko.http.scaladsl.model.HttpEntity$Default and org.apache.pekko.http.javadsl.model.HttpEntity$Strict are in unnamed module of loader ‘app’)

Thanks for your help!

sdeigm, 2025-05-04:
Hi, in the Shelly App the uni-meter can only be added, but then stays in the 'Pending connection' state. That is the normal behavior. You will not be able to receive any data using the Shelly App since therefore a cloud connection of the virtual shelly would be necessary which is not possible.

You can only check your virtual shelly locally by for instance opening the following URL with your broswer as @Trinsic82 already did:

``http://<uni-meter-ip>/rpc/EM.GetStatus?id=0``

I am currently not able to reproduce your error message " exception in HTTP server ...". When do you get that error message? When you try to add the uni-meter to the Shelly App or do you get it permanently?

Trinsic82, 2025-05-04:
Hi,

Thank you for the clarification. In my case, the error message always comes up when I press “Check device” in the Shelly app when adding. The error message comes up every time I press this button in the app. Otherwise it does not appear in the log.

So everything seems to be working fine and I'm looking forward to linking the virtual Shelly with the Markstek Venus.

Thanks for your great work !

Regards

sdeigm, 2025-05-04:
Ok @Crown3r can you also check if you get the meter readings from you home assistant system using

```http://<uni-meter-ip>/rpc/EM.GetStatus?id=0```

and close this issue if you get the correct data?

By the way, the configuration of the udp-port in your sample config from above is in a wrong location. It has to be moved to the shelly-pro3em output device section:

```
uni-meter {
  output-devices {
    shelly-pro3em {
      udp-port = 1010
    }
  }
}
```

Crown3r, 2025-05-04:
Hey, thanks! I correct the config.

I also getting data from the system:

[image]

Im still not quit sure, which sensors are correct:

      power-sensor = "sensor.current_power" -> current power including - power of pv production in W which would be the same as enery-consumption-power?

      energy-consumption-sensor = "sensor.energy_imported" -> power needed from grid? In W or kWh?

      energy-production-sensor = "sensor.energy_exported" --> only the production sensor from my pv or the amount in kWh which I exporting? Or the power in W?

Greets and thanks!

sdeigm, 2025-05-05:
The power-sensor should provide the current power usage in W. It should be positive if you are getting power from the grid and negative if you are delivering power to the grid. If you have a split setup with one (positive) sensor providing the power consumed from the grid and one positive sensor for the power delivered to the grid, you have to additionally configure the power-production-sensor. In that case the current power is calculated from both sensors as:

"current power" = "power-sensor" - "power-production-sensor"

The energy-consumption-sensor and energy-production-sensor should provide the energy consumed from / delivered to the grid in Wh. If your home assistant system delivers them as kWh you have to scale the to Wh using the scaling factor.

closed 2025-05-07 by sdeigm