sdeigm/uni-meter #141: Visibility of Home Assistant Power Sensor data [help wanted]

ChrisLambert1, 2025-05-03:
Hi,
first of all thanks for this shelly emulator, which I implemented in my HAS environment right now.
Just to checkout the functionality of the emulator and not having a storage system (like Hoymiles) available, I'm trying to check out the functionality of the emulator and looking for answers on following questions:

- The connection to the emulated device is set to 'Pending connection' 
- I'm able to get the emulated Shelly Pro 3EM in the Shelly app, but within the Shelly App I've no data (No data available)

And I wonder, if this is a normal behaviour (no real connection and no data is shown by the Shelly app)?
Is it possible to print the transmitted values on a terminal or protocol window in home assistant, to check the link between a HAS sensor?

Thank you very much!

sdeigm, 2025-05-04:
Hi, yes this is the expected behavior. The Shelly App expects that the Shelly builds up a connection to the cloud. That is not possible for the uni-meter since that is an undocumented protocol real shellies do on an encrypted connection. That behavior can not be emulated and so the virtual Shelly stays in the state 'Pending connection'.

The Hoymiles and Marstek storages only need a local connection to the Shelly. You can check if your uni-meter delviers the correct electrical meter readings by opening the following URL in your Browser:

``http://<uni-meter-ip>:<shelly-http-port>/rpc/EM.GetStatus?ip=0``

The shelly-http-port can be omitted if it is 80.

ChrisLambert1, 2025-05-04:
Hey,thank you for the fast reply, that was the missing hint 👍 !

closed 2025-05-04 by ChrisLambert1