sdeigm/uni-meter #147: Docker Compose Error -  "host" network_mode is incompatible with port_bindings

martin-st-81, 2025-05-07:
Wenn man Uni-Meter mittels DockerCompose starten möchte schlägt dies mit dem Error
"host" network_mode is incompatible with port_bindings
fehl.

Wenn man in dem file docker-compose.yaml die ports entfernt - startet das Uni-Meter, allerdings wird es dann nicht vom Hoymiles MS-A2 gefunden

[image]

martin@Frigate:~/docker-compose/Uni-Meter$ sudo docker-compose up
Pulling uni-meter (sdeigm/uni-meter:latest)...
latest: Pulling from sdeigm/uni-meter
f18232174bc9: Pull complete
2d619958c952: Pull complete
e6df77c4327c: Pull complete
Digest: sha256:d767b195a18f6c5d1b7c23d6abcbcd2cb7a017ca975363cbabb0e0a2bd3ef7cb
Status: Downloaded newer image for sdeigm/uni-meter:latest
Creating uni-meter ...

ERROR: for uni-meter  "host" network_mode is incompatible with port_bindings

ERROR: for uni-meter  "host" network_mode is incompatible with port_bindings
Traceback (most recent call last):
  File "/usr/bin/docker-compose", line 33, in <module>
    sys.exit(load_entry_point('docker-compose==1.29.2', 'console_scripts', 'docker-compose')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 81, in main
    command_func()
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 203, in perform_command
    handler(command, command_options)
  File "/usr/lib/python3/dist-packages/compose/metrics/decorator.py", line 18, in wrapper
    result = fn(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 1186, in up
    to_attach = up(False)
                ^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 1166, in up
    return self.project.up(
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/project.py", line 697, in up
    results, errors = parallel.parallel_execute(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/parallel.py", line 108, in parallel_execute
    raise error_to_reraise
  File "/usr/lib/python3/dist-packages/compose/parallel.py", line 206, in producer
    result = func(obj)
             ^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/project.py", line 679, in do
    return service.execute_convergence_plan(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/service.py", line 559, in execute_convergence_plan
    return self._execute_convergence_create(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/service.py", line 473, in _execute_convergence_create
    containers, errors = parallel_execute(
                         ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/parallel.py", line 108, in parallel_execute
    raise error_to_reraise
  File "/usr/lib/python3/dist-packages/compose/parallel.py", line 206, in producer
    result = func(obj)
             ^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/service.py", line 478, in <lambda>
    lambda service_name: create_and_start(self, service_name.number),
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/service.py", line 457, in create_and_start
    container = service.create_container(number=n, quiet=True)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/service.py", line 330, in create_container
    container_options = self._get_container_create_options(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/service.py", line 939, in _get_container_create_options
    container_options['host_config'] = self._get_container_host_config(
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/service.py", line 1014, in _get_container_host_config
    return self.client.create_host_config(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/docker/api/container.py", line 596, in create_host_config
    return HostConfig(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/docker/types/containers.py", line 336, in __init__
    raise host_config_incompatible_error(
docker.errors.InvalidArgument: "host" network_mode is incompatible with port_bindings

sdeigm, 2025-05-08:
Hallo, der Fehler liegt an dem von dir verwendeten ``docker-compose``. Es ist zwar richtig, dass die ports Sektion in host Netzwerkmodus nicht ausgewertet wird, aber alle neueren docker Versionen ignorieren sie einfach bzw. weisen nur durch eine Warnung darauf hin. Ich denke das wird auch bei dir passieren, wenn du anstelle von ``docker-compose up`` das bei neueren Versionen eigentlich zu verwendende ``docker compose up`` ohne Bindestrich zwischen dem docker und dem compose verwendest.

Mit dem Nicht-Finden des uni-meters durch den Hoymiles Storage wird es komplizierter. Da kann es eine Reihe von Gründen geben. Bei mDNS ist es zwingend notwendig, dass die mDNS Broadcast Pakete an Port 5353/udp auch den Docker Container erreichen. Das klappt nur im Host Netzwerkmodus und auch nur wenn Docker selbst nicht schon eine Bridge verwendet wie das z.B. unter Windows (und ich glaube auch Mac) sowie z.B. bei Portainer der Fall ist. Was verwendest du denn genau? Einen puren Docker-Daemon auf einem Raspberry Pi oder etwas anderes? Weiterhin kann es der Fall sein, dass auf dem Raspberry Pi schon ein Avahi Daemon läuft, so dass die mDNS Pakete den Docker Container nicht erreichen. 

Ganz im Notfall kannst du einen Avahi Daemon auf dem Host-Computer instalieren unn das Service-File, das uni-meter im /etc/avahi/services Verzeichnis im Docker Container anlegt aus dem Container herauskopieren und in das /etc/avahi/services Verzeichnis des Host Computers legen. Dann sollte es auch funktionieren.

Aber teil mir vielleicht erst mal mit in was für einer Umgebung der Container bei dir läuft, dann können wir weiter schauen ...

martin-st-81, 2025-05-08:
Danke für deine rasche Hilfe

Ich verwende eine Debian 12 Server mit Docker 28.0.4
Mit docker compose up startet der Container. In der Shelly App wird der Shelly3ProEM jetzt auch im WLAN gefunden.
In der Hoymiles App allerdings nicht, auch wenn die IP manuell eingegeben wird. 

Denke aber, wenn die Shelly App das Device findet sollte der Avahi Daemon eigentlich passen.

martin@Frigate:~/docker-compose/Uni-Meter$ docker version
Client: Docker Engine - Community
 Version:           28.0.4
 API version:       1.48
 Go version:        go1.23.7
 Git commit:        b8034c0
 Built:             Tue Mar 25 15:07:22 2025
 OS/Arch:           linux/amd64
 Context:           default

[image]

martin-st-81, 2025-05-08:
danke für die Hilfe und Bestätigung, dass es jetzt eigentlich klappen sollte und der Fehler nicht mehr am Docker Container liegt

sdeigm, 2025-05-09:
Gut, dass du es so zum Laufen bekommen hast. Irgendwie wird bei dir also der mDNS Verkehr nicht in den Container geleitet. Ich habe es bei mir noch einmal auf dem Raspberry Pi ausprobiert und da funktioniert es mit der Konfiguration, ohne das ich den avahi Daemon des Host-Rechners verwenden muss. Es muss also irgendwie am Network-Setup des Docker Daemon bei dir auf dem Debian Server liegen. Da kann ich so aus der Entfernung leider nicht viel helfen.

Interessant wäre wenn du den avahi-daemon stoppst und den uni-meter Docker Container startest, ob bei dir dann irgendwer auf dem mdns Port lauscht. Bei mir auf dem Pi sieht das so aus (evtl. muss du bei dir mdns durch 5353 ersetzen):

```
sdeigm@node-01:~ $ netstat -a | grep mdns
udp        0      0 0.0.0.0:mdns            0.0.0.0:*
udp6       0      0 [::]:mdns               [::]:*
sdeigm@node-01:~ $
```

martin-st-81, 2025-05-09:
Ich habe den lokalen avahi-daemon jetzt testhalber komplett deinstalliert und den Server neu gestartet
Die Hoymiles App findet den MSA2 jetzt nicht mehr

[image]

wenn der UniMeter Docker Container beendet wird, schaut es so aus
[image]

sdeigm, 2025-05-09:
Ok, das lauscht also auch ohne avahi-daemon und uni-meter etwas auf dem mdns Port. Habe mal gegoogelt. Auf einem Debian Server scheint es auch einen mdnsd-Daemon zu geben (https://installati.one/install-mdnsd-debian-12/).

Kannst du den vielleicht einmal deinstallieren bzw. mit ``sudo systemctl stop mdnsd`` (oder ähnlich) dekativieren? Dann sollte ohne uni-meter bei ``netstat -a | grep mdns`` überhaupt nichts mehr gefunden werden. Anschließend dann mal den Docker Container starten und schauen ob es funktioniert?

martin-st-81, 2025-05-09:
das Service mdnsd ist nicht installiert

[image]

Ich muss leider nachträglich ergänzen, dass hier noch drei andere Docker Container laufen
Wenn ich alle anderen Container beende und der UniMeter alleine läuft, wird der Storage aber leider trotzdem nicht gefunden

[image]

Wenn ich auch den UniMeter Container beende, lauscht dann gar nichts mehr

[image]

In den anderen Containern sind bei dem docker-compose file, die ports mit zB "8050:8050" eingetragen
aber leider auch mit - siehe unten - funktioniert es leider nicht

services:
  uni-meter:
    image: sdeigm/uni-meter:latest
    container_name: uni-meter
    network_mode: host
    restart: unless-stopped
    ports:
      - "80:80/tcp"
      - "1010:1010/udp"
      - "5353:5353/udp"
    expose:
      - "80/tcp"
      - "1010/udp"
      - "5353/udp"
    volumes:
      - ./uni-meter.conf:/etc/uni-meter.conf
      - ./logback.xml:/opt/uni-meter/config/logback.xml

sdeigm, 2025-05-09:
Und was bekommst du, wenn du zunächst die Container Id bestimmst:

```
sdeigm@node-01:~ $ docker ps
CONTAINER ID   IMAGE                    COMMAND                  CREATED        STATUS        PORTS     NAMES
34d3950f87da   sdeigm/uni-meter:1.1.9   "/opt/uni-meter/bin/…"   18 hours ago   Up 14 hours             uni-meter
sdeigm@node-01:~ $
```

und dann mit der Container-Id ein inspect aufrufst (da interessieren insbesondere die "NetworkSetting"):

```
sdeigm@node-01:~ $ docker inspect 34d3950f87da
[
 ....
        "NetworkSettings": {
            "Bridge": "",
            "SandboxID": "f2563a56ccdddbb0ea873f27069c112ad07c3609379e8331fee36a4b5c9567b0",
            "SandboxKey": "/var/run/docker/netns/default",
            "Ports": {},
            "HairpinMode": false,
            "LinkLocalIPv6Address": "",
            "LinkLocalIPv6PrefixLen": 0,
            "SecondaryIPAddresses": null,
            "SecondaryIPv6Addresses": null,
            "EndpointID": "",
            "Gateway": "",
            "GlobalIPv6Address": "",
            "GlobalIPv6PrefixLen": 0,
            "IPAddress": "",
            "IPPrefixLen": 0,
            "IPv6Gateway": "",
            "MacAddress": "",
            "Networks": {
                "host": {
                    "IPAMConfig": null,
                    "Links": null,
                    "Aliases": null,
                    "MacAddress": "",
                    "DriverOpts": null,
                    "NetworkID": "96f9943af0aec33765da88c55dc53a4605e997066ab49b3c654f9d7ca1fa4205",
                    "EndpointID": "8cd0e46efb641a3353d73908a4f02f4859e6557fda51ad7128af246b85aee245",
                    "Gateway": "",
                    "IPAddress": "",
                    "IPPrefixLen": 0,
                    "IPv6Gateway": "",
                    "GlobalIPv6Address": "",
                    "GlobalIPv6PrefixLen": 0,
                    "DNSNames": null
                }
            }
        }
    }
]
```

martin-st-81, 2025-05-09:
Es ist auffällig, dass zB beim Container "frigate" die Ports mit "docker ps" angezeigt werden, bei UniMeter nicht kA
Bei den anderen Container sind keine Ports definiert

Falls du denkst es liegt nicht an dem Unimeter Container, würde ich es dabei belassen den Avahi Daemon einfach lokal laufen zu lassen.

Wenn gewünscht kann ich aber auch gerne weitere Infos liefern 

[image]

[image]

martin-st-81, 2025-05-09:
Kann es ev. daran liegen, dass ich eine MAC vorgebe ?
beim lokalen avahi-daemon adaptiere ich diese auch in den beiden files

[image]

[image]

martin-st-81, 2025-05-10:
Habe jetzt testhalber das file "shellypro3em-B327EB36xxxx.service" aus dem Container kopiert und testhalber statt den beiden anderen lokal (/etc/avahi/services/) eingesetzt. -> klappt problemlos ... MSA2 wird in der Hoymiles App gefunden

Da der avahi-daemon eigentlich nur für die Ersteinrichtung benötigt wird und anschließend eigentlich deaktiviert werden kann, da die Verbindung direkt via IP erfolgt, könnte man dies ggf. in die Doku schreiben. Damit andere User welche den Docker Container verwenden wollen, hier nicht in das Problem laufen.

martin-st-81, 2025-05-10:
Ich konnte das Verhalten eingrenzen und zumindest für mich lösen.
Es muss am lokalen Host der avahi-daemon mittels apt install avahi-daemon installiert werden.
In dem Ordner /etc/avahi/services dürfen aber keine files liegen.
Wenn dann der UniMeter Docker gestartet wird, ist der Shelly3ProEM in der Hoymiles App sichtbar und kann hinzugefügt werden.

closed 2025-05-10 by martin-st-81

martin-st-81, 2025-05-10:
[image]

[image]

martin-st-81, 2025-05-12:
@sdeigm 
Am Host wurde der avahi-daemon deinstalliert und im docker-compose file "privileged: true" ergänzt.
Der MSA2 Speicher kann jetzt problemlos gefunden und eingerichtet werden

services:
  uni-meter:
    image: sdeigm/uni-meter:latest
    container_name: Uni-Meter
    network_mode: host
    privileged: true
    restart: unless-stopped
    ports:
      - 80:80/tcp
      - 1010:1010/udp
      - 5353:5353/udp
    expose:
      - 80/tcp
      - 1010/udp
      - 5353/udp
    volumes:
      - ./uni-meter.conf:/etc/uni-meter.conf
      - ./logback.xml:/opt/uni-meter/config/logback.xml

sdeigm, 2025-05-12:
Ok, das ist wirklich interessant was du da herausgefunden hast. Ich werde das Flag auf jeden Fall mal mit einem Komentar in das docker-compose File aufnehmen. Allerdings benötigt man das Flag nicht überall. Ich denke bei dir liegt es daran, das auf deinem Debian Host AppArmor läuft, dann hat Docker standardmässig Sicherheitseinschränkungen (könnte auch SELinux sein was bei dir läuft, aber AppArmor ist unter Debian wahrscheinlicher).

Kannst du mal mit ``aa-status`` überprüfen ob bei dir AppArmor läuft? Falls ja könntest du es dann mit ``aa-teardown`` herunterfahren und dann schauen ob das ganze auch ohne das privileged Flag läuft?

sdeigm, 2025-05-12:
Bzw. vielleicht müsste es auch ``systemctl stop apparmor`` sein.

martin-st-81, 2025-05-13:
Habe dem Docker jetzt eine eigene manuelle IP zugordnet und somit kann man etwaige Port bzw MDNS Überschneidungen mit anderen Containern am gleichen Host ausschließen. Pairing und Betrieb läuft damit stabil.

Denke diese Variante sollte in den meistens Konstellationen am problemlosesten funktionieren. 

services:
  uni-meter:
    image: sdeigm/uni-meter:latest
    container_name: Uni-Meter
    restart: unless-stopped
    networks:
      my_network:
        ipv4_address: 192.168.25.100
    ports:
      - 80:80/tcp
      - 1010:1010/udp
      - 5353:5353/udp
    expose:
      - 80/tcp
      - 1010/udp
      - 5353/udp
    volumes:
      - ./uni-meter.conf:/etc/uni-meter.conf
      - ./logback.xml:/opt/uni-meter/config/logback.xml

networks:
    my_network:
        driver: macvlan
        driver_opts:
          parent: enp1s0
        ipam:
          config:
            - subnet: 192.168.25.0/24
              gateway: 192.168.25.1