sdeigm/uni-meter #149: Shelly 3EM as output device

LukeTheMast3r, 2025-05-11:
Hi sdeigm,

would it be possible to implement a Shelly 3EM as an output device?

Want to try to simulate one for my "Indielux Stromwächter" it only accepts the old 3EM.

Pro is online and does not work :(

Greetings

sdeigm, 2025-05-12:
Hi, sure I can try to add support for your device. But I need some more information. What is the current problem? I thought that the new Pro 3EM is an enhancement of the old 3EM and that all interfaces of the old version are still supported. So my questions are

* Do you have a problem, that the "Indielux Stromwächter" is not discoverying the uni-meter? If that is the problem, how is the uni-meter discovered? Via mDNS, via Shelly Cloud or do you directly specifiy the IP address of the uni-meter?

* If the uni-meter is discovered but not working, you should enable the TRACE output as described [here](https://github.com/sdeigm/uni-meter#troubleshooting). Then we get the information, which REST/RPC calls are triggered and currently not implemented.

Looking forward to get this information ...

LukeTheMast3r, 2025-05-12:
Hey, so actually i left the project unattendet overnight and after work i started over again.

The device is still not automaticly discovered. But i added my Homeassistant IP within the "Stromwächter" configuration.
That solved the problem and the "Stromwächter" showed the actual power readings.

It seems to be a bit slow, but should not be a uni-meter problem i think.
But for educational purposes: How fast do you process the data within uni-meter from a Homeassistant instance?

My Setup is also a bit strange .... but works now ;)
eastron meter -> modbus Ethernet converter - NodeRed- Homeassistant - Uni-Meter - Stromwächter 

Thanks for offering The help.
It could be advantageous to mention the "Stromwächter" in your supported devices section if you like.

Used uni-meter.conf:

`uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.home-assistant"

  input-devices {
    home-assistant {
      url = "http://192.XXX.XXX.XXX:8123"
      access-token = "ey....."

      power-phase-mode = "tri-phase"
      energy-phase-mode = "mono-phase"

      power-l1-sensor = "sensor.activepower_l1"
      power-l2-sensor = "sensor.activepower_l2"
      power-l3-sensor = "sensor.activepower_l3"
      energy-consumption-sensor = "sensor.totalimportactiveenergy"
      energy-production-sensor = "sensor.totalexportactiveenergy"
    }
  }
}`

Greetings from Germany

sdeigm, 2025-05-13:
In the default configuration the uni-meter polls the home assistant sensors once a second. You can configure this by setting the 

``polling-interval = 1000ms``

configuration value within the home-assistant section. But I think it is not the polling frequency. I guess your setup is a little bit "sluggish" due to the large number of hops in your setup. 

Is the input device a Eastron SDM630? The single phase SDM120 device via Protoss PE11 Modbus converter is already directly supported. Would be easy to add the SDM630, but I have no test device. I guess that would make your setup a little bit more lively. 

I guess the reason why the device is not automatically discovered is the wrong mDNS information send by the uni-meter. It announces itself as shelly-pro3emXXXXXX. So I will add a Shelly 3EM output device as you intially request which announces itself as shelly-3emXXXX and basically provides the same interface.

LukeTheMast3r, 2025-05-16:
The input device is an SDM 72D-M

Tried to set the polling interval. But no noticeable change. The values in HomeAsssistant are updated like within a second. So fast enugh. Even on the android app with widgets. - so not a problem for now :)

Actually I Have the problem, that the Stromwächter is in error mode While there is negative energy.

Within the debug section of uni-meter everything seems to be OK. So i have to investigate within the Stromwächter. I'l give an update as soon as i know more.

Have seen (within debug mode) that unimeter provides the Frequency and Voltage as output. Is it possibe to configure these values as input data so that they also can be delivered ... same as the power and energy?

Greetings