sdeigm/uni-meter #151: Solaredge input device provides power values with wrong sign

sdeigm, 2025-05-14:


hendrikmuhs, 2025-05-15:
I was pleasantly surprised that my SE energymeter is supported out of the box. But then was wondering why my battery charged instead of discharged. Now I know why :-)

Great project, watching for the resolution of this issue, happily act as tester if needed.

sdeigm, 2025-05-16:
I should have mentioned it in the comment to this issue. There is already a workaround for the problem. See this [comment](https://github.com/sdeigm/uni-meter/issues/78#issuecomment-2878935603)

closed 2025-05-16 by sdeigm

sdeigm, 2025-05-16:
There is now a snapshot build available which should fix this issue:

[GithHub Packages](https://github.com/sdeigm/uni-meter/packages/2336213)

as well as a docker container:

sdeigm/uni-meter:1.1.10-SNAPSHOT

Would be great if you can test that build.

It should not be necessary, but you can change the sign of the power values by the configuration parameter

``invert-power = true``

which defaults to false.

hendrikmuhs, 2025-05-16:
:+1: works great, thank you!