sdeigm/uni-meter #153: Support a "don't charge" as well as a "don't discharge" mode

sdeigm, 2025-05-16:


closed 2025-05-27 by sdeigm

VolMi, 2025-05-30:
Hi, I was excited to see this in the changelog but I wonder if this is even possible?
I use Home Assistant to generate a signal to throw at uni-meter and my storage is a Marstek Venus E (finally arrived this week).

Just yesterday I tried to achieve the same thing with manipulations to my HA signal, but finally decided that this is impossible if the actual power input or power draw of the storage is unknown. The grid power can be both positive or negative with the power at the storage being any value imaginable. You must know the current power of the battery to decide which value the Shelly signal (a delta between the current and target value) is allowed to have.

How does your logic work?

And: Thank you sooo much for this great project! I excitedly set uni-meter up a month ago, knowing I'll have a several weeks wait for the arrival of the storage. :-)

sdeigm, 2025-05-31:
Hi @VolMi, you are correct that it is in general not possible to control the storage by manipulated electrical meter readings when you don't have the information at what rate the storage is currently charging/discharging. An electrical meter reading of 0 watts just tells the storage to keep its current state.

In this feature I am using the fact, that the storage (at least the Hoymiles storage does this) falls back to a default behavior when it does not receive any readings from the electrical meter anymore. When I want to prevent the storage from charging, the uni-meter does not send any negative electrical meter readings to the storage. Instead it stops answering the requests from the storage. So after a short while the storage stops to charge/discharge until it again gets a positive electrical meter reading which then brings the storage to discharge mode again. 

As said with my hoymiles storage that works in an acceptable manner, would be great if you could test the feature with your Marstek Venus and provide some feedback.