sdeigm/uni-meter #156: 2 Tibber pulse, one readable, one not

rschoell, 2025-05-24:
Hi, 
I've got 2 Tibber PUlse in the house, one is readable by Uni-meter, the other one not. 
I've already reset the non-working, but still can't get it to work.

on the node/1/data Page of the working, the data looks like this:

[image]

For the one not working, I'm getting fully readable data in the webinterface:

[image]

How can I get the 2nd one into uni-meter?

Many thanks

closed 2026-03-11 by sdeigm

sdeigm, 2026-03-11:
@0xnor0 but your problem is slighly different again. Your pulse bridge provides the power values for all 3 phases 21.7.0, 41.7.0 and 61.7.0.

1-0:21.7.0*255(-000128.41*W)
1-0:41.7.0*255(000000.00*W)
1-0:61.7.0*255(000011.64*W)

The other examples from above always provided the combined phase power channel 16.7.0

1-0:16.7.0*255(000458.90*W)

So some more work to do :-)

rschoell, 2026-03-11:
Thank you very much. I can confirm, that text-mode works with TIBBER3 release. Like you say, the overall consumption is then divided by 3 currently. But that's ok for me for the moment.

sdeigm, 2026-03-12:
@0xnor0 you did nothing wrong. The current Pulse input device only handled the Obis channel 16.7.0 which contains the combined power data for all 3 phases. Then you can configure if you want to use that data on all three phases (divided by 3) or on a single phase. 

Your device is the first one that provides all 3 phases separately (21.7.0, 41.7.0, 61.7.0). In the version from yesterday I calculated the combined power of all 3 phases and distributed it using the same configuration logic like before.

I now changed the behavior for exactly your 3 phase case to publish the exact data for all 3 phases. But then the other configuration parameter phase-mode and power-phase are meaningless. I don't know if that's the best solution ...

0xnor0, 2026-03-12:
> [@rschoell](https://github.com/rschoell) yes I can also add support for your case. I found a good overview of the Obis codes:
> 
> https://tech.enectiva.cz/en/installation-instructions/others/obis-codes-meaning/
> 
> The difference between your codes 36.7.0 to 76.7.0 and [@0xnor0](https://github.com/0xnor0) 's obis codes 21.7.0 to 61.7.0 is that the first ones are signed while the second ones are always positive and you have additional 22.7.0 to 62.7.0 which contain the negative power values. I will support both cases.
> 
> But with these changes we are getting an incompatibility between the binary and the text mode pulse input. Since you [@rschoell](https://github.com/rschoell) seem to have both versions, can you volunteer to also test the binary mode if I am changing it accordingly?

My Data shows the following:
```
1-0:21.7.0*255(-000032.67*W)
1-0:41.7.0*255(000001.41*W)
1-0:61.7.0*255(000011.85*W)
1-0:1.7.0*255(-000018.68*W)
```
My balcony solar power system is connected to phase 1, but according to the code, the value shouldn't be negative. However, it is negative here. Or am I missing something?
`21.7.0	Positive active instantaneous power (A+) in phase L1 [kW]`

Or is all of this correct and it's the balance of positive and negative values?
Due to the backflow prevention, an exclamation mark is displayed and the positive values ​​are negative, but have no effect on the overall consumption.

GuenterD, 2026-03-12:
> Now there is a 1.4.0-TIBBER4 release and a docker container which should support all formats from this issue, including the multi phase format of [@0xnor0](https://github.com/0xnor0)

Thank you very much, TIBBER4 is working fine. One maybe minor thing I noticed is that ther is sometime rounding issue in total_power:  "total_power":-496.83000000000004  "total_power":-706.4399999999999,"

```
TIBBER4 is anna-lena@annalenaPC:~$ curl ShellyPro3EM-D0577BA1BD0F.local/status
{"wifi_sta":{"connected":true,"ssid":"kabellos","ip":"192.168.178.99","rssi":-83},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"09:59","unixtime":1773305995,"serial":1,"has_update":false,"mac":"D0577BA1BD0F","ram_total":50648,"ram_free":38376,"ram_lwm":32968,"fs_size":233681,"fs_free":174194,"uptime":4200,"temperature":28.08,"overtemperature":false,"tmp":{"tC":28.08,"tF":82.54,"is_valid":true},"modbus":{"enabled":false},"total_power":-496.83000000000004,"fs_mounted":true,"em:0":{"id":0,"a_current":-0.72,"a_voltage":230.00,"a_act_power":-165.61,"a_aprt_power":-165.61,"a_pf":1.00,"a_freq":50.00,"b_current":-0.72,"b_voltage":230.00,"b_act_power":-165.61,"b_aprt_power":-165.61,"b_pf":1.00,"b_freq":50.00,"c_current":-0.72,"c_voltage":230.00,"c_act_power":-165.61,"c_aprt_power":-165.61,"c_pf":1.00,"c_freq":50.00,"total_current":-2.16,"total_act_power":-496.83,"total_aprt_power":-496.83},"emdata:0":{"id":0,"a_total_act_energy":1309.19,"a_total_act_ret_energy":432.21,"b_total_act_energy":1309.19,"b_total_act_ret_energy":432.21,"c_total_act_energy":1309.19,"c_total_act_ret_energy":432.21,"total_act":3927.57,"total_act_ret":1296.63}}anna-lena@annalenaPC:~$ 
anna-lena@annalenaPC:~$ curl ShellyPro3EM-D0577BA1BD0F.local/status
{"wifi_sta":{"connected":true,"ssid":"kabellos","ip":"192.168.178.99","rssi":-83},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"10:11","unixtime":1773306675,"serial":1,"has_update":false,"mac":"D0577BA1BD0F","ram_total":50648,"ram_free":38376,"ram_lwm":32968,"fs_size":233681,"fs_free":174194,"uptime":4879,"temperature":28.08,"overtemperature":false,"tmp":{"tC":28.08,"tF":82.54,"is_valid":true},"modbus":{"enabled":false},"total_power":-705.99,"fs_mounted":true,"em:0":{"id":0,"a_current":-1.02,"a_voltage":230.00,"a_act_power":-235.33,"a_aprt_power":-235.33,"a_pf":1.00,"a_freq":50.00,"b_current":-1.02,"b_voltage":230.00,"b_act_power":-235.33,"b_aprt_power":-235.33,"b_pf":1.00,"b_freq":50.00,"c_current":-1.02,"c_voltage":230.00,"c_act_power":-235.33,"c_aprt_power":-235.33,"c_pf":1.00,"c_freq":50.00,"total_current":-3.06,"total_act_power":-705.99,"total_aprt_power":-705.99},"emdata:0":{"id":0,"a_total_act_energy":1309.19,"a_total_act_ret_energy":432.24,"b_total_act_energy":1309.19,"b_total_act_ret_energy":432.24,"c_total_act_energy":1309.19,"c_total_act_ret_energy":432.24,"total_act":3927.57,"total_act_ret":1296.72}}anna-lena@annalenaPC:~$ 
anna-lena@annalenaPC:~$ curl ShellyPro3EM-D0577BA1BD0F.local/status
{"wifi_sta":{"connected":true,"ssid":"kabellos","ip":"192.168.178.99","rssi":-83},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"10:11","unixtime":1773306695,"serial":1,"has_update":false,"mac":"D0577BA1BD0F","ram_total":50648,"ram_free":38376,"ram_lwm":32968,"fs_size":233681,"fs_free":174194,"uptime":4899,"temperature":28.08,"overtemperature":false,"tmp":{"tC":28.08,"tF":82.54,"is_valid":true},"modbus":{"enabled":false},"total_power":-706.4399999999999,"fs_mounted":true,"em:0":{"id":0,"a_current":-1.02,"a_voltage":230.00,"a_act_power":-235.48,"a_aprt_power":-235.48,"a_pf":1.00,"a_freq":50.00,"b_current":-1.02,"b_voltage":230.00,"b_act_power":-235.48,"b_aprt_power":-235.48,"b_pf":1.00,"b_freq":50.00,"c_current":-1.02,"c_voltage":230.00,"c_act_power":-235.48,"c_aprt_power":-235.48,"c_pf":1.00,"c_freq":50.00,"total_current":-3.06,"total_act_power":-706.44,"total_aprt_power":-706.44},"emdata:0":{"id":0,"a_total_act_energy":1309.19,"a_total_act_ret_energy":432.25,"b_total_act_energy":1309.19,"b_total_act_ret_energy":432.25,"c_total_act_energy":1309.19,"c_total_act_ret_energy":432.25,"total_act":3927.57,"total_act_ret":1296.75}}
```

sdeigm, 2026-03-12:
Ok, there is now a 1.4.0-TIBBER5 (container and release) which supports the obis channels from @rschoell and @0xnor0 in text mode.

The binary mode is not yet modified

rschoell, 2026-03-15:
@sdeigm Thanks for the explanation. I've not tested the auto detection on textmode yet, but binary mode is auto detected. Works like a charm. Many thanks for your work here.