sdeigm/uni-meter #161: possibly conflicting ports while polling data?

tobleu, 2025-05-30:
Hi! 

First of all thanks a lot for this great tool! I want to try to get it running with the battery Growatt NOAH 2000. I installed uni-meter on my Raspberry Pi which has also iobroker running. When executing the uni-meter.sh script after configuring for Shelly Pro 3EM with input from the iobroker API, I get the following error message: 

> ERROR uni-meter.input          - failed to execute polling: Tcp command [Connect(127.0.0.1/<unresolved>:8088,None,List(),Some(10 seconds),true)] failed because of java.net.ConnectException: Verbindungsaufbau abgelehnt

which can be resolved by changing the port of the output instance (i.e. Shelly) to 8088, resulting in another error:

> ERROR uni-meter.input          - Failed to parse response: Unrecognized token 'The': was expecting (JSON String, Number, Array, Object or token 'null', 'true' or 'false')

The input, however, works as expected, by delivering json data... 

Any idea, where the problem might be?

sdeigm, 2025-05-31:
Hi, would be great if you could provide your configuration here. Without your uni-meter.conf I have to guess a little bit.

The first error "ConnectException: Verbindungsaufbau abgelehnt" seems to be related to the fact that there is no IO Broker REST API listening on port 8088. I think you have configured it to listen on another port (or you haven't added that REST extension at all?).

Then you configured the uni-meter virtual shelly to provide its data on port 8088. Now your uni-meter is talking to itself. The input device part of the uni-meter queries the output device of the uni-meter. Since the uni-meter does not provide the information the IO Broker REST API normally provides, your are getting the second failure regarding the JSON parsing.

You have to make sure the the url configured in the generic-http section points to your IO Broker rest API like explained in the IO Broker sample. You can test that URL in a normal Web-Browser and if it is correct you should see a meaningful output. The URL in the sample already provides the correct structure. You just have to adjust it to match your IP-Address, your IO-Broker REST API port and your IO-Broker datapoint names.

See the [io-broker-sample](https://github.com/sdeigm/uni-meter/blob/main/samples/ioBroker/uni-meter.conf) line 18 to 25.

tobleu, 2025-05-31:
Thanks for your thoughts thus far! 

Here's my original config, for which the first error appears, the socond with the port 8088 for Shelly:

`uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.vz-logger"

  http-server {
    port = 8080
  }

  input-devices {
    generic-http {
      url = "http://192.168.178.78:8087/getBulk/smartmeter.0.1-0:1_8_0__255.value,smartmeter.0.1-0:2_8_0__255.value,smartmeter.0.1-0:16_7_0__255.value/?json"

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"

      channels = [{
        type = "json"
        channel = "energy-consumption-total"
        json-path = "$[0].val"
        scale = 0.001
      },{
        type = "json"
        channel = "energy-production-total"
        json-path = "$[1].val"
        scale = 0.001
      },{
        type = "json"
        channel = "power-total"
        json-path = "$[2].val"
      }]
    }
  }

  output-devices {
    shelly-pro3em {
      # udp-port = 1010 # ...
      # port = 8088
    }
  }
}
`

tobleu, 2025-05-31:
Okay, I set back the server port for uni-meter to 80 and configured the API to port 8088 (as this is obviously expected - despite different URL in the config) - changing also the port for polling in the config it now works without an error. 

Next thing is, that the ShinePhone App from Growatt does not find the Shelly: "Stellen Sie sicherm dass ihr Balkon-Energiespeicherung und der everHome EcoTracker im selben lokalen Netzwerk sind" (hm... vielleicht sollte ich es als EcoTracker konfigurieren - habe aber Shelly pro 3EM angetippt...)

tobleu, 2025-06-01:
Indeed, this was also what I realized yesterday. Unfortuately the battery went offline, so I cannot proceed at the moment. I will report back... Have a nice Sunday!

tobleu, 2025-06-02:
> Okay, I set back the server port for uni-meter to 80 and configured the API to port 8088 (as this is obviously expected - despite different URL in the config) - changing also the port for polling in the config it now works without an error.
> 
> After starting the necessary service there ist following message at port 80 (which was obviously fetched in the "self talking" case above - see "The"):
> 
> > The requested resource could not be found but may be available again in the future.

I think this should be traced back in code, in order to make sure, the correct port is used for polling - or was this the standard setting for vz-logger?

tobleu, 2025-06-02:
> Indeed, this was also what I realized yesterday. Unfortuately the battery went offline, so I cannot proceed at the moment. I will report back... Have a nice Sunday!

The battery is back online, but for integrating Shelly the app requires to log in at Shelly and authorize the data request - which seems to be impossible, without having a physical Shelly device.

Using Eco Tracker also no response is given - an error is thrown, that both devices do net seem to be in the same network :-(

tobleu, 2025-08-19:
@rschoell so what's the Avahi-config-script? Could you guide me? 
Would also be nice to have this info in the Wiki as well...
Thanks a lot in advance!

rschoell, 2025-08-20:
parallel to the "normal" uni-meter.sh script there's the uni-meter-and-avahi.sh script. this script also add's config files into /etc/avahi and executes the mDNS daemon on your system. This helps to announce your ecotracker service in the network and allow the Noah 2000 to find your Uni-Meter as an ecotracker. 
It definitively is listed somewhere in the doc's, but maybe not so directly visible. Partially mentions are inside the "baremetal" install, which I assume you used to set this up. I've also followed a lot of "issues" in here, which in github is always a second source of truth with lots of informations. 
Anyways. your uni-meter must be able to announce itself through mDNS. if that works and no blockings inside the network exist, you should be able to configure it. Another bonus: with the latest firmware 15.x.x.x, I've no more seen disconnects from the uni-meter!