sdeigm/uni-meter #162: Change simulated MAC address of Shelly in Home Assistant Add-on

SpaceTeddy, 2025-05-31:
Hi There,
with my current installation of unimeter I have to use a customized MAC address of the Shelly. For that, I use the /etc/avahi/services/shellypro3em.http.tcp.service file.
is this possible also in HA integration?

sdeigm, 2025-06-01:
Hi, with recent uni-meter versions it is not necessary anymore to customize the MAC address. The uni-meter automatically uses the mac address of the first detected physical network device. Anyhow you can still override this MAC address in the uni-meter.conf file as before.

The HA integration does not use the avahi daemon to announce the service via mDNS. Instead the zero-conf service of the HA system is used. For this to work you have to install the pyscript addon and copy the provided python script to its script directory as described [here](https://github.com/sdeigm/uni-meter/blob/main/doc/install/HomeAssistant.md#mdns-support)

SpaceTeddy, 2025-06-02:
Hi there,
Well, I guess I made all following the documentation, but unfortunately the MS-a2 did not recognize the unimeter shelly in HA. To get the Shelly information by http://<ha-ip>/status ect is not an issue, therefore I guess, that there is something wrong with the mdns service. 
May you give me a hint how to test, that mdns is well configured in HA? … copy the mdns py script to /config/pyscript is done, but I see no influence.

sdeigm, 2025-06-05:
Good morning, the list of services does not contain the service ``uni_meter_mdns_register`` which is used by the uni-meter to register the mDNS information. That can have multiple reasons:

* You did not install the pyscript integration at all
* You did not copy the ``uni-meter-mdns.py`` file into the pyscript configuration directory of your homeassistant instance:

[image]

* You did not enable the right to use all imports:

[image]

closed 2025-06-05 by SpaceTeddy