sdeigm/uni-meter #163: HA add-on receives http polling error

oldgitdaddy, 2025-06-07:
Hello,
I installed the add-on for home assistant and followed the instructions here: https://github.com/sdeigm/uni-meter/blob/main/doc/install/HomeAssistant.md

I receive constant update in the log with this error: 
25-06-07 17:27:33.614 ERROR uni-meter.input          - failed to execute http polling: The http server closed the connection unexpectedly before delivering responses for 1 outstanding requests

I did not setup mDNS, as I do not use hoymiles. I just have this simple configuration. I tried various URLs (local IP, localhost, etc.) and none of this worked. I am also not sure if it is a good idea to call http. At least I would prefer HTTPS instead.

```
uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"

  input = "uni-meter.input-devices.home-assistant"
  
   input-devices {
    home-assistant {
      url = "http://homeassistant:8123"
      access-token = "XXX"

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"

      power-sensor = "sensor.current_power"
      energy-consumption-sensor = "sensor.pconsume"
      energy-production-sensor = "sensor.psupply"
    }
  }
}
```

oldgitdaddy, 2025-06-07:
Quick update. Home assistant add-on works with external dynamic DNS name. But no access with internal IP. Not sure, why. Any idea? 

Anyway polling error was also coming from that the power-sensor was misconfigured. BTW: Thanks for the tool. ✌️

oldgitdaddy, 2025-06-08:
As it did not feel right with using the external dynamic url from my home assistant server, I switched to MQTT, which works out of the box. Again thanks!

sdeigm, 2025-06-11:
What exactly do you mean by internal IP? Do you mean http://supervisor?