sdeigm/uni-meter #173: add dynamic throttling

VolMi, 2025-06-17:
### Context
I am never happy when I play around with the `min-sample-period` setting. Actually, my lack of happiness is probably due the incoherent sampling rate of my [tibber pulse IR sensor via Home Assistant](https://github.com/marq24/ha-tibber-pulse-local). I do get occasional overshoots and I can reduce them a bit by increasing the min-sample-period, but then I have a much smaller update frequency than I would like, which is also noticable if I switch on or off a large electrical load.

<details>
<summary>In Home Assistant, I actually log timestamps when I got the last reading of the sensor via this simple template sensor:</summary>

```
  {% set s = states.sensor.tibber_local_0100100700ff %}
  {{ s.state and (as_timestamp(s.last_updated) - as_timestamp("2025-06-01 00:00:00+00:00")) | round(2) }}
```
</details>
Then I created a Statistics sensor to log the difference of the last two data points, so I get the time passed since the last update.
This way, I can create a plot like this which shows that it usually takes 2 or 4 seconds to update the power value, but there are frequent 6s delays and even outliers well beyond 10s:

[image]

At first, I did this to optimize exactly this value itself, by fiddling around everything that is possible with the Tibber Pulse itself. But this is about the best I can achieve.

### Feature Request
Now I was thinking that it should be possible for an uni-meter that is configured to read sensors from home assistant to read my last-update time stamp (or something similar but better) and use it for an adaptive replacement of the `min-sample-period`. That is, read the current power value and the time stamp every `Δt` and delay the output until the sensor value becomes less old than `Δt + eps`.

Does that make any sense?

schalterDev, 2026-09-07:
Same conclusion here, but coming from the opposite direction — which I think
makes the case stronger than an unstable source alone would.

My input is an ESPHome SML reader on the meter's optical port. Unlike your
Tibber Pulse, it is almost perfectly regular. Timestamps from the debug log:
- 10:05:42.102
- 10:05:46.511
- 10:05:50.972
- 10:05:55.381

Intervals: 4409 / 4461 / 4409 ms — fixed rate, ~50 ms jitter, no dropped
telegrams. Basically the best case a time-based throttle could hope for.

I started at `min-sample-period = 5000ms`. Against a 4409 ms source that drifts
by ~590 ms per cycle, so roughly every 7th–8th telegram two input updates land
in one throttle window and one measurement is never delivered to the storage at
all. Tightening it to 4600 ms — just above my source period — reduces that to
about every 23rd telegram, so the skipping is largely solved.

What remains is the part I find more damaging, and it follows directly from the
trigger being time-based:

The throttle window still runs free relative to the input, now drifting by
~191 ms per cycle. The age of the delivered value therefore sweeps slowly
through the full 0 to 4.4 s range with a period of roughly 100 seconds. The
storage sees a dead time that is never constant — it creeps up to ~4.4 s, snaps
back to near zero, and starts over. A control loop copes with constant dead time
far better than with dead time that keeps moving underneath it.

No time-based value avoids this. Matching the source period exactly would only
work if both clocks were locked, which they aren't, and it would have to be
re-tuned per meter anyway. So the issue isn't that your sensor is irregular —
it's that the trigger is time-based rather than sample-based, which is what your
adaptive proposal fixes.

### What the behaviour could look like

From a user's point of view the goal is simple: the storage should get exactly
one fresh reading per real measurement — no more, no less. No drift, no skipped
readings, no per-meter tuning.

Worth separating two kinds of input, because they differ in what's knowable:

- **Push-based inputs** (MQTT and similar): a new message already *is* a new
  measurement, so the event is available without extra information.
- **Polling inputs** (Home Assistant, Tasmota, generic HTTP): re-reading the
  same value looks identical to a genuine update. This is where your
  `last_updated` idea is needed — and as far as I can tell, HA already returns
  that timestamp today, so the information is there for the taking.

Something like this, with the existing behaviour as the default so nobody is
affected who doesn't opt in:

```hocon
shelly-pro3em {
  sample-mode   = "on-input-update"   # default: "throttle"
  max-hold-time = 15s                 # fallback if the input stalls
}
```

The fallback matters: if the source goes quiet, the storage still needs an
answer eventually.

### Offer, with a disclaimer

@sdeigm  — I'd be happy to put together a PR if you're open to the idea. Full
transparency: I'd be doing it with Claude Code, and I'm not a Java developer but
professional Frontend developer and able to understand the code.
I'd test it against my own setup (Marstek Venus E 3.0) and I'd keep the scope
tight, but you should review it on that basis. If you'd rather not have
AI-assisted contributions in this repo, that's entirely fair — just say so and
I'll leave it as a feature request.

If you are open to it, I'd want your view on a few things first:

1. Does this fit how you'd want it solved, or do you have a different design in
   mind?
2. Naming preference for the config option?
3. Tri-phase mode with three independently updating sensors. My suggestion would
   be to emit only once **every** sensor has produced a new sample since the last
   emit, with `max-hold-time` catching the case where one of them stalls.

   Reasoning: in the common case all three sensors come from the same physical
   meter, split into three entities. They update near-simultaneously, so firing
   on *any* fresh sensor would emit up to three times in quick succession — with
   a partially updated snapshot each time. That reproduces exactly the
   multiple-corrections-per-measurement problem this whole issue is about, and it
   hands the storage phase values of mixed age, which seems like a bad idea given
   that at least the Marstek does its own phase-detection logic on that data.

   The alternative — fire on any fresh sensor — is defensible when the three
   phases genuinely come from three independent sources with different update
   rates. "Wait for all" is then gated by the slowest one, so you'd lose update
   frequency on the two fast channels. If that setup is common in the field, it
   might be worth making it selectable rather than fixed.

   Caveat: I run mono-phase, so this is reasoning from the design rather than
   from experience. You'll have a much better sense of which setup is typical.

I'd start with the push-based inputs only and leave `last_updated` handling for
a follow-up.

sdeigm, 2026-09-08:
Hi, thanks for the lengthy input. First of all I have to say, that the min-sample-period configuration parameter was never intended to solve problems like you described above. The min-sample-period parameter is intended to solve problems, where the next parameter provided by the input device does not yet reflect the changes initiated by the steering storage which then leads to an oversteering. So in my case I have an optical reader which provides a new reading every second, but that reading does not yet contain the changes initiated by the storage. Therefore I am using a min-sample-period of 2500ms to ensure that the next answer to the storage already contains the feedback of the changed storage behaviour.

With an input reading interval of 4000ms like in your case, I would expect, that the next reading already contains the feedback to the changed loading/exporting behavior of the storage. But statistically you have an average "age" of 2000ms of that data, that's the only thing that would be minimized with that new mode. The failure of that aged data is in 50% of the cases in the wrong direction and in the other 50% in the correct direction. So I would not expected great improvements from that new mode, but we can add it. That's not a big change.

For the implementation details, I would

* not send a replacement value if the input stalls. That makes no sense. I would not send any data anymore. In that case all the storages I know fall back to their default behaviour to do nothing neither charge nor discharge. That's normally the best thing the storage can do

* For the problem with the independently arriving single phase values I would use a behavior like the TCP linger algorithm, which collects data for a very small time period and sends the collected data after a configurable delay period.

Since I am very critical against AI generated code I can not guarantee, that I will merge each pull request. There is already another AI generated PR I will not merge, since it is just a bad design. But would be interesting to see the AI generated solution. 

Anyhow I will think about a solution myself.

schalterDev, 2026-09-08:
Hi @sdeigm, I've opened #412 with an implementation of the mode, following your two points: no replacement value when the input stalls, and a linger period instead of waiting for all phases. It also contains a small addition to the Home Assistant input that makes the mode useful with a polling source.

I've been running it on my own setup (ESPHome SML reader via Home Assistant, Marstek Venus E 3.0) for a while: one reply per meter reading, age of the delivered value constant at about 0.1 s instead of drifting between 0.2 s and 3.6 s. Whether that makes the control loop noticeably calmer I can't say yet, that needs a longer comparison of the grid power, and I'll report back when I have it.

No rush at all, please have a look whenever you find the time. The option names are just proposals, and I'm happy to change anything about the design that doesn't fit the way you'd want it solved.

closed 2026-09-14 by sdeigm