sdeigm/uni-meter #179: Which sensors in uni-meter.config for Home Assistant?

skater175, 2025-06-21:
I have a Hichi IR reader (TASMOTA) that transmits its values ​​to Home Assistant via MQTT. The uni-meter add-on is installed and running in Home Assistant. Regarding the uni-meter.config, I'm not sure which Home Assistant sensors I should use. Which sensors are these usually:

power-phase-mode = "tri-phase"
power-l1-sensor = "sensor.tasmota_z1_power_p1"
power-l2-sensor = "sensor.tasmota_z1_power_p2"
power-l3-sensor = "sensor.tasmota_z1_power_p3"

energy-consumption-sensor = "sensor.tasmota_z1_zaehlerstand_total_z1"
energy-production-sensor = "sensor.tasmota_z1_einspeisung_total_z1"

I only have voltage and current for the three phases, and power as power_curr only as a total value. So what are the sensors sensor.tasmota_z1_power_p1, ... _p2, ... p3? Thank you in advance.

sdeigm, 2025-06-23:
Then you can use the power_curr sensor as power-l1-sensor. Just remove the other sensor mappings power-l2-sensor and power-l3-sensor. The virtual Shelly then provides the entire power data on phase 1 which should work for the Marstek and the Hoymiles storage.

skater175, 2025-06-23:
Ich frage jetzt mal in deutsch, damit ich bei den Sensoren nicht durcheinander komme: "sensor.tasmota_z1_power_p1" ist der Sensor, der die aktuelle Leistung (am Zähler) in Watt zeigt, also das, was in dem Moment in den Zähler rein oder raus geht... Hat also nichts mit Power im Sinne von Stromstärke (in Ampere) zu tun, oder? Und die anderen beiden Sensoren sind dem Namen entsprechend der Gesamtbezug aus und die Einspeisung ins Netz in kWh.

skater175, 2025-06-23:
Herzlichen Dank, dann hatte ich den  power-l1-sensor richtig verstanden. Inzwischen läuft auch alles und ich warte auf dem MS-A2.

sdeigm, 2025-06-24:
Ok, gut zu hören, dass alles läuft. Ich schließe das Ticket dann mal ...

closed 2025-06-24 by sdeigm