sdeigm/uni-meter #181: MSA 2 dont discover shelly EM3 emulator

Nicky85g, 2025-06-29:
bonjour la communauté. désolé pour mon anglais...
j'ai besoin d'aide et peut être pas tout compris je débute...
j'ai installé uni-meter dans home assistant pour simuler un shelly em 3 pro
pour faire fonctionner ma batterie MSA-2 ...
tout à l'air de fonctionner les données remonte bien  sur http://192.168.xxx.xxx/rpc/EM.GetStatus?id=0
mais impossible de se connecter a la batterie l'appli "S mile Home" demande à être connecté au même réseau wifi
mais HA est en Ethernet sur mini pc , est-ce important ou uni-meter le gère ?
voici mes log et fichier de config
Merci @sdeigm pour cet excellent travail

Hello community. sorry for my English...
I need help and maybe not all understand I start...
I installed uni-meter in home assistant to simulate a shelly em 3 pro
to run my MSA-2 battery...
Incidentally, the data is going well on http://192.168.xxx.xxx/rpc/EM.GetStatus? id 0
but impossible to connect to the battery the "S mile Home" app needs to be connected to the same WiFi network
But HA is in Ethernet on mini pc, is it important or uni-meter manages it?
These are my logs and config file.
Thank you @sdeigm sdeigm for this excellent work.

----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[16:50:36] INFO: using custom configuration file: uni-meter.conf
25-06-29 16:50:38.928 INFO  uni-meter                - ###################################################################
25-06-29 16:50:38.939 INFO  uni-meter                - # Universal electric meter converter 1.1.11 (2025-05-29 03:21:57) #
25-06-29 16:50:38.940 INFO  uni-meter                - ###################################################################
25-06-29 16:50:38.941 INFO  uni-meter                - initializing actor system
25-06-29 16:50:40.643 INFO  org.apache.pekko.event.slf4j.Slf4jLogger - Slf4jLogger started
25-06-29 16:50:41.503 INFO  uni-meter.controller     - creating ShellyPro3EM output device
25-06-29 16:50:41.547 INFO  uni-meter.controller     - creating HomeAssistant input device
25-06-29 16:50:43.882 INFO  uni-meter.output         - UDP server is listening on /[0:0:0:0:0:0:0:0]:1010
25-06-29 16:50:44.905 INFO  uni-meter.http.port-80   - HTTP server is listening on /[0:0:0:0:0:0:0:0]:80
------------------------------------------------------------------------------------------------------------------
uni-meter.conf

uni-meter {
  output = "uni-meter.output-devices.shelly-pro3em"
  input = "uni-meter.input-devices.home-assistant"
  

  output-devices {
     shelly-pro3em {
        udp-port = 1010
      }
    }

  input-devices {
    home-assistant {
         url = "http://192.168.xxx.xxx:8123"
         access-token = "mon token"
     

      power-phase-mode = "mono-phase"
      energy-phase-mode = "mono-phase"
     
      power-sensor = "sensor.routeur_source_rms_puissance_m_soutiree"
      energy-consumption-sensor = "sensor.routeur_source_rms_puissance_t_soutiree"
      energy-production-sensor = "sensor.routeur_source_rms_puissance_t_injectee"

    }
  }
}

sdeigm, 2025-06-30:
Hi, from your log I can see, that the mDNS configuration is missing. The Hoymiles storage discovers the Shelly using mDNS. So you have to add the pyscript Home Assistant Add-on and copy one python script to the correct location as described [here](https://github.com/sdeigm/uni-meter/blob/main/doc/install/HomeAssistant.md#mdns-support)

mDNS uses UDP broadcase packages. Not all switches forward these packages, so the Hoymiles storage and the uni-meter should be in the same network without any switches inbetween them.

Nicky85g, 2025-06-30:
Bonjour j'ai bien installé pyscript et le fichier mais il n'a pas l'air de
fonctionner.
Deplus il met qu'une instance est déjà démarré mais pas d'entités trouvé je
vais chercher de ce côté merci